espressif/arduino-esp32 #2942: Timer + Tasks + WiFi AP + SPIFFS = IllegalInstruction crash

ivoras, 2019-06-29:
### Hardware:
Board:							DEVKIT 1
Core Installation version:			1.0.2
IDE name:						Arduino IDE
Flash Frequency:					80MHz
PSRAM enabled:                                    no
Upload Speed:					921600
Computer OS:                                        Windows 10

### Description:

I have a simple LED audio visualisation project at https://github.com/ivoras/SevenBoard1 and everything was working fine until I tried adding WiFi AP functionality for configuration purposes.

Now I get random, very rare InstrFetchProhibited crashes, which I can't reproduce, and a reliable hard crash when trying to send a file to the AsyncWebServer client over WiFi. The code is:

```
  web.on("/", HTTP_GET, [](AsyncWebServerRequest *req) {
    Serial.println("Got request");
    req->send(SPIFFS, "/index.html", "text/html");
  });
```

and it crashes on the send() call with:

```
Got request
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40152abc: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40152abc: bad00bad bad00bad bad00bad
Guru Meditation Error: CoGuru Meditation Error: Core  1 panic'ed (
```

it doesn't finish the last printout, that's why it's truncated in the message above, and it hangs there, doesn't even reset the board.

This is devkit v1, the uploader sees it as:

```
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
```

### Sketch

https://github.com/ivoras/SevenBoard1

### Debug Messages:
```
starting setup
Setting up SPIFFS
Setting up WiFi
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 13 - AP_START
Web server started
setup over
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 15 - AP_STACONNECTED
dhcps: send_offer>>udp_sendto result 0
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 17 - AP_STAIPASSIGNED
Got request
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40152d10: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40152d10: bad00bad bad00bad bad00bad
Guru Meditation Error: CoGuru Meditation Error: Core  1 panic'ed (Guru Meditation Error: Core  1 panic'ed (Guru Meditation Error: Core  1 panic'ed (Unknown reason)
Guru Meditation Error: Core  1 panic'ed (
```

atanisoft, 2019-06-29:
This is expected if you are doing things inside an ISR that is not supposed to be done.

If you have a hardware timer active it will disable spi_flash access while it is running. The AsyncWebServer callbacks are not disabled during spi_flash disabled periods (and can't easily be disabled). You best option is disable the timer when web access is being used.

ivoras, 2019-06-29:
The ISR is to the best of my knowledge, fine - it only does ADC and memory access.

But the timer interfering with SPI ... that's another story. That looks surprisingly specific, any ideas why a hardware timer would disable all SPI flash activities by default?

ivoras, 2019-06-29:
And another question: since PROGMEM is on SPI flash, I don't suppose I could get around this by moving data from SPIFFS to PROGMEM?

... and for future reference, this also fails: https://github.com/espressif/arduino-esp32/issues/2943

atanisoft, 2019-06-29:
Serving web data from progmem is faster than spiffs but it would be a band-aid on the real problem.

All ISR code will lock spi_flash and is done intentionally. It is not specific to arduino-esp32 nor can it be bypassed.

Any memory access inside an ISR also must be uint32_t based and read-only. Anything you want to access as read-only should be tagged as DRAM_ATTR similar to the ISR itself being IRAM_ATTR. The ISR needs to be very fast so I'd suggest move ADC put of the ISR code and instead invoke it from a task that is blocking and woke up via ulTaskNotifyFromIsr (I think is the right name for it, there is a task notification API which works from an ISR context)

atanisoft, 2019-06-29:
As for #2943, you are sharing a very small memory space between your code and everything else that runs in IRAM and DRAM. There isn't much space left usually in that area.

ivoras, 2019-07-07:
Surrounding SPIFFS operations with timerStop(myTimer) and timerStart(myTimer) appears to solve this issue.

closed 2019-07-07 by ivoras