espressif/arduino-esp32 #2943: DRAM segment data does not fit when using PROGMEM

ivoras, 2019-06-29:
### Hardware:
Board:							DEVKIT 1
Core Installation version:			1.0.2
IDE name:						Arduino IDE
PSRAM enabled:                                    no
Computer OS:                                        Windows 10

### Description:

I have a couple of large structures - byte arrays of parts of a web page actually, which I thought I'd serve from PROGMEM via AsyncWebServer.

However, I get the "DRAM segment data does not fit" error -- which confuses me because:

1. The data is less than 100 KiB in size and I have more than 200 KiB of free memory even if it were to fit in RAM
2. The structures are declared with PROGMEM, which I thought would mark them for inclusion into the code portions of flash, not loaded into RAM.

So... why does this happen?

### Sketch:  
```
unsigned char bootstrap.min.css[] PROGMEM = {
...
};
```

### Debug Messages:
```

c:/users/ivoras/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\ivoras\AppData\Local\Temp\arduino_build_459698/SevenBoard1.ino.elf section `.dram0.bss' will not fit in region `dram0_0_seg'

c:/users/ivoras/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: DRAM segment data does not fit.

c:/users/ivoras/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: region `dram0_0_seg' overflowed by 8456 bytes
```

lbernstone, 2019-06-30:
PROGMEM is just a [compatibility word in esp32](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/pgmspace.h#L32).  Use the [Preferences library](https://github.com/espressif/arduino-esp32/tree/master/libraries/Preferences) to store data in flash (the NVS partition).

atanisoft, 2019-06-30:
@lbernstone he is embedding a file in the code not trying to store it in nvs/flash. Progmem is fine for this.

The issue though is there is something tagged as DRAM_ATTR that is exceeding the capacity of the dram segment declared in the ld file.

ivoras, 2019-06-30:
Nope, see EDIT2 above, and the memory sizes reported by the compiler.

OTOH, declaring the data as const together with PROGMEM works, in the sense that it doesn't get counted in "dynamic memory". 

That makes sense since, per ibernstone's link, PROGMEM is a no-op on ESP32.

ivoras, 2019-06-30:
I'll close this issue with that conclusion: PROGMEM is no-op on ESP32, it does nothing. Declaring a variable as const does something similar to what PROGMEM does in other platforms, makes the data reside in flash until it's needed, without special handling.

closed 2019-06-30 by ivoras