espressif/arduino-esp32 #2947: Error on running esp_now_init() after setCpuFrequencyMhz(10);

tdnet12434, 2019-07-01:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.1-rc4
IDE name:						IDF component
Flash Frequency:					40Mhz
PSRAM enabled:                                    no
Upload Speed:					921600
Computer OS:                                        Ubuntu

### Description:
Error on running esp_now_init() after setCpuFrequencyMhz(10);

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

// Not required for Arduino IDE
#include <esp_now.h>
#include <esp32-hal.h>
#include <esp32-hal-gpio.h>
#include <freertos/portable.h>
#include <freertos/task.h>
#include <HardwareSerial.h>
#include <stddef.h>
#include <freertos/semphr.h>

// IGNORE EVERYTHING ABOVE THIS LINE - NOT REQUIRED FOR ARDUINO IDE

TaskHandle_t Task0;
TaskHandle_t Task1;
SemaphoreHandle_t Semaphore;

#define LED0 27
#define LED1 25

// Global variables, available to all
static volatile unsigned long count0 = 0;
//static volatile unsigned long count1 = 0;

struct {
    unsigned long countA;
    unsigned long countB;
}volatile myCounters;

void loop0(void * parameter) {
    for (;;) {
        Serial.print("\t\t\t\t\t\t\t\tLoop 0 - Running on core: ");
        Serial.print(xPortGetCoreID());

        Serial.print(" (");

        xSemaphoreTake(Semaphore, portMAX_DELAY);
        Serial.print(++count0);
        xSemaphoreGive(Semaphore);

        Serial.println(")");

        digitalWrite(LED0, HIGH);
        delay(200);
        digitalWrite(LED0, LOW);
        //delay(300);
        // Above is Arduino-speak for:
        vTaskDelay(300 / portTICK_PERIOD_MS);
        // Note it is the TASK that is blocked not the entire thread
    }
}

void loop1(void * parameter) {
    for (;;) {
        Serial.print("Loop 1 - Running on core: ");
        Serial.print(xPortGetCoreID());

        Serial.print(" (");

        xSemaphoreTake(Semaphore, portMAX_DELAY);
        Serial.print(++count0);
        xSemaphoreGive(Semaphore);

        Serial.println(")");

        digitalWrite(LED1, LOW);
        delay(100);
        digitalWrite(LED1, HIGH);
        delay(100);
    }
}

// Init ESP Now with fallback
void InitESPNow() {
    //WiFi.disconnect();
    if (esp_now_init() == ESP_OK) {
        Serial.println("ESPNow Init Success");
    }
    else {
        Serial.println("ESPNow Init Failed");
        // Retry InitESPNow, add a counte and then restart?
        // InitESPNow();
        // or Simply Restart
    }
}
void setup()
{
    Serial.begin(115200);
    Serial.println("Setup started.");

    // Options are: 240 (default), 160, 80, 40, 20 and 10 MHz
    setCpuFrequencyMhz(10);
    int cpuSpeed = getCpuFrequencyMhz();
    Serial.println("Running at " + String(cpuSpeed) + "MHz");

    InitESPNow();
    pinMode(LED0, OUTPUT);
    pinMode(LED1, OUTPUT);

    // Simple flag, up or down
    Semaphore = xSemaphoreCreateMutex();

    xTaskCreatePinnedToCore(
            loop0, /* Function to implement the task */
            "Task0", /* Name of the task */
            1000, /* Stack size in words */
            NULL, /* Task input parameter */
            10, /* Priority of the task */
            &Task0, /* Task handle. */
            1); /* Core where the task should run */

    xTaskCreatePinnedToCore(
            loop1, /* Function to implement the task */
            "Task1", /* Name of the task */
            1000, /* Stack size in words */
            NULL, /* Task input parameter */
            //configMAX_PRIORITIES - 1, /* Priority of the task */
            1,
            &Task1, /* Task handle. */
            1); /* Core where the task should run */

    //xSemaphoreGive(Semaphore);
    Serial.println("Setup completed.");
}

void loop()
{
//  Suspending and resuming tasks...
//  static TickType_t previousWakeTime = xTaskGetTickCount();
//  static bool taskSuspended = false;
//
//  vTaskDelayUntil(&previousWakeTime, portTICK_PERIOD_MS * 5000);
//
//  if (taskSuspended) {
//    vTaskResume(Task1);
//    taskSuspended = false;
//    Serial.println("Task 1 resumed.");
//  } else {
//    xSemaphoreTake(Semaphore, portMAX_DELAY);
//    vTaskSuspend(Task1);
//    taskSuspended = true;
//    Serial.println("Task 1 suspended.");
//  }
//
//  previousWakeTime = xTaskGetTickCount();

    //delay(1);
    //taskYIELD();
    //vPortYield();
    //yield();
    vTaskDelete(NULL);
}
```

### Debug Messages:
```

PC: 0x400e1897: _vfprintf_r at ../../../.././newlib/libc/stdio/vfprintf.c line 1754
EXCVADDR: 0x00000044

Decoding stack results
0x400e1897: _vfprintf_r at ../../../.././newlib/libc/stdio/vfprintf.c line 1754
0x400d38df: search_object at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libgcc/unwind-dw2-fde.c line 985
0x400d3986: __register_frame_info_bases at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libgcc/unwind-dw2-fde.c line 98
0x40085201: panicPutHex at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 85
```

stickbreaker, 2019-07-01:
@tdnet12434 None of the Radio Functions(WiFi,Esp_now,BlueTooth) will work if the frequency is reduced below 80mhz,  The low speed clock modes are only provided to reduce power consumption when preforming stand alone tasks.  Any task that needs high preformance(SD,SPI,Wifi ..) needs the processors at full speed.

Chuck.

stickbreaker, 2019-07-01:
@tdnet12434 When you want to power consumption, you first have to shut down all radio functions, then reduce clock speed. When you want to reuse the Radio, first restore clock speed, then start up the Wifi or Esp_Now.

Chuck.

lbernstone, 2019-07-01:
You missed a not in there, Chuck.  The radio functions will NOT work at cpu frequencies below 80MHz.

tdnet12434, 2019-07-01:
Thank you @stickbreaker. I'm trying to reduce wifi bandwidth to 10mhz as show in [here](https://hackaday.com/2019/01/04/underclocking-the-esp8266-leads-to-wifi-weirdness/). His result make me want to try espnow in stealth mode.

stickbreaker, 2019-07-01:
That example was on a 8266 not esp32?  The Esp32's radio code is not the same.

Chuck.

tdnet12434, 2019-07-02:
Ok, then i close this issue for now. Thanks @stickbreaker krub. :)

closed 2019-07-02 by tdnet12434