espressif/arduino-esp32 #2949: ESP32 I2C PCF8574 read issue

Gustavomurta, 2019-07-01:
In February 2018, I ran my sketch (arduino IDE) in my project - ESP32 digital voltmeter and it worked fine - Arduino IDE 1.8.5 and ESP32 Board version 1.0.0.
The communication of the digital voltmeter is made with PCF8574 I2C interface. 
Pullups resistors are 3K3 ohms (VCC = 3.3V)

Links of my project:
[https://www.elektormagazine.com/labs/esp32-digital-voltmeter#/comments/labs/1459](url)
[https://github.com/Gustavomurta/ESP32-DVM](url)

### Hardware:
Board: DOIT ESP32 DEVKIT V1
Core Installation version: 1.0.2
IDE name: Arduino IDE 1.8.9
Flash Frequency: 80 MHz
PSRAM enabled: NO
Upload Speed:  115200
Computer OS:  Windows 10

The sketch I use to read ESP32 voltmeter  is:
[https://github.com/Gustavomurta/ESP32-DVM/blob/master/ESP32_DVM7135_PT_V1.ino](url)

Simple as -
`<pre>
<font color="#00979c">void</font> <font color="#000000">ReadData</font> <font color="#000000">(</font><font color="#000000">)</font>
<font color="#000000">{</font>
 &nbsp;<font color="#d35400">Wire</font><font color="#434f54">.</font><font color="#d35400">begin</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font> &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#434f54">&#47;&#47; Start I2C</font>
 &nbsp;<font color="#d35400">Wire</font><font color="#434f54">.</font><font color="#d35400">requestFrom</font><font color="#000000">(</font><font color="#000000">PCF8574</font><font color="#434f54">,</font> <font color="#000000">1</font><font color="#000000">)</font><font color="#000000">;</font> &nbsp;&nbsp;<font color="#434f54">&#47;&#47; Read One Byte from PCF8574</font>
 &nbsp;<font color="#000000">pcfData</font> <font color="#434f54">=</font> <font color="#d35400">Wire</font><font color="#434f54">.</font><font color="#d35400">read</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font> &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#434f54">&#47;&#47; Byte Read</font>
<font color="#000000">}</font>

</pre>`

Now with Arduino 1.8.9 and ESP32 Board version 1.0.2, the I2C interface  runs in crazy mode. The SCL pin sends continuous pulses and the SDA sends erroneous data (checked with logic analyzer). 
My ESP32 is DOIT ESP32 DevKit V1 and the all hardware is OK and checked with my I2C scanner. (based on Nick Gammon).

[https://github.com/Gustavomurta/ESP32-DVM/blob/1bc07438e89eac39ae83f6d824451ad7248b3536/ESP32_I2Cscanner2.ino](url)

Messages at Arduino Serial Console:
```rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:8424
ho 0 tail 12 room 4
load:0x40080400,len:5868
entry 0x4008069c

I2C scanner. Scanning ...
Found address: 32 (0x20)
Found 1 device(s).
```

Researching this problem with the Debug option, I found these messages. 
But I have not identified the error yet. No error is being detected.
I suspect there is now a conflict between I2C and PWM Channel 0 that I use to generate clock of 480 KHz to TLC 7135 (Voltmeter chip). Note that my project worked fine with old libraries. 

```
rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:8424
ho 0 tail 12 room 4
load:0x40080400,len:5868
entry 0x4008069c
[D][esp32-hal-i2c.c:1337] i2cProcQueue():  Busy Timeout start=0xd6, end=0xd6, =0, max=50 error=0
[E][esp32-hal-i2c.c:319] i2cDumpI2c(): i2c=0x3ffbebe0
[I][esp32-hal-i2c.c:320] i2cDumpI2c(): dev=0x60013000 date=0x16042000
[I][esp32-hal-i2c.c:322] i2cDumpI2c(): lock=0x3ffb84ec
[I][esp32-hal-i2c.c:324] i2cDumpI2c(): num=0
[I][esp32-hal-i2c.c:325] i2cDumpI2c(): mode=1
[I][esp32-hal-i2c.c:326] i2cDumpI2c(): stage=3
[I][esp32-hal-i2c.c:327] i2cDumpI2c(): error=0
[I][esp32-hal-i2c.c:328] i2cDumpI2c(): event=0x3ffb8570 bits=0
[I][esp32-hal-i2c.c:329] i2cDumpI2c(): intr_handle=0x3ffb85a0
[I][esp32-hal-i2c.c:330] i2cDumpI2c(): dq=0x3ffb854c
[I][esp32-hal-i2c.c:331] i2cDumpI2c(): queueCount=1
[I][esp32-hal-i2c.c:332] i2cDumpI2c(): queuePos=0
[I][esp32-hal-i2c.c:333] i2cDumpI2c(): errorByteCnt=0
[I][esp32-hal-i2c.c:334] i2cDumpI2c(): errorQueue=0
[I][esp32-hal-i2c.c:335] i2cDumpI2c(): debugFlags=0x00020000
[I][esp32-hal-i2c.c:289] i2cDumpDqData(): [0] 7bit 20 R STOP buf@=0x3ffbfeb8, len=1, pos=0, ctrl=11110
[I][esp32-hal-i2c.c:307] i2cDumpDqData(): 0x0000: .                                00 
[I][esp32-hal-i2c.c:347] i2cDumpInts(): 0 row	count	INTR	TX	RX	Tick 
[I][esp32-hal-i2c.c:438] i2cTriggerDumps(): after ProcQueue
[I][esp32-hal-i2c.c:347] i2cDumpInts(): 0 row	count	INTR	TX	RX	Tick 
```

The same topic at ESP32 forum:
[https://www.esp32.com/viewtopic.php?f=19&t=11261#p45725](url)

Can someone help me? Thanks in advance.

atanisoft, 2019-07-01:
@stickbreaker

Gustavomurta, 2019-07-01:
Logic analyzer diagram with old libraries- works fine (feb/2018) 
https://github.com/Gustavomurta/ESP32-DVM/blob/master/ESP32_DVM7135_data01.JPG

Now I2C runs in wrong mode. The SCL pin sends continuous pulses and the SDA returns with erroneous data.
https://github.com/Gustavomurta/ESP32-DVM/blob/master/ESP32_DVM7135_bug%20at%20I2C.JPG

Gustavomurta, 2019-07-01:
**Correcting important information!**

I was able to do some ESP32 Voltmeter tests on my Laptop.
The Arduino IDE version is 1.8.5 and the CORE version 1.0.0.
The symptom is different, but it does not work either! Only now, Over Voltage does not appear.
The I2C interface still has a problem.

I remembered that the first version of Core I used was the previous version of 1.0.0.
And I used the old method to install the package:
Old method for installing ESP32 in the Arduino IDE:
https://jgamblog.wordpress.com/2018/02/08/conhecendo-o-esp32-usando-arduino-ide/

I will try to recreate the environment I used when the project worked well.

atanisoft, 2019-07-01:
@Gustavomurta many of the LCD libraries use this same chip IIRC. I haven't seen any problems with LCDs on ant version of arduino-esp32. I suspect it is a bug in the PCF8574 library you are using.

stickbreaker, 2019-07-01:
This is the error message.  It is saying the I2C bus was not in an idle state when it attempted to use it.  The Busy signal is generated when the I2C peripheral see activity on the i2c bus.
>[D][esp32-hal-i2c.c:1337] i2cProcQueue():  Busy Timeout start=0xd6, end=0xd6, =0, max=50 error=0

try reporting the errors.
```c++

void ReadData (){

  if( Wire.begin()){                   // Start I2C
    uint8_t count =Wire.requestFrom(PCF8574, 1);   // Read One Byte from PCF8574
    if(count ==1 ) {
       pcfData = Wire.read();          // Byte Read
    }
    else {
      Serial.printf(" Wire.requestFrom() failed=%d(%s)\n",Wire.lastError(),Wire.getErrorText(Wire.lastError());
    }
  }
  else {
    Serial.printf(" Wire.begin() failed\n");  
  }
}
```

Chuck.

Gustavomurta, 2019-07-01:
Hi Atanisoft, 
I´m not using PCF8574 Library . Only wire library.
#include <Wire.h>

stickbreaker, 2019-07-01:
Verify your pins,  `Wire.begin()` without specifying pins uses the defaults defined in the selected board's `pins_arduino.h` file. 

Chuck.

Gustavomurta, 2019-07-01:
> Verify your pins, `Wire.begin()` without specifying pins uses the defaults defined in the selected board's `pins_arduino.h` file.
> 
> Chuck.

In my Sketch 

# define SDApin 21     // PCF8574 SDA pin15       => GPIO_21
# define SCLpin 22     // PCF8574 SCL pin14       => GPIO_22
# define PCF8574 0x20  // PCF8574 Address

Serial.begin(115200);
  attachInterrupt(digitalPinToInterrupt(STB), TestData, FALLING); // External Interrupt GPIO_4
  Wire.begin (SDApin, SCLpin);           // sda= GPIO_21 /scl= GPIO_22

void ReadData ()
{
  Wire.begin();                   // Start I2C
  Wire.requestFrom(PCF8574, 1);   // Read One Byte from PCF8574
  pcfData = Wire.read();          // Byte Read
}

stickbreaker, 2019-07-01:
put the error test in and see what is returned.

Chuck.

stickbreaker, 2019-07-01:
Why do you code a `Wire.begin()` in the read code?  It is only needed at boot to configure hardware, or after a bus failure.  `Wire.begin()`  Is used to recover from a hung bus. Change your read code to this:
```c++
void ReadData (){
   uint8_t count =Wire.requestFrom(PCF8574, 1);   // Read One Byte from PCF8574
    if(count ==1 ) {
       pcfData = Wire.read();          // Byte Read
    }
    else {
      Serial.printf(" Wire.requestFrom() failed=%d(%s)\n",Wire.lastError(),Wire.getErrorText(Wire.lastError());
    }
}

```
if during the `Wire.begin()` it senses a hung bus, the recovery procedure is to produce upto 9 clock cycles on SCL while monitoring SDA as soon as SDA goes high the recovery procedure ends.

Chuck.

Gustavomurta, 2019-07-01:
> This is the error message. It is saying the I2C bus was not in an idle state when it attempted to use it. The Busy signal is generated when the I2C peripheral see activity on the i2c bus.
>
>
> try [...]

```
ESP32_DVM7135_Chuck:76:113: error: expected ')' before ';' token

       Serial.printf(" Wire.requestFrom() failed=%d(%s)\n", Wire.lastError(), Wire.getErrorText(Wire.lastError());
```

Gustavomurta, 2019-07-01:
> ```c++
> void ReadData (){
>    uint8_t count =Wire.requestFrom(PCF8574, 1);   // Read One Byte from PCF8574
>     if(count ==1 ) {
>        pcfData = Wire.read();          // Byte Read
>     }
>     else [...]

What is wrong with this line ?
Serial.printf(" Wire.requestFrom() failed=%d(%s)\n", Wire.lastError(), Wire.getErrorText(Wire.lastError());

Gustavomurta, 2019-07-01:
I corrected that !

stickbreaker, 2019-07-01:
Add another ')' before the semicolon, my mistake.

stickbreaker, 2019-07-01:
drop the extra `Wire.begin()` from the read function.

Gustavomurta, 2019-07-01:
Thanks for your patience

stickbreaker, 2019-07-01:
Did removing the extra `Wire.begin()` change anything?

Gustavomurta, 2019-07-01:
Serial Console message 
```
 Wire.requestFrom() failed=5(BUSY)
```

stickbreaker, 2019-07-01:
so, something is happening on the bus,  The busy error is generated by the hardware peripheral.  Try looking at the signal with your scope, something is not right.  
What exact board have you selected in the arduino enviroment? and What exact hardware board are you using?  I am thinking there is some mismatch between the hardware SCL/SDA pins and the software defined pins.

Gustavomurta, 2019-07-01:
I´m verifying the interface I2C with Logic analyzer. 
Now the ESP32 try to read PCF8574, but cannot print reading voltages.

Hardware:
Board: DOIT ESP32 DEVKIT V1
Core Installation version: 1.0.2
IDE name: Arduino IDE 1.8.9
Flash Frequency: 80 MHz
PSRAM enabled: NO
Upload Speed: 115200
Computer OS: Windows 10

Gustavomurta, 2019-07-01:
The pins of I2C are correct and verified ! 

My ESP32 is DOIT ESP32 DevKit V1 and the all hardware is OK and checked with my I2C scanner. (based on Nick Gammon).

https://github.com/Gustavomurta/ESP32-DVM/blob/1bc07438e89eac39ae83f6d824451ad7248b3536/ESP32_I2Cscanner2.ino

Messages at Arduino Serial Console:

´´´
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:8424
ho 0 tail 12 room 4
load:0x40080400,len:5868
entry 0x4008069c

I2C scanner. Scanning ...
Found address: 32 (0x20)
Found 1 device(s).
´´´

Gustavomurta, 2019-07-01:
I can not understand that.
Why are there SCL pulses sometimes even when there is no I2C communication?

Gustavomurta, 2019-07-01:
For me this is strange, no?

stickbreaker, 2019-07-01:
ok, from the doit variant it defines the pins correctly,

Can you copy your full sketch into a [glist](https://gist.github.com/)  so I can review it.  The symptoms/problem you are reporting has not be seen by anyone else.  I wrote the I2C subsystem. I use it extensively, I am biased :grin:, but I am suspecting there is something else interfering. 

Chuck.

Gustavomurta, 2019-07-01:
The sketch I use to read ESP32 voltmeter is:
https://github.com/Gustavomurta/ESP32-DVM/blob/master/ESP32_DVM7135_PT_V1.ino

I suspect there is now a conflict between I2C and PWM Channel 0 that I use to generate clock of 480 KHz to TLC 7135 (Voltmeter chip). Note that my project worked fine with old libraries.

stickbreaker, 2019-07-01:
the pulses you are seeing is the i2c subsystem trying to clear the bus.  If a READ operation is in progress and there is a esp32 reset, the slave i2c device could be in the middle of outputting a request. It could be driving SDA low.  When the ESP32 boots and `Wire.begin()` is executed and it sees SCL /SDA not in an idle state it tries to recover the bus.  It does this by manually generating SCL pulses in [checklinestate](https://github.com/espressif/arduino-esp32/blob/7d7824701fe5e22f08555d3e1ce3180a922b2151/cores/esp32/esp32-hal-i2c.c#L1400).  I think that is what you are seeing.

Gustavomurta, 2019-07-01:
What is meaning hal in esp32-hal-i2c.c?

stickbreaker, 2019-07-01:
ok, now I know what is going on, you are calling `Wire()` from an interrupt, your interrupt is triggering while `Wire()` is processing the the prior interrupt.

>What is meaning hal in esp32-hal-i2c.c?

Hardware Abstraction Layer

Gustavomurta, 2019-07-01:
Thanks. 

https://www.elektormagazine.com/labs/esp32-digital-voltmeter#/comments/labs/1459

How digital section of my project works:

Digital Circuit  :

In my design, the  STROBE, RUN/HOLD, CLOCK IN  control signals are connected directly on ESP32. 
CLOCK IN receives clock pulses from ESP32. RUN/HOLD is used to enable/disable the DVM. 
And STROBE line generates interrupts pulses in the ESP32. The other data from DVM is acquired by the PCF8574 I/O Expander. It uses I2C interface to communicate with the ESP32. 

The only digit strobe used is DS5 (MSD) to receive data, and the BUSY output is ignored. The STROBE output goes low five times per conversion cycle. Each STROBE pulse of the TLC7135 generates an interrupt in the ESP32 to acquire BCD data and status of Over-voltage, Under-voltage and Polarity lines. 

The first STROBE pulse occurs in the middle of DS5 when BCD data for the most significant digit (MSD) is available on outputs B1–B8. STROBE also pulses LOW during the following DS4 through DS1 signals, after which STROBE remains high until the next conversion cycle. Therefore, only one STROBE pulse occurs for each digit select, and each STROBE corresponds to a BCD digit in MSD-to-LSD order.  When reading the ADC data, the ESP32 simply reads BCD data during each STROBE pulse and stores that data in memory corresponding to the number of STROBE pulses received.

stickbreaker, 2019-07-01:
change your code such that the interrupt sets a flag that is triggered in the loop or a separate task. 

The default `Wire()` bus speed is 100k.
Recommended changes to your code:
* get rid of the `Wire.begin()` in the read code
* don't call `Wire()` in an interrupt, (`Wire()` is not re-enterent) 
* increase the bus speed to 400, (`Wire.begin(sda,scl,400000);`)

Chuck.

Gustavomurta, 2019-07-01:
Theoretically It is not possible to increase the I2C speed of PCF8574. 
http://www.ti.com/lit/ds/symlink/pcf8574.pdf

The max I2C frequecy of PCF8574 is 100 KHz. 
But I will try this - 400 KHz

stickbreaker, 2019-07-01:
```c++
Serial.begin(115200);
  attachInterrupt(digitalPinToInterrupt(STB), Sample, FALLING); // External Interrupt GPIO_4
  Wire.begin (SDApin, SCLpin,400000);           // sda= GPIO_21 /scl= GPIO_22
}

volatile bool getSample=false;
volatile int missedCount = 0;

void Sample(){
 if(getSample){
  missedCount++;
  }
  getSample = true;
}

void loop(){
if(getSample){
  getSample = false;
  int temp=missedCount;
  missedCount = 0;
  if(temp) {
   Serial.printf("  Missed %d sample pulse(s)\n",temp); 
  }
  testdata();
}
```

Gustavomurta, 2019-07-01:
Console message ( I2C clock 400 KHz) 
`Wire.requestFrom() failed=5(BUSY)`

The PCF8574 can not respond. (logic analyzer visualization)

stickbreaker, 2019-07-01:
try the code I just posted, add it to/modify your code, keep the bus speed at 100khz.   The busy is happening because of overlapping interrupts.  the 'second' interrupt `Wire.begin()`  is checking the busy flag (with is set because the prior transaction is still in progress), and failing.  When `Wire.begin()` is called it INITIALIZES the i2c hardware.  If the hardware is currently processing a prior resquest, IT KILLS it.  You cannot overlap `Wire()` calls.

stickbreaker, 2019-07-02:
you might be seeing multiple interrupts per/strobe,  If the falling edge is not fast (MHz) the esp32 can see multiple transitions because of glitches of the signal (micro volts).

Gustavomurta, 2019-07-02:
Ok , I am preparing this test. 

In my project there are no glitches. The frequency of interrupt is low. The signal is clear!
(seeing in my osclloscope)

Gustavomurta, 2019-07-02:
There is no answer at Serial Console> 
Seeing the pulses of Interrupt pin (GPIO-4) - five pulses of 1 us cycling at frequency of 12 Hz. 
Important - The GPIO-2 must generate the clock of 480 KHz to Voltmeter run. 

`#include <Wire.h>

# define CLOCK 2       // TCL7135 pin22 CLKIN     => GIPO_2 
# define STB  4        // TCL7135 pin26 -STB      => GPIO_4
# define RHLD 5        // TCL7135 pin25 RUN/-HOLD => GPIO_5
# define SDApin 21     // PCF8574 SDA pin15       => GPIO_21
# define SCLpin 22     // PCF8574 SCL pin14       => GPIO_22
# define PCF8574 0x20  // PCF8574 Address

byte Digit5, Digit4, Digit3, Digit2, Digit1 = 0;  // Digits of TLC7135
byte pcfData = 0;
int DigitCount = 0 ;
char Signal ;
float Voltage = 0;

volatile bool getSample = false;
volatile int missedCount = 0;

void setup()
{

  pinMode (STB, INPUT);                    // Interrupt - ESP32 GPIO_4
  pinMode (RHLD, OUTPUT);                  // RUN DVM   - ESP32 GIPO_5

  // TLC7135 pin22 CLKIN => GPIO_2 generate CLOCK 480 KHz

  pinMode(CLOCK, OUTPUT);       // GPIO_2 as Output
  ledcAttachPin(CLOCK, 0);      // GPIO_2 attached to PWM Channel 0
  ledcSetup(0, 480000, 2);      // Channel 0 , freq 480 KHz , 2 bit resolution
  ledcWrite(0, 2);              // Enable frequency with duty cycle 50%

  digitalWrite(RHLD, LOW);                // Reset DVM TLC7135
  delay(100);
  digitalWrite(RHLD, HIGH);               // Run DVM TLC7135
  
  Serial.begin(115200);
  attachInterrupt(digitalPinToInterrupt(STB), Sample, FALLING);  // External Interrupt GPIO_4
  Wire.begin (SDApin, SCLpin);                                   // sda= GPIO_21 /scl= GPIO_22
}

void Sample()
{
  if (getSample)
  {
    missedCount++;
  }
  getSample = true;
}

void loop()
{
  if (getSample)
  {
    getSample = false;
    int temp = missedCount;
    missedCount = 0;
    if (temp)
    {
      Serial.printf("  Missed %d sample pulse(s)\n", temp);
    }    
  }
}`

stickbreaker, 2019-07-02:
add call to your testData() code.  or
```c++
void loop()
{
  if (getSample)
  {
    getSample = false;
    int temp = missedCount;
    missedCount = 0;
    if (temp)
    {
      Serial.printf("  Missed %d sample pulse(s)\n", temp);
    }
  Serial.printF("Sampling\n");
  testdata();    
  }
}
```

chuck.

Gustavomurta, 2019-07-02:
```
[code]
#include <Wire.h>

# define CLOCK 2       // TCL7135 pin22 CLKIN     => GIPO_2 
# define STB  4        // TCL7135 pin26 -STB      => GPIO_4
# define RHLD 5        // TCL7135 pin25 RUN/-HOLD => GPIO_5
# define SDApin 21     // PCF8574 SDA pin15       => GPIO_21
# define SCLpin 22     // PCF8574 SCL pin14       => GPIO_22
# define PCF8574 0x20  // PCF8574 Address

byte Digit5, Digit4, Digit3, Digit2, Digit1 = 0;  // Digits of TLC7135
byte pcfData = 0;
int DigitCount = 0 ;
char Signal ;
float Voltage = 0;

volatile bool getSample = false;
volatile int missedCount = 0;

void setup()
{

  pinMode (STB, INPUT);                    // Interrupt - ESP32 GPIO_4
  pinMode (RHLD, OUTPUT);                  // RUN DVM   - ESP32 GIPO_5

  // TLC7135 pin22 CLKIN => GPIO_2 generate CLOCK 480 KHz

  pinMode(CLOCK, OUTPUT);       // GPIO_2 as Output
  ledcAttachPin(CLOCK, 0);      // GPIO_2 attached to PWM Channel 0
  ledcSetup(0, 480000, 2);      // Channel 0 , freq 480 KHz , 2 bit resolution
  ledcWrite(0, 2);              // Enable frequency with duty cycle 50%

  digitalWrite(RHLD, LOW);                // Reset DVM TLC7135
  delay(100);
  digitalWrite(RHLD, HIGH);               // Run DVM TLC7135

  Serial.begin(115200);
  attachInterrupt(digitalPinToInterrupt(STB), Sample, FALLING);  // External Interrupt GPIO_4
  Wire.begin (SDApin, SCLpin);                                   // sda= GPIO_21 /scl= GPIO_22
}

void Sample()
{
  if (getSample)
  {
    missedCount++;
  }
  getSample = true;

  Wire.requestFrom(PCF8574, 1);   // Read One Byte from PCF8574
  pcfData = Wire.read();          // Byte Read
}

void PrintData ()
{
  Serial.print(" TLC7135 Data = ");
  //Serial.print(Signal);              // Signal Plus or Minus
  Serial.print(" ");
  Serial.print(Digit5, HEX);
  Serial.print(",");
  Serial.print(Digit4, HEX);
  Serial.print(Digit3, HEX);
  Serial.print(Digit2, HEX);
  Serial.print(Digit1, HEX);
}

void PrintVoltage ()
{
  Voltage = ((Digit5 * 10000) + (Digit4 * 1000) + (Digit3 * 100) + (Digit2 * 10) + Digit1) / 10000.0 ;
  Serial.print("   Voltage = ");
  //Serial.print(Signal);
  Serial.println(Voltage, 3);
}

void TestData ()
{
  Sample();
  // Polarity();
  if ( bitRead(pcfData, 4))             // Test if TLC7135 Digit 5 is on (bit 4 of PCF8574)
  {
    Digit5 = pcfData & 0x0F ;           // Read first Byte
    DigitCount = 0 ;                    // Reset Digit Count
    DigitCount = ++DigitCount ;         // Digit Count increment
  }
  if (! (bitRead(pcfData, 4)))          // Test If bit 4 of PCF8574 is Off - Not Digit 5
    switch (DigitCount)
    {
      case 1 :
        Digit4 = pcfData & 0x0F ;       // Read 2nd Byte - Digit 4
        DigitCount = ++DigitCount ;     // Digit Count increment
        break ;
      case 2 :
        Digit3 = pcfData & 0x0F ;       // Read 3rd Byte - Digit 3
        DigitCount = ++DigitCount ;     // Digit Count increment
        break ;
      case 3 :
        Digit2 = pcfData & 0x0F ;       // Read 4th Byte - Digit 2
        DigitCount = ++DigitCount ;     // Digit Count increment
        break ;
      case 4 :
        Digit1 = pcfData & 0x0F ;       // Read 5th Byte - Digit 1
        DigitCount = ++DigitCount ;     // Digit Count increment
        PrintData ();
        PrintVoltage ();
        break ;
      default :
        break;
    }
  // OverVoltage ();
  // UnderVoltage ();
}

void loop()
{
  if (getSample)
  {
    getSample = false;
    int temp = missedCount;
    missedCount = 0;
    if (temp)
    {
      Serial.printf("  Missed %d sample pulse(s)\n", temp);
    }
    Serial.printf("Sampling\n");
    TestData ();
  }
}

[/code]
```

Gustavomurta, 2019-07-02:
Console Messages : 

```
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:8424
ho 0 tail 12 room 4
load:0x40080400,len:5868
entry 0x4008069c
Sampling
Sampling
Sampling
Sampling
Sampling
Sampling
Sampling
Sampling
Sampling
Sampling
Sampling
Sampling
 TLC7135 Data =  1,2770   Voltage = 1.277
Sampling
Sampling
Sampling
Sampling
Sampling
Sampling
Sampling
Sampling
Sampling
Sampling
Sampling
Sampling
 TLC7135 Data =  1,7720   Voltage = 1.772
Sampling
Sampling
Sampling
Sampling
Sampling
Sampling
Sampling
Sampling
Sampling
Sampling
Sampling
 TLC7135 Data =  1,7727   Voltage = 1.773
Sampling
Sampling
Sampling
Sampling
Sampling
Sampling
 TLC7135 Data =  1,7720   Voltage = 1.772
Sampling
Sampling
Sampling
Sampling
Sampling
Sampling
 TLC7135 Data =  1,7720   Voltage = 1.772
Sampling
Sampling
Sampling
Sampling
Sampling
Sampling
Sampling
Sampling
Sampling
Sampling
Sampling
 TLC7135 Data =  1,7727   Voltage = 1.773
Sampling
Sampling
Sampling
Sampling
Sampling
Sampling
 TLC7135 Data =  1,7720   Voltage = 1.772
Sampling
Sampling
Sampling
Sampling
Sampling
Sampling
 TLC7135 Data =  1,7620   Voltage = 1.762
Sampling
Sampling
Sampling
Sampling
Sampling
Sampling
Sampling
Sampling
Sampling
Sampling
  Missed 5 sample pulse(s)
Sampling
Sampling
Sampling
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x40088bfd  PS      : 0x00060533  A0      : 0x80088007  A1      : 0x3ffbcf60  
A2      : 0x3ffb8074  A3      : 0x3ffbd0ec  A4      : 0x00000001  A5      : 0x00000001  
A6      : 0x00060523  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3ffb8074  
A10     : 0x3ffb8074  A11     : 0x00000000  A12     : 0x00060520  A13     : 0x3ffbbcc8  
A14     : 0x0000000a  A15     : 0x3ffbc030  SAR     : 0x00000000  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000004  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x40088bfd:0x3ffbcf60 0x40088004:0x3ffbcf80 0x40085a1a:0x3ffbcfa0 0x40085a95:0x3ffbcfc0 0x4008869a:0x3ffbcfe0 0x4008877b:0x3ffbd020 0x40085aa1:0x3ffbd050

Rebooting...
ets Jun  8 2016 00:22:57

```

stickbreaker, 2019-07-02:
Not quite.

```c++
#include <Wire.h>

# define CLOCK 2       // TCL7135 pin22 CLKIN     => GIPO_2 
# define STB  4        // TCL7135 pin26 -STB      => GPIO_4
# define RHLD 5        // TCL7135 pin25 RUN/-HOLD => GPIO_5
# define SDApin 21     // PCF8574 SDA pin15       => GPIO_21
# define SCLpin 22     // PCF8574 SCL pin14       => GPIO_22
# define PCF8574 0x20  // PCF8574 Address

byte Digit5, Digit4, Digit3, Digit2, Digit1 = 0;  // Digits of TLC7135
byte pcfData = 0;
int DigitCount = 0 ;
char Signal ;
float Voltage = 0;

volatile bool getSample = false;
volatile int missedCount = 0;

void setup()
{

  pinMode (STB, INPUT);                    // Interrupt - ESP32 GPIO_4
  pinMode (RHLD, OUTPUT);                  // RUN DVM   - ESP32 GIPO_5

  // TLC7135 pin22 CLKIN => GPIO_2 generate CLOCK 480 KHz

  pinMode(CLOCK, OUTPUT);       // GPIO_2 as Output
  ledcAttachPin(CLOCK, 0);      // GPIO_2 attached to PWM Channel 0
  ledcSetup(0, 480000, 2);      // Channel 0 , freq 480 KHz , 2 bit resolution
  ledcWrite(0, 2);              // Enable frequency with duty cycle 50%

  digitalWrite(RHLD, LOW);                // Reset DVM TLC7135
  delay(100);
  digitalWrite(RHLD, HIGH);               // Run DVM TLC7135

  Serial.begin(115200);
  attachInterrupt(digitalPinToInterrupt(STB), Sample, FALLING);  // External Interrupt GPIO_4
  Wire.begin (SDApin, SCLpin);                                   // sda= GPIO_21 /scl= GPIO_22
}

void Sample()
{
  if (getSample)
  {
    missedCount++;
  }
  getSample = true;
}

void PrintData ()
{
  Serial.print(" TLC7135 Data = ");
  //Serial.print(Signal);              // Signal Plus or Minus
  Serial.print(" ");
  Serial.print(Digit5, HEX);
  Serial.print(",");
  Serial.print(Digit4, HEX);
  Serial.print(Digit3, HEX);
  Serial.print(Digit2, HEX);
  Serial.print(Digit1, HEX);
}

void PrintVoltage ()
{
  Voltage = ((Digit5 * 10000) + (Digit4 * 1000) + (Digit3 * 100) + (Digit2 * 10) + Digit1) / 10000.0 ;
  Serial.print("   Voltage = ");
  //Serial.print(Signal);
  Serial.println(Voltage, 3);
}

void TestData ()
{

  uint8_t count =Wire.requestFrom(PCF8574, 1);   // Read One Byte from PCF8574
  if(count == 1) {
     pcfData = Wire.read();          // Byte Read
  }
else {
   Serial.printf(" requestFrom() failed = %d (%s)\n",Wire.lastError(),Wire.getErrorText(Wire.lastError()));
   return; // error
}
  // Polarity();
  if ( bitRead(pcfData, 4))             // Test if TLC7135 Digit 5 is on (bit 4 of PCF8574)
  {
    Digit5 = pcfData & 0x0F ;           // Read first Byte
    DigitCount = 0 ;                    // Reset Digit Count
    DigitCount = ++DigitCount ;         // Digit Count increment
  }
  if (! (bitRead(pcfData, 4)))          // Test If bit 4 of PCF8574 is Off - Not Digit 5
    switch (DigitCount)
    {
      case 1 :
        Digit4 = pcfData & 0x0F ;       // Read 2nd Byte - Digit 4
        DigitCount = ++DigitCount ;     // Digit Count increment
        break ;
      case 2 :
        Digit3 = pcfData & 0x0F ;       // Read 3rd Byte - Digit 3
        DigitCount = ++DigitCount ;     // Digit Count increment
        break ;
      case 3 :
        Digit2 = pcfData & 0x0F ;       // Read 4th Byte - Digit 2
        DigitCount = ++DigitCount ;     // Digit Count increment
        break ;
      case 4 :
        Digit1 = pcfData & 0x0F ;       // Read 5th Byte - Digit 1
        DigitCount = ++DigitCount ;     // Digit Count increment
        PrintData ();
        PrintVoltage ();
        break ;
      default :
        break;
    }
  // OverVoltage ();
  // UnderVoltage ();
}

void loop()
{
  if (getSample)
  {
    getSample = false;
    int temp = missedCount;
    missedCount = 0;
    if (temp)
    {
      Serial.printf("  Missed %d sample pulse(s)\n", temp);
    }
    Serial.printf("Sampling\n");
    TestData ();
  }
}
```

Don't call `Wire()` from an interrupt context, Set a flag, then let the main loop do the work.  

Chuck.

Gustavomurta, 2019-07-02:
Now , the ESP32 can read some bytes of PCF8574 ! 
Some values are wrong. Maybe out of sync with the correct reading. 
What is your conclusion, Master ?

stickbreaker, 2019-07-02:
just posted a change to your code, try it.
chuck.

stickbreaker, 2019-07-02:
I would code such that when a missed pulse is detected, the current sample sequence drops into a resync mode.  It looks for the last digit or the first.  have a state machine that knows what it is expecting.  If it does not receive what it is expecting, then fall into to resync.

Since the "sampling" debug message is working, delete it,  the `loop()` code will hang until all print data can be added to the 127byte uart buffer, so only print what you need.  at 115200 baud you can send a max of 11,520 bytes per second before the `Serial.print()` calls start hanging.

Chuck.

Gustavomurta, 2019-07-02:
I supressed   // Serial.printf("Sampling\n");
It much better now ! 
```
rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:8424
ho 0 tail 12 room 4
load:0x40080400,len:5868
entry 0x4008069c
 TLC7135 Data =  1,2600   Voltage = 1.260
 TLC7135 Data =  1,2595   Voltage = 1.260
 TLC7135 Data =  1,2599   Voltage = 1.260
 TLC7135 Data =  1,2606   Voltage = 1.261
 TLC7135 Data =  1,2608   Voltage = 1.261
 TLC7135 Data =  1,2609   Voltage = 1.261
 TLC7135 Data =  1,2610   Voltage = 1.261
 TLC7135 Data =  1,2612   Voltage = 1.261
 TLC7135 Data =  1,2615   Voltage = 1.262
 TLC7135 Data =  1,2616   Voltage = 1.262
 TLC7135 Data =  1,2616   Voltage = 1.262
 TLC7135 Data =  1,2616   Voltage = 1.262
 TLC7135 Data =  1,2616   Voltage = 1.262
 TLC7135 Data =  1,2615   Voltage = 1.262
 TLC7135 Data =  1,2614   Voltage = 1.261
 TLC7135 Data =  1,2614   Voltage = 1.261
 TLC7135 Data =  1,2611   Voltage = 1.261
 TLC7135 Data =  1,2610   Voltage = 1.261
 TLC7135 Data =  1,2610   Voltage = 1.261
 TLC7135 Data =  1,2611   Voltage = 1.261
 TLC7135 Data =  1,2611   Voltage = 1.261
 TLC7135 Data =  1,2612   Voltage = 1.261
 TLC7135 Data =  1,2613   Voltage = 1.261
 TLC7135 Data =  1,2614   Voltage = 1.261
 TLC7135 Data =  1,2614   Voltage = 1.261
 TLC7135 Data =  1,2615   Voltage = 1.262
 TLC7135 Data =  1,2615   Voltage = 1.262
 requestFrom() failed = 2 (ACK)
 requestFrom() failed = 2 (ACK)
 TLC7135 Data =  1,2616   Voltage = 1.262
 TLC7135 Data =  1,2615   Voltage = 1.262
 TLC7135 Data =  1,2615   Voltage = 1.262
 requestFrom() failed = 2 (ACK)
 TLC7135 Data =  1,2615   Voltage = 1.262
 TLC7135 Data =  1,2614   Voltage = 1.261
 TLC7135 Data =  1,2615   Voltage = 1.262
 TLC7135 Data =  1,2615   Voltage = 1.262
 TLC7135 Data =  1,2615   Voltage = 1.262

```

Gustavomurta, 2019-07-02:
This problem occurs because the I2C routines was reformulated? 
In past my sketch runs perfectly... 

But now I can change my program ! 
Thank you very much, Chuck !

stickbreaker, 2019-07-02:
resync code:
```c++

static bool reSync=true; // start with a resync operation, don't know which digit
//  the meter is currently sending on boot.

void TestData ()
{

  uint8_t count =Wire.requestFrom(PCF8574, 1);   // Read One Byte from PCF8574
  if(count == 1) {
     pcfData = Wire.read();          // Byte Read
  }
else {
   Serial.printf(" requestFrom() failed = %d (%s)\n",Wire.lastError(),Wire.getErrorText(Wire.lastError()));
   return; // error
}
  // Polarity();
  if ( bitRead(pcfData, 4))             // Test if TLC7135 Digit 5 is on (bit 4 of PCF8574)
  {
   reSync = false; // Ok, now I know where I am.
    Digit5 = pcfData & 0x0F ;           // Read first Byte
    DigitCount = 0 ;                    // Reset Digit Count
  }
else if(!reSync) { // bit 4 is not set, so must be digits 1..4 and not in reSync mode
   DigitCount++; //now working on new digit
    switch (DigitCount)
    {
      case 1 :
        Digit4 = pcfData & 0x0F ;       // Read 2nd Byte - Digit 4
        break ;
      case 2 :
        Digit3 = pcfData & 0x0F ;       // Read 3rd Byte - Digit 3
        break ;
      case 3 :
        Digit2 = pcfData & 0x0F ;       // Read 4th Byte - Digit 2
        break ;
      case 4 :
        Digit1 = pcfData & 0x0F ;       // Read 5th Byte - Digit 1
        PrintData ();
        PrintVoltage ();
        reSync = true; // start a new sample, this one is done!
        break ;
      default :
        reSync = true; // something wrong
        break;
    }
  // OverVoltage ();
  // UnderVoltage ();
}
```

Try this.

Chuck.

stickbreaker, 2019-07-02:
>This problem occurs because the I2C routines was reformulated? In past my sketch runs perfectly... But now I can change my program ! Thank you very much, Chuck !
--

You ran into some timing issue.   Your original code was not designed to work with the esp32, I don't know why it worked, my i2c code has been in since 1.0.0.  I haven't done any major changes since 1.0.1.

If you are happy with the result, close this issue.

Chuck.

Gustavomurta, 2019-07-02:
I worked with an old  package of ESP32 Core, using this procedure to install:
https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/windows.md

I will close this issue.
Thanks a lot !

Gustavomurta, 2019-07-02:
I will change my program to run with the new I2C required needs. 
Thanks a lot !

closed 2019-07-02 by Gustavomurta