espressif/arduino-esp32 #2952: ESP-IDF Release v3.2.2

joseitor, 2019-07-02:
Espressif have just release a new ESP-IDF v3.2.2 which fix many problems. We must update it here too.

asetyde, 2019-07-02:
I think it's reason we are pre-realese , I hope it updated too ! This is good level of quality to use 

Alexandro Todeschini
----------------------------
Develop & Research
Domotics Division 
+393351881688 - Skype : Electric Labs srl
alexandro.todeschini@electriclabs.it
---------------------------------------------------

Le informazioni, i dati e le notizie contenute nella presente comunicazione e i relativi allegati sono di natura privata e come tali possono essere riservate e sono, comunque, destinate esclusivamente ai destinatari indicati in epigrafe. La diffusione, distribuzione e/o la copiatura del documento trasmesso da parte di qualsiasi soggetto diverso dal destinatario è proibita, sia ai sensi dell’art. 616 c.p., sia ai sensi del D.Lgs. n. 196/2003. Se avete ricevuto questo messaggio per errore, vi preghiamo di distruggerlo e di darcene immediata comunicazione anche inviando un messaggio di ritorno all’indirizzo e-mail del mittente. 
This e-mail (including attachments) is intended only for the recipient(s) named above. It may contain confidential or privileged information and should not be read, copied or otherwise used by any other person. If you are not the named recipient, please contact (e-mail dell’azienda) and delete the e-mail from your system. Rif. D.L. 196/2003.

atanisoft, 2019-08-02:
@me-no-dev I believe you have pulled in the latest IDF 3.2.x already?

closed 2019-10-15 by stale[bot]