espressif/arduino-esp32 #2953: Guru meditation when SPIFFS jquery.min.js

matbks, 2019-07-02:
Hi,

SPIFFS.h and IRremote.h are not working together.

When the ESP32 run the following line it crashes:
```
server.on("/jquery.min.js", HTTP_GET, [](AsyncWebServerRequest *request){
    request->send(SPIFFS, "/jquery.min.js", "text/javascript");
```
BUT, it only happens if i use the following line in my setup:
`irrecv.enableIRIn(); `
// Begin the receiving process. This will enable the timer interrupt which consumes a small amount of CPU every 50 µs.`
from 
``#include <IRremote.h>`

Looks like SPIFFS and IRremote can't work together. idk if it's something to do with Timer.
I saved the error and decoded it. Follows:

DECODED ERROR:
```
Decoding stack results
0x400dd11c: IRTimer() at C:\Users\mateu\Documents\Arduino\libraries\ESP32-IRremote\IRremote.cpp line 131
0x40082561: spi_flash_read at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spi_flash/flash_ops.c line 537
0x400e76d6: esp_partition_read at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spi_flash/partition.c line 242
0x400eb6dd: spiffs_api_read at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spiffs/spiffs_api.c line 36
0x400ef873: spiffs_phys_rd at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spiffs/spiffs/src/spiffs_cache.c line 161
0x400ecfe3: spiffs_obj_lu_find_entry_visitor at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spiffs/spiffs/src/spiffs_nucleus.c line 166
0x400eeb5e: spiffs_object_find_object_index_header_by_name at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spiffs/spiffs/src/spiffs_nucleus.c line 1694
0x400ec4d8: SPIFFS_stat at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spiffs/spiffs/src/spiffs_hydrogen.c line 765
0x400eb0bc: vfs_spiffs_stat at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spiffs/esp_spiffs.c line 536
0x400e4ba1: esp_vfs_stat at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/vfs/vfs.c line 492
0x400e6911: stat at ../../../.././newlib/libc/syscalls/sysstat.c line 12
0x400d4a22: VFSImpl::open(char const*, char const*) at C:\Users\mateu\Documents\Arduino\hardware\espressif\esp32\libraries\FS\src\vfs_api.cpp line 41
0x400d400a: SPIFFSImpl::exists(char const*) at C:\Users\mateu\Documents\Arduino\hardware\espressif\esp32\libraries\SPIFFS\src\SPIFFS.cpp line 42
0x4016e0c6: fs::FS::exists(char const*) at C:\Users\mateu\Documents\Arduino\hardware\espressif\esp32\libraries\FS\src\FS.cpp line 199
0x400d413b: fs::FS::exists(String const&) at C:\Users\mateu\Documents\Arduino\hardware\espressif\esp32\libraries\FS\src\FS.cpp line 204
0x400d6965: AsyncWebServerRequest::send(fs::FS&, String const&, String const&, bool, std::function ) (C:\Program Files at x86)\Arduino\libraries\ESPAsyncWebServer-master\src\WebRequest.cpp line 771
0x400d2149: std::_Function_handler   >::_M_invoke(const std::_Any_data &,  ) at C:\Users\mateu\Documents\Arduino\WifiLoginIR/WifiLoginIR.ino line 181
0x400d8f75: AsyncCallbackWebHandler::handleRequest(AsyncWebServerRequest*) at c:\users\mateu\documents\arduino\hardware\espressif\esp32\tools\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0/functional line 2271
0x400d6bf1: AsyncWebServerRequest::_parseLine() (C:\Program Files at x86)\Arduino\libraries\ESPAsyncWebServer-master\src\WebRequest.cpp line 576
0x400d6cb5: AsyncWebServerRequest::_onData(void*, unsigned int) (C:\Program Files at x86)\Arduino\libraries\ESPAsyncWebServer-master\src\WebRequest.cpp line 121
0x400d6f1d: std::_Function_handler   >::_M_invoke(const std::_Any_data &,  ,  ,  ,  ) (C:\Program Files at x86)\Arduino\libraries\ESPAsyncWebServer-master\src\WebRequest.cpp line 75
0x400d9d4d: AsyncClient::_recv(tcp_pcb*, pbuf*, signed char) at c:\users\mateu\documents\arduino\hardware\espressif\esp32\tools\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0/functional line 2271
0x400d9dc9: AsyncClient::_s_recv(void*, tcp_pcb*, pbuf*, signed char) (C:\Program Files at x86)\Arduino\libraries\AsyncTCP-master\src\AsyncTCP.cpp line 1144
0x400da356: _async_service_task(void*) (C:\Program Files at x86)\Arduino\libraries\AsyncTCP-master\src\AsyncTCP.cpp line 145
0x40088685: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143
```
ERROR:
```
05:35:37.214 -> PC      : 0x400dd11c  PS      : 0x00060034  A0      : 0x40081800  A1      : 0x3ffbe7a0  
05:35:37.214 -> A2      : 0x00000001  A3      : 0x00000002  A4      : 0x000000ff  A5      : 0x4008b3fc  
05:35:37.214 -> A6      : 0x00000000  A7      : 0x00000001  A8      : 0x8008133c  A9      : 0x3ff5f024  
05:35:37.214 -> A10     : 0x3ffbec3c  A11     : 0x20000000  A12     : 0x00000400  A13     : 0x3ffce6b0  
05:35:37.248 -> A14     : 0x00000026  A15     : 0x3ffc2b68  SAR     : 0x00000017  EXCCAUSE: 0x00000007  
05:35:37.248 -> EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffc  
05:35:37.248 -> Core 1 was running in ISR context:
05:35:37.248 -> EPC1    : 0x40082564  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400dd11c
05:35:37.248 -> 
05:35:37.248 -> Backtrace: 0x400dd11c:0x3ffbe7a0 0x400817fd:0x3ffbe7c0 0x40082561:0x3ffce6f0 0x400e76d6:0x3ffce760 0x400eb6dd:0x3ffce790 0x400ef873:0x3ffce7c0 0x400ecfe3:0x3ffce7f0 0x400eeb5e:0x3ffce830 0x400ec4d8:0x3ffce870 0x400eb0bc:0x3ffce8a0 0x400e4ba1:0x3ffce900 0x4000bcc5:0x3ffce920 0x400e6911:0x3ffce940 0x400d4a22:0x3ffce960 0x400d400a:0x3ffce9e0 0x4016e0c6:0x3ffcea20 0x400d413b:0x3ffcea40 0x400d6965:0x3ffcea60 0x400d2149:0x3ffceab0 0x400d8f75:0x3ffceb00 0x400d6bf1:0x3ffceb40 0x400d6cb5:0x3ffceb80 0x400d6f1d:0x3ffcebc0 0x400d9d4d:0x3ffcebe0 0x400d9dc9:0x3ffcec20 0x400da356:0x3ffcec40 0x40088685:0x3ffcec70
```

I am using:
ESP32 DEV KIT
WINDOWS 10 64b

libraries:
#include <WiFi.h>
#include "SPIFFS.h"
#include "ESPAsyncWebServer.h"
#include <ArduinoOTA.h>
#include <IOXhop_FirebaseESP32.h>
#include <IRremote.h>

context:
Its an ESP32 webserver offline.
When the user conects to the AcessPoint it can see a page through browser, this page contains a JavaScript code that allows me to get some information from the user throw text boxes.
To use JavaScript in this code without WiFi conection i need to bring it with the ESP32 throw SPIFFS.
This project also use a Infrared LED to comunicate with other devices.
So i can't afford not using Infrared libraries or SPIFFS library.

stickbreaker, 2019-07-02:
Which Guru mediation error?  If it is "cache region accessed but cache disabled" look at the code of `IRTimer()` see if it is small enough to fit in IRAM and is not marked IRAM.

Chuck.

matbks, 2019-07-02:
Yes thats the errror. I've tried just writing IRAM_ATTR bettwen voind and onTimer() {} but didnt work, idk if thats just as simple as that, if it is, didnt work. 
Thank you.

atanisoft, 2019-07-02:
@matbks with a 50 microsecond interval on the timer ISR and auto-reload you won't be able to access SPIFFS (or flash for that matter) anytime the IR code is "active". You have two options:
1. Move the web content to static blobs compiled in the code and serve them from there.
2. disable the IR code *BEFORE* calling ```request->send(SPIFFS, "/jquery.min.js", "text/javascript");``` and re-enable it *AFTER* that request is complete.

Also note that though the comment claims a 50 nanosecond interval it is actually using a 50 microsecond interval based on the divider (default APB clock used by the HW timers is 80Mhz).

matbks, 2019-07-02:
@atanisoft  I appreciate your help.
Could you please tell me how to disable IR, cause as far as I know there is only this enable function.
Thanks again.

atanisoft, 2019-07-02:
As I don't use that library or know anything about it I'm not sure how to help on it.

This is also not the first time someone has had a problem with this particular library and SPIFFS access.

MorpheusDe97, 2019-07-03:
Hello @matbks, this worked for me: 
I implemented a disableIRIn-Function that calls the reverse functions of the enableIRIn()-Function. Maybe not the most elegant way, but at least it worked on my side.
`void IRrecv::disableIRIn()
{
	// Initialize state machine variables
	irparams.rcvstate = STATE_IDLE;
	irparams.rawlen = 0;
	timerEnd(timer);
	timerAlarmDisable(timer);
	timerDetachInterrupt(timer);
}`

matbks, 2019-07-04:
@MorpheusDe97 Thank you so much!
I am going to try this.
I ended up using https://github.com/markszabo/IRremoteESP8266 that is now working on esp32 and i haven't look up the library yet but i does not cause the crash and actually recognizes lots of  protocols.
Anyway, i am going to try  your code and post here if i was able to enable and disable irrecv with sucess.
Thank you again!

closed 2019-09-16 by stale[bot]