espressif/arduino-esp32 #2954: Wrong Magic Byte error when trying to upload spiffs.bin

GeorgeFlorian, 2019-07-02:
OS: Linux Mint 19.1
Board: ESP32-EVB
IDE: Platformio IDE on VSCode
Hello !

I am trying to implement Firmware Update via file upload in my project.
I have followed @lbernstone 's [example](https://github.com/lbernstone/asyncUpdate/blob/master/AsyncUpdate.ino) and I made it to work when uploading a `firmware.bin` file. But when I'm trying to upload `spiffs.bin` it returns A LOT of `Wrong Magic Byte` lines.

I get `spiffs.bin` after I upload files into the ESP using `pio run -t uploadfs`.

Any ideas ?

lbernstone, 2019-07-02:
Make sure it works manually.
`python esptool.py write_flash 0x290000 spiffs.bin`

GeorgeFlorian, 2019-07-02:
> Make sure it works manually.
> `python esptool.py write_flash 0x290000 spiffs.bin`
> That indexOf compare probably ought to be >=0. I didn't account for the filename starting with spiffs.

What do I do with `python esptool.py write_flash 0x290000 spiffs.bin` ?

lbernstone, 2019-07-02:
Run it from the command line.  The esptool will upload your file over usb to the esp32.

GeorgeFlorian, 2019-07-02:
> Run it from the command line. The esptool will upload your file over usb to the esp32.

Too bad I've never installed esptool. It doesn't mater though.
You were right with `>=`. I've changed to:
```
int cmd = (filename.indexOf("spiffs") >= 0) ? U_SPIFFS : U_FLASH;
```
and now it works. I can upload both `firmware.bin' and `spiffs.bin`.

Also, I've ditched the MDNS. I couldn't find its purpose.
I am using both handlers: `void handleUpdate` and `void handleDoUpdate` and I am using the `server.on` handlers to server them plain, alongside the others `server.on` that I have in the project:
```
void setup() {
// ... a lot of server.on
// ...
      server.on("/update", HTTP_GET, [](AsyncWebServerRequest *request){handleUpdate(request);});
      server.on("/doUpdate", HTTP_POST,
      [](AsyncWebServerRequest *request) {},
      [](AsyncWebServerRequest *request, const String& filename, size_t index, uint8_t *data,
                    size_t len, bool final) {handleDoUpdate(request, filename, index, data, len, final);}
      );
      server.onNotFound([](AsyncWebServerRequest *request){request->send(404);});

      server.begin();
}
```

GeorgeFlorian, 2019-07-02:
> Run it from the command line. The esptool will upload your file over usb to the esp32.

Too bad I've never installed esptool. It doesn't mater though.
You were right with `>=`. I've changed to:
```
int cmd = (filename.indexOf("spiffs") >= 0) ? U_SPIFFS : U_FLASH;
```
and now it works. I can upload both `firmware.bin` and `spiffs.bin`.

Also, I've ditched the MDNS. I couldn't find its purpose.
I am using both handlers: `void handleUpdate` and `void handleDoUpdate` and I am using the `server.on` handlers to server them plain, alongside the others `server.on` that I have in the project:
```
void setup() {
// ... a lot of server.on
// ...
      server.on("/update", HTTP_GET, [](AsyncWebServerRequest *request){handleUpdate(request);});
      server.on("/doUpdate", HTTP_POST,
      [](AsyncWebServerRequest *request) {},
      [](AsyncWebServerRequest *request, const String& filename, size_t index, uint8_t *data,
                    size_t len, bool final) {handleDoUpdate(request, filename, index, data, len, final);}
      );
      server.onNotFound([](AsyncWebServerRequest *request){request->send(404);});

      server.begin();
}
```

It should work even without MDNS, right ?

GeorgeFlorian, 2019-07-02:
Also, uploading `spiffs.bin` to the ESP32 means that it will overwrite whatever files were before and just put the new ones on top, correct ?

If that is true, is there a way of only updating the files inside SPIFFS and not overwrite them ?

lbernstone, 2019-07-02:
mDNS has nothing to do with the update.
Updating a partition replaces the partition.  If you want to manage the files, you need to manage the files. https://github.com/me-no-dev/ESPAsyncWebServer/tree/master/examples/ESP_AsyncFSBrowser.

closed 2019-09-14 by stale[bot]