espressif/arduino-esp32 #2956: Stuck at mbedtls_err_ssl_want_write()

thedim-witted, 2019-07-03:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do?
Describe your system( Hardware, computer, O/S, core version, environment).
Describe what is failing.
Show the shortest possible code that will duplicate the error.
Show the EXACT error message(it doesn't work is not enough).
All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ESP32 Dev Module? ?node32? ?ttgo_lora?
Core Installation version:			?1.0.0? ?1.0.1-rc4? ?1.0.1? ?1.0.1-git?
IDE name:						?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
PSRAM enabled:                                    ?no? ?yes?
Upload Speed:					?115200?
Computer OS:                                        ?Windows 10? ?Mac OSX? ?Ubuntu?

### Description:
Describe your problem here

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

lbernstone, 2019-07-03:
CSB. If you actually want some help, make a minimal sketch.  If you are working from the [esp-idf example](https://github.com/espressif/esp-idf/blob/v3.2.2/examples/protocols/https_mbedtls/main/https_mbedtls_example_main.c), you may get more traction asking your question there.

thedim-witted, 2019-07-08:
Reproduced this from the WiFiClientSecure example code

`#include <ArduinoJson.h>
#include <WiFiClientSecure.h>

const char* ssid     = "xxxxxxxxx";     // your network SSID (name of wifi network)
const char* password = "xxxxxxxxxx"; // your network password

const char*  host = "xxxxxxxxx";  // Server URL
const char* url   = "/xxxxxxxxx";
const char* fingerprint = "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx";
int connection_iterator;
int SAMPLE_SIZE = 350;
uint64_t chipid;
uint32_t low;
uint32_t high;

WiFiClientSecure client;

void setup() {
  //Initialize serial and wait for port to open:
  Serial.begin(115200);
  delay(100);
  chipid = ESP.getEfuseMac();
  low = chipid % 0xFFFFFFFF;
  high = (chipid >> 32) % 0xFFFFFFFF;
  Serial.println(String(low, HEX) + String(high, HEX));
  Serial.println("2002");

  pinMode(2, OUTPUT);
  digitalWrite(2, LOW);
}

DynamicJsonBuffer jsonBuffer(14700);

void loop() {
  connection_iterator = 0;

  WiFi.begin(ssid, password);

  //   attempt to connect to Wifi network:
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    // wait 1 second for re-trying
    delay(1000);
  }

  Serial.print("Connected to ");
  Serial.println(ssid);

  jsonBuffer.clear();
  JsonObject &rootObject = jsonBuffer.createObject();
  JsonObject &dataObject = rootObject.createNestedObject("data");
  JsonArray &X1dataObject = dataObject.createNestedArray("X1");
  JsonArray &Y1dataObject = dataObject.createNestedArray("Y1");
  JsonArray &Z1dataObject = dataObject.createNestedArray("Z1");
  Serial.print("Heap:  ");
  Serial.println((int)esp_get_free_heap_size());

  for (int iterator = 0; iterator < SAMPLE_SIZE; iterator++) {
    X1dataObject.add(iterator*0.1);
    Y1dataObject.add(iterator*0.2);
    Z1dataObject.add(iterator*0.3);
  }

  rootObject["coreid"] = String(low, HEX) + String(high, HEX);
  //rootObject["coreid_high"] = String(high, HEX);
  rootObject["sample_time"] = 554;
  rootObject["firmware_version"] = 2002;
  rootObject.prettyPrintTo(Serial);

  Serial.print("connecting to ");
  Serial.println(host);

  while ((!client.connect(host, 443))) {
    connection_iterator++;
    Serial.println("connection failed! Retrying..");
    delay(5000);

    if (connection_iterator >= 10)
      ESP.restart();
  }
    digitalWrite(2, HIGH);
  Serial.print("requesting URL: ");
  Serial.println(url);
  client.println(String("POST ") + url + " HTTP/1.1");
  client.println(String("Host: ") + host);
  client.println("Cache-Control: no-cache");
  client.println("Content-Type: application/json");
  client.print("Content-Length: ");
  client.println(rootObject.measureLength());
  client.println();
  rootObject.printTo(client);
  Serial.println("request sent");
  //  free(NULL);
  connection_iterator = 0;
  while ((client.connected())) {
    String line = client.readStringUntil('\n');
    connection_iterator++;
    if (line == "\r") {
      Serial.println("headers received");
      break;
    }
    Serial.println("Waiting for Headers");
    delay(500);
    if (connection_iterator > 30)
      break;
  }
  String line = client.readStringUntil('\n');
  if (line.startsWith("{")) {
    Serial.println("esp32/Arduino CI successfull!");
//    successfull_attempts++;
  } else {
    Serial.println("esp32/Arduino CI has failed");
  }
  Serial.println("reply was:");
  Serial.println("==========");
  Serial.println(line);
  Serial.println("==========");
  Serial.println("closing connection");
  digitalWrite(2, LOW);
}`

lbernstone, 2019-07-08:
Scoping matters.  Calling WiFi.begin every loop is not going to help your cause.
```
#include <ArduinoJson.h>
#include <WiFiClientSecure.h>

//const char* ssid = "xxxxxxxxx"; // your network SSID (name of wifi network)
//const char* password = "xxxxxxxxxx"; // your network password

const char* host = "www.google.com"; // Server URL
const char* url = "/";
int connection_iterator;
int SAMPLE_SIZE = 350;
uint32_t low;
uint32_t high;

WiFiClientSecure client;

void setup() {
//Initialize serial and wait for port to open:
  Serial.begin(115200);
  delay(100);
  uint64_t chipid = ESP.getEfuseMac();
  low = chipid % 0xFFFFFFFF;
  high = (chipid >> 32) % 0xFFFFFFFF;
  Serial.println(String(low, HEX) + String(high, HEX));
  Serial.println("2002");
  
  pinMode(2, OUTPUT);
  digitalWrite(2, LOW);

  WiFi.begin();
  if (!WiFi.waitForConnectResult()) {
    Serial.println("Connection failed");
    return;  
  }
  Serial.print("Connected to ");
  Serial.println(WiFi.SSID());  
}

void jsonOut() {
  DynamicJsonBuffer jsonBuffer(14700);
  connection_iterator = 0;
  
  jsonBuffer.clear();
  JsonObject &rootObject = jsonBuffer.createObject();
  JsonObject &dataObject = rootObject.createNestedObject("data");
  JsonArray &X1dataObject = dataObject.createNestedArray("X1");
  JsonArray &Y1dataObject = dataObject.createNestedArray("Y1");
  JsonArray &Z1dataObject = dataObject.createNestedArray("Z1");
  Serial.printf("Heap: %d\n", esp_get_free_heap_size());
  
  for (int iterator = 0; iterator < SAMPLE_SIZE; iterator++) {
    X1dataObject.add(iterator*0.1);
    Y1dataObject.add(iterator*0.2);
    Z1dataObject.add(iterator*0.3);
  }
  
  rootObject["coreid"] = String(low, HEX) + String(high, HEX);
  //rootObject["coreid_high"] = String(high, HEX);
  rootObject["sample_time"] = 554;
  rootObject["firmware_version"] = 2002;
  rootObject.prettyPrintTo(Serial);
  
  client.println(rootObject.measureLength());
  client.println();
  rootObject.printTo(client);
}

void loop() {
  Serial.print("connecting to ");
  Serial.println(host);
  
  while ((!client.connect(host, 443))) {
    connection_iterator++;
    Serial.println("connection failed! Retrying..");
    delay(5000);
    
    if (connection_iterator >= 10)
      ESP.restart();
    }
    digitalWrite(2, HIGH);
    Serial.print("requesting URL: ");
    Serial.println(url);
    client.println(String("POST ") + url + " HTTP/1.1");
    client.println(String("Host: ") + host);
    client.println("Cache-Control: no-cache");
    client.println("Content-Type: application/json");
    client.print("Content-Length: ");
    jsonOut();
    Serial.println("request sent");
    connection_iterator = 0;
    while (client.connected()) {
      String line = client.readStringUntil('\n');
      connection_iterator++;
      if (line == "\r") {
        Serial.println("headers received");
        break;
      }
      Serial.println("Waiting for Headers");
      delay(500);
      if (connection_iterator > 30) break;
    }
    String line = client.readStringUntil('\n');
    if (line.startsWith("{")) {
      Serial.println("esp32/Arduino CI successfull!");
      // successfull_attempts++;
    } else {
      Serial.println("esp32/Arduino CI has failed");
    }
  Serial.println("reply was:");
  Serial.println("==========");
  Serial.println(line);
  Serial.println("==========");
  Serial.println("closing connection");
  digitalWrite(2, LOW);
}
```

thedim-witted, 2019-07-08:
It's great that you restructured my test code. Thank You. However, this doesn't solve the issue. Also, I am getting this particular more often than getting stuck in the mbedtls_err_ssl_want_write() function

>21:11:13.151 -> [V][ssl_client.cpp:280] send_ssl_data(): ret < 0
>21:11:13.151 -> [E][ssl_client.cpp:33] handle_error(): UNKNOWN ERROR CODE (0050)
>21:11:13.151 -> [E][ssl_client.cpp:35] handle_error(): MbedTLS message code: -80
>21:11:13.184 -> [V][ssl_client.cpp:243] stop_ssl_socket(): Cleaning SSL connection.

closed 2019-09-20 by stale[bot]