espressif/arduino-esp32 #2957: How to measure battery lipo/lifepo4 with internal ADC

mixaobo, 2019-07-03:
i know there were a lots of issus about internal ADC accuaray. i use
`adc1_config_width(ADC_WIDTH_11Bit);
adc1_config_channel_atten(ADC1_CHANNEL_0, ADC_ATTEN_DB_11);`
My circuit is usb cable to power up esp32. GPIO36 to +3.32v (lifepo4), ground to ground
if i understand correctly, adc value will return 2047 if 3.9v applied to GPIO36.
but when i applied 3.32v to gpio it's always 2047.
is the scale limit by the voltage power up esp32 board (3.3v?) so i will receive 2047 always?
Code for reading 
`for (int i = 0; i < 500; i++)
    {
        sum += adc1_get_voltage(ADC1_CHANNEL_0);
        delayMicroseconds(1000);
    }
    // calculate the voltage
    voltage = sum / (float)500;`
the same if i use lipo 18650 3.62v. norminal/4.2v max
but when i you voltage divider 4.2v-->27k-->measure here-->10k-->ground. i receive 3.66v in monitor, it's close to 3.62v. so i always need voltage divider

lbernstone, 2019-07-03:
You already have the answer.  Use a voltage divider. Always.  It not only allows you to control the range, but limits the current draw so you don't discharge the battery through your ADC.  Please note, this forum is not a tutorial site, it is for issues with the arduino-esp32 code.

closed 2019-07-04 by mixaobo

mixaobo, 2019-07-04:
> You already have the answer. Use a voltage divider. Always. It not only allows you to control the range, but limits the current draw so you don't discharge the battery through your ADC. Please note, this forum is not a tutorial site, it is for issues with the arduino-esp32 code.

I run voltage divider like use. i 4.2v -->100k-->measure here-->27k-->ground and use internal 1.1v reference, but  i dont know why i only see ~3v at monitor
`adc1_config_width(ADC_WIDTH_BIT_11);
adc1_config_channel_atten(ADC1_CHANNEL_0,ADC_ATTEN_DB_0);
float battery_read()
{
    //read battery voltage per %
    long sum = 0;                  // sum of samples taken
    float voltage = 0.0;           // calculated voltage
    float output = 0.0;            //output value
    const float battery_max = 4.2; //maximum voltage of battery
    const float battery_min = 3.0; //minimum voltage of battery before shutdown

    float R1 = 100000.0; // resistance of R1 (100K)
    float R2 = 27000.0;  // resistance of R2 (10K)`
[Link to my circuit](https://www.upsieutoc.com/image/z1441710576761-a9c7c4541ccba6eef0366b2f8d6416fb.BtpaiE)