espressif/arduino-esp32 #2958: SD Card power usage in deep sleep (or when idle)

ducalex, 2019-07-03:
### Hardware:
Board:							Lolin32
Core Installation version:			ESP-IDF: v3.3-beta3-103-ga9d266b92, Arduino as a component: 7d7824701fe5e22f08555d3e1ce3180a922b2151
IDE name:						VS Code
Flash Frequency:					40Mhz
PSRAM enabled:                                     no
Upload Speed:				      15200
Computer OS:                                        Windows 8.1

### Description:
When I use the SD library and then go into deep sleep the SD Card uses an absurd amount of power (10-12mA). This does not occur if I use esp_vfs_fat_sdmmc_mount instead of the SD library even if I do not uninitialize it before sleeping.

I've investigated and the IO pins are freed correctlty (although card->ssPin isn't done explicitely). So the SD Card must be left in a weird state because of a wrong or incomplete command.

I fixed my problem by adding:
        `sdTransaction(pdrv, GO_IDLE_STATE, 0, NULL);`

to `sdcard_uninit`

But the real problem is probably elsewhere.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp
#include <Arduino.h>
#include <SD.h>
#include <esp_sleep.h>

void setup() {
    SD.begin();
    SD.end();
    esp_sleep_enable_timer_wakeup(10 * 1000 * 1000);
    esp_deep_sleep_start();
}

void loop() {
}
```

### Debug Messages:
```
None
```

closed 2019-09-15 by stale[bot]

NikolaiRadke, 2020-08-19:
Dear ducalex,

i encountered the same problems, the sd card draws too much current. I am looking for a solution for a long time, so i wonder if your fix works. Were did you add the line ` sdTransaction(pdrv, GO_IDLE_STATE, 0, NULL);`  in` sdcard_uninit` in file `sd_diskio.cpp`? 
  
I couldn't apply your fix, the card is still draining my poor battery. Maybe you are reading this after a year...

Best wishes from Berlin!

lbernstone, 2020-08-19:
Put it at line 651 before the ff_diskio_register call.  Let me know if this works and I will submit a PR.

NikolaiRadke, 2020-08-19:
Yes!!
It works flawless! 12mA less current, now the ESP32 deep sleep isn't senseless anymore. 

Thanks for your reply and best regarts from Berlin.

tueddy, 2020-12-19:
I wonder if this is valid/fixed for SD_MMC also? So if you call SD_MMC.end() the SD card goes to idle state and save power?
Thanks!