espressif/arduino-esp32 #2962: Esp32 BLE+ETH+OTA

robcim, 2019-07-04:
Hi all,
I'm Roberto. I use a ESP32 ( flash size 16MB) with BLE and ETH communication. I add the OTA mode (WebUpdate) but he's not work. 
I have this error in console: [E][WiFiClient.cpp:338] write(): 11

Anyone can help me please.
Thank you

my code is:

#include <WiFi.h>
#include <WiFiClient.h>
#include <WebServer.h>
#include <ESPmDNS.h>
#include <Update.h>
#include <ETH.h>

#include <HardwareSerial.h>
HardwareSerial Serial485(2);

int porta_stato_check = 0;
static bool eth_connected = false;
bool deviceConnected = false;
bool oldDeviceConnected = false;

#define ETH_CLK_MODE    ETH_CLOCK_GPIO0_IN   // nostre board ETH_CLOCK_GPIO0_IN      //  ETH_CLOCK_GPIO17_OUT
#define ETH_POWER_PIN   17                     // nostre board 17
#define ETH_TYPE        ETH_PHY_LAN8720  
#define ETH_ADDR        1                      // mostre board 1
#define ETH_MDC_PIN     23
#define ETH_MDIO_PIN    18

String MAIN_page;
String MAIN_page1;
String MAIN_page2;
String MAIN_page3;

WebServer server(80);

static const char serverIndex[] PROGMEM = R"(
<!DOCTYPE html>
<html>
 <head><meta charset="utf-8"><meta name="viewport" content="width=device-width,initial-scale=1,user-scalable=no">
      <title>Upgrade</title>
      <style>div,fieldset,input,select{padding:10px;font-size:1em;}
        fieldset{background-color:yellow;}
        p{margin:0.5em 0;button:hover{background-color:#32CD32;}.bred{background-color:green;}}
        input{width:100%;box-sizing:border-box;-webkit-box-sizing:border-box;-moz-box-sizing:border-box;button:hover{background-color:#32CD32;}.bred{background-color:green;}}
        input[type=checkbox],input[type=radio]{width:1em;margin-right:6px;vertical-align:-1px;}
        select{width:100%;}
        textarea{resize:none;width:98%;height:318px;padding:5px;overflow:auto;}
        body{text-align:center;font-family:verdana;font-size:1.6rem;}
        td{padding:0px;}
        button{border:0;border-radius:0.3rem;background-color:#00FF00;color:red;line-height:2.4rem;font-size:1.2rem;width:100%;-webkit-transition-duration:0.4s;transition-duration:0.4s;cursor:pointer;}     
      </style>
  <body>    
  <div style="text-align:left;display:newline-block;min-width:300px;">
    <div style="text-align:center;">
      <h2>Upgrade Firmware</h2></div>
    <form method='POST' action='/update' enctype='multipart/form-data'>
      <p></p>
      <p></p>
      <input type='file' name='update'>
      <p></p>
      <p></p>
      <input type='submit' value='Update'>
    </form>
      <p></p>
      <br>
      <p></p>
      <br>  
[... 331 lines cut ...]
//      Serial.print("ip_add= "); Serial.println(ip);
//      Serial.print("mask= "); Serial.println(msk);
//      Serial.print("gateway= "); Serial.println(gw);
//      Serial.print("dns= "); Serial.println(dn);
//    }
//  });
  server.begin();
  MDNS.addService("http", "tcp", 80);
}

void setup(void) {
  Serial.begin(115200);
  Serial.println();
  Serial.println("Booting Sketch...");

  ETH.begin(ETH_ADDR, ETH_POWER_PIN, ETH_MDC_PIN, ETH_MDIO_PIN, ETH_TYPE, ETH_CLK_MODE);
  
  OTA();
  
}

void loop(void) {
  server.handleClient();
  delay(1);
}

lbernstone, 2019-07-06:
tl;dr
Please [format your code](https://help.github.com/articles/creating-and-highlighting-code-blocks/), cut out the huge chunk of comment, and follow the [issue template](https://github.com/espressif/arduino-esp32/blob/master/.github/ISSUE_TEMPLATE/bug_report.md).  Provide as much logging as possible- a single error line with no context is not useful.

robcim, 2019-07-08:
#include <WiFi.h>
#include <WiFiClient.h>
#include <WebServer.h>
#include <ESPmDNS.h>
#include <Update.h>
#include <ETH.h>

#include <HardwareSerial.h>
HardwareSerial Serial485(2);

int porta_stato_check = 0;
static bool eth_connected = false;
bool deviceConnected = false;
bool oldDeviceConnected = false;

#define ETH_CLK_MODE ETH_CLOCK_GPIO0_IN
#define ETH_POWER_PIN 17
#define ETH_TYPE ETH_PHY_LAN8720
#define ETH_ADDR 1
#define ETH_MDC_PIN 23
#define ETH_MDIO_PIN 18

WebServer server(80);

static const char serverIndex[] PROGMEM = R"(

<title>Upgrade</title> 
<style>div,fieldset,input,select{padding:10px;font-size:1em;} fieldset{background-color:yellow;} 
    p{margin:0.5em 0;button:hover{background-color:#32CD32;}.bred{background-color:green;}} 
    input{width:100%;box-sizing:border-box;-webkit-box-sizing:border-box;-moz-box-sizing:border- 
    box;button:hover{background-color:#32CD32;}.bred{background-color:green;}} 
    input[type=checkbox],input[type=radio]{width:1em;margin-right:6px;vertical-align:-1px;} 
    select{width:100%;} textarea{resize:none;width:98%;height:318px;padding:5px;overflow:auto;} 
    body{text-align:center;font-family:verdana;font-size:1.6rem;} td{padding:0px;} button{border:0;border- 
    radius:0.3rem;background-color:#00FF00;color:red;line-height:2.4rem;font-size:1.2rem;width:100%;- 
    webkit-transition-duration:0.4s;transition-duration:0.4s;cursor:pointer;} 
</style>
  <body>    
  <div style="text-align:left;display:newline-block;min-width:300px;">
    <div style="text-align:center;">
      <h2>Upgrade Firmware</h2></div>
    <form method='POST' action='/update' enctype='multipart/form-data'>
      <input type='file' name='update'>
      <input type='submit' value='Update'>
    </form>
    <form action="." method="get">
      <button style=" background-color: #00FF00;color: red;">Main Menu</button>
    </form><p></p>
  </body>
          <div style="text-align:right;font-size:15px;"><hr>
          <a href="" target="_blank" style="color:#000000;">Smart Device</a>
          </div></div>
  </head>
</html>
)";

void WiFiEvent(WiFiEvent_t event)
{
  switch (event) {
    case SYSTEM_EVENT_ETH_START:
      Serial.println("ETH Started");
      //set eth hostname here
      ETH.setHostname("esp32-ethernet");
      break;
    case SYSTEM_EVENT_ETH_CONNECTED:
      Serial.println("ETH Connected");
      break;
    case SYSTEM_EVENT_ETH_GOT_IP:
      Serial.print("ETH MAC: ");
      Serial.print(ETH.macAddress());
[... 157 lines cut ...]
  if (eth_connected) {
    testClient("google.com", 80);
  }
  if (deviceConnected) {
    delay(100);
    read_statusBOX();
  }
  else {
    Serial.println("BLE not connected");

    pCharacteristicStart->setValue("");
    pCharacteristicP1->setValue(" ");
    pCharacteristicP2->setValue(" ");
    int check_sens_porta = controllo_porta(); // digitalRead(contatto_porta); 

    if (check_sens_porta == porta_open) {
      statoled_allarme = 1;
      pCharacteristicOpen->setValue("3");

      Serial.println("non connesso BLE, allarme porta aperta");
    }
  }
   server.handleClient();
   delay(1);
}

closed 2019-09-20 by stale[bot]