espressif/arduino-esp32 #2965: Can i use BasicOTA on AP mode ?

MkLHX, 2019-07-05:
Hi there!

I'm looking for solution about OTA with ESP32. Actually i use OTAWebUpdate like method to upload my firmware but my devices aren't all connected through my local network, some of thems are on AP mode and the ESPAsyncWebserver have some troubles to render light html view from progmem in AP mode.
So i'd like to know if STA is mandatory to run [BasicOTA](https://github.com/espressif/arduino-esp32/blob/master/libraries/ArduinoOTA/examples/BasicOTA/BasicOTA.ino)

lbernstone, 2019-07-05:
The OTA components do not have any dependency on WiFi.  So, they should work in either STA or AP mode.

MkLHX, 2019-07-05:
Yep i have the same idea but in fact when i run this example nothing works:
```
#include <WiFi.h>
#include <ArduinoOTA.h>

void setup() {
  Serial.begin(115200);
  Serial.println("Booting");
  WiFi.mode(WIFI_AP_STA);
  WiFi.softAP("esp32");
  
  Serial.print("local IP address: ");
  Serial.println(WiFi.softAPIP());

  // Port defaults to 3232
  // ArduinoOTA.setPort(3232);

  // Hostname defaults to esp3232-[MAC]
  // ArduinoOTA.setHostname("myesp32");

  // No authentication by default
  // ArduinoOTA.setPassword("admin");

  // Password can be set with it's md5 value as well
  // MD5(admin) = 21232f297a57a5a743894a0e4a801fc3
  // ArduinoOTA.setPasswordHash("21232f297a57a5a743894a0e4a801fc3");

  ArduinoOTA
    .onStart([]() {
      String type;
      if (ArduinoOTA.getCommand() == U_FLASH)
        type = "sketch";
      else // U_SPIFFS
        type = "filesystem";

      // NOTE: if updating SPIFFS this would be the place to unmount SPIFFS using SPIFFS.end()
      Serial.println("Start updating " + type);
    })
    .onEnd([]() {
      Serial.println("\nEnd");
    })
    .onProgress([](unsigned int progress, unsigned int total) {
      Serial.printf("Progress: %u%%\r", (progress / (total / 100)));
    })
    .onError([](ota_error_t error) {
      Serial.printf("Error[%u]: ", error);
      if (error == OTA_AUTH_ERROR) Serial.println("Auth Failed");
      else if (error == OTA_BEGIN_ERROR) Serial.println("Begin Failed");
      else if (error == OTA_CONNECT_ERROR) Serial.println("Connect Failed");
      else if (error == OTA_RECEIVE_ERROR) Serial.println("Receive Failed");
      else if (error == OTA_END_ERROR) Serial.println("End Failed");
    });

  ArduinoOTA.begin();

  Serial.println("Ready");
}

void loop() {
  ArduinoOTA.handle();
}
```
board esp32devmodule
project over platformio

lbernstone, 2019-07-05:
Works for me, as does https://github.com/lbernstone/asyncUpdate.  Set your debug level to verbose and see if there is another error somewhere.

MkLHX, 2019-07-09:
Hi @lbernstone, thanks for shared code. But this method is for update through webview. And it's possible to make OTA like this in both AP and STA mode.
Today, i'd like to know if BasicOTA is possible on AP mode. The goal is to make OTA when you're connected on the ESP32 through his own AP. 
And then run CTRL+U on Arduino IDE when connected on network port or update through 
```pio run -t upload --upload-port <ipaddress> like 192.168.4.1```

MkLHX, 2019-07-09:
Ok i find the trouble just now!! fucking windows network managment!

So many reading later, i find this comment on this thread 
https://community.platformio.org/t/esp32-ota-via-ethernet-doesnt-want-to-work/8308/18

And i check the default configuration of AP network on my desktop and getting a fucking public category for each ESP32 AP SSID i have on my wifi list.

So change this public in private and voilà the BasicOTA works fine!

So, now i'm going to send this information in the PlatformIO documentation. Thx for all

closed 2019-07-09 by MkLHX

Florencio333, 2022-01-08:
#include <Arduino.h>
#include <WiFi.h>
#include <AsyncTCP.h>
#include <ESPAsyncWebServer.h>
#include <AsyncElegantOTA.h>

 
const char *ssid = "ESP32-AP-WebServer";
const char* password = "12345678";
 
const char* PARAM_INPUT_1 = "state"; //guarda el estado en PARAM_INPUT

const int output = 2;// salida en la pata 2 y esta esta pata unida al led azul del esp32
const int buttonPin = 4; //boton pulsador normalmente abierto que cuando se pushea se conecta a positivo y el esp32 lee una señal de entrada

// Variables will change:
int buttonState=LOW;             // the current reading from the input pin

// Create AsyncWebServer object on port 80

// Create AsyncWebServer object on port 80
AsyncWebServer server(80);// crea un servidor asíncrono El término asíncrono se refiere al concepto de que más de una cosa ocurre al mismo tiempo, o múltiples cosas relacionadas ocurren sin esperar a que la previa se haya completado
AsyncWebSocket ws("/ws");// para procesar las solicitudes de servicio de red. 
//Index.html es la página por defecto dentro de los directorios de los servidores de los sitios webs que se carga siempre que se solicita un dominio., index sería la página principal o el home de nuestra web
// PROGMEM  es una manera de guardar una pagina web en el programa de arduino y volverlo a llamar sin causar que se congele el esp32. PROGMEM guarda y sirve de nuevo paginss web
const char index_html[] PROGMEM = R"rawliteral(  
<!DOCTYPE HTML>
<html>
<head>
  <title>ESP Web Server</title>
  <meta name="viewport" content="width=device-width, initial-scale=1">
  <link rel="icon" href="data:,">
  <style>
  html {
    font-family: Arial, Helvetica, sans-serif;
    text-align: center;
  }
  h1 {
    font-size: 1.8rem;
    color: white;
  }
  h2{
    font-size: 1.5rem;
    font-weight: bold;
    color: #143642;
  }
  .topnav {
    overflow: hidden;
    background-color: #143642;
  }
  body {
    margin: 0;
  }
  .content {
    padding: 30px;
    max-width: 600px;
    margin: 0 auto;
  }
  .card {
    background-color: #F8F7F9;;
    box-shadow: 2px 2px 12px 1px rgba(140,140,140,.5);
    padding-top:10px;
    padding-bottom:20px;
  }
  .button {
    padding: 15px 50px;
    font-size: 24px;
    text-align: center;
    outline: none;
    color: #fff;
[... 134 lines cut ...]
  });

    // Start ElegantOTA
 
  AsyncElegantOTA.begin(&server);
    server.begin();
    //Ahora manejaremos la declaración de rutas, que es donde podemos controlar los métodos permitidos por HTTP.
    server.on("/state", HTTP_GET, [] (AsyncWebServerRequest *request) {   // CON la ruta  "/state" LE DECIMOS A LA PAGINA WEB que actualice los botones cuando fisicamente activamos el switch
    request->send(200, "text/plain", String(digitalRead(output)).c_str());
  });
}
  
void loop() {
 
  if (digitalRead(buttonPin) == HIGH && buttonState == LOW) { //digitalRead() – lee el valor del pin fisico correspondiente en este caso si esta en low entonces ejecuta activa el relay (RelayPin1, HIGH);
    digitalWrite(output, HIGH);
    buttonState = HIGH;
 
  }

  if (digitalRead(buttonPin) == LOW && buttonState == HIGH) {
    digitalWrite(output, LOW);
    buttonState = LOW;
      }
    }

italocjs, 2022-03-10:
I've added a sample code for anyone trying to make it works, and get the serverIndex error.   https://github.com/italocjs/ESP32_OTA_APMODE

italisecdir, 2022-06-07:
The simplest solution to OTA in AP Mode: [https://github.com/pangodream/ESP2SOTA](https://github.com/pangodream/ESP2SOTA)
Two lines of code and nothing else.

pangodream, 2022-06-18:
You can use ESP2SOTA library for this. In Arduino IDE / Library Manager search for the library ESP2SOTA, install it and then go to Samples and open AP_Mode sample
[image]