espressif/arduino-esp32 #2969: ESP8266 issue

Esp8266-hax0r, 2019-07-08:
I have an issue here.
My code:
`#include <WiFi.h>
#include <WebServer.h>
#include "DHT.h"

#define DHTTYPE DHT11

/*Put your SSID & Password*/
const char* ssid = "myssid";  // Enter SSID here
const char* password = "mypass";  //Enter Password here

WebServer server(80);

uint8_t DHTPin = 4; 
               
DHT dht(DHTPin, DHTTYPE);                

float Temperature;
float Humidity;
 
void setup() {
  Serial.begin(115200);
  delay(100);
  
  pinMode(DHTPin, INPUT);

  dht.begin();              

  Serial.println("Connecting to ");
  Serial.println(ssid);

  //connect to your local wi-fi network
  WiFi.begin(ssid, password);

  //check wi-fi is connected to wi-fi network
  while (WiFi.status() != WL_CONNECTED) {
  delay(1000);
  Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected..!");
  Serial.print("Got IP: ");  Serial.println(WiFi.localIP());

  server.on("/", handle_OnConnect);
  server.onNotFound(handle_NotFound);

  server.begin();
  Serial.println("HTTP server started");

}
void loop() {
  
  server.handleClient();
  
}

void handle_OnConnect() {

 Temperature = dht.readTemperature(); // Gets the values of the temperature
  Humidity = dht.readHumidity(); // Gets the values of the humidity 
  server.send(200, "text/html", SendHTML(Temperature,Humidity)); 
}

void handle_NotFound(){
  server.send(404, "text/plain", "Not found");
}

String SendHTML(float Temperaturestat,float Humiditystat){
  String ptr = "<!DOCTYPE html> <html>\n";
  ptr +="<head><meta name=\"viewport\" content=\"width=device-width, initial-scale=1.0, user-scalable=no\">\n";
  ptr +="<title>ESP32 Weather Report</title>\n";
  ptr +="<style>html { font-family: Helvetica; display: inline-block; margin: 0px auto; text-align: center;}\n";
  ptr +="body{margin-top: 50px;} h1 {color: #444444;margin: 50px auto 30px;}\n";
  ptr +="p {font-size: 24px;color: #444444;margin-bottom: 10px;}\n";
  ptr +="</style>\n";
  ptr +="</head>\n";
  ptr +="<body>\n";
  ptr +="<div id=\"webpage\">\n";
  ptr +="<h1>ESP32 Weather Report</h1>\n";
  
  ptr +="<p>Temperature: ";
  ptr +=(int)Temperaturestat;
  ptr +="°C</p>";
  ptr +="<p>Humidity: ";
  ptr +=(int)Humiditystat;
  ptr +="%</p>";
  
  ptr +="</div>\n";
  ptr +="</body>\n";
  ptr +="</html>\n";
  return ptr;
}`

Error it's showing...
`Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:8424
ho 0 tail 12 room 4
load:0x40080400,len:5868
entry 0x4008069c
Connecting to 
myssid
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4012d9c6  PS      : 0x00060830  A0      : 0x8012e6e5  A1      : 0x3ffb55a0  
A2      : 0x3ffb6064  A3      : 0x3ffb6064  A4      : 0x00000001  A5      : 0x3ffc67c8  
A6      : 0x60033084  A7      : 0x3ffbf63c  A8      : 0x3ffc6e90  A9      : 0x3ffb6070  
A10     : 0x00000001  A11     : 0xff000fff  A12     : 0xdfffffff  A13     : 0xbfffffff  
A14     : 0x80000000  A15     : 0x3ffae738  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x8012e6e9  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x4012d9c6:0x3ffb55a0 0x4012e6e2:0x3ffb55d0 0x4008f941:0x3ffb5650 0x4008877d:0x3ffb5690`

How do I fix it?

Daemach, 2019-07-08:
> String ptr = " \n";
> ptr +="<meta name="viewport" content="width=device-width,
> ptr +="\n";
> ESP32 Weather Report\n";

Start by installing and using this, per the usage instructions at the
bottom of the page:  https://github.com/me-no-dev/EspExceptionDecoder

Esp8266-hax0r, 2019-07-08:
I did. It only throws that.

lbernstone, 2019-07-08:
Likely a power issue.  WiFi init takes more power than many USB ports can give it.  Plug it into a real power supply.

Esp8266-hax0r, 2019-07-08:
Like what?

lbernstone, 2019-07-08:
https://www.aliexpress.com/item/32954620239.html

Esp8266-hax0r, 2019-07-08:
Thanks. How did you understand it is a power issue?

lbernstone, 2019-07-08:
Uh, b/c it is crashing when you try to init the WiFi.  It will typically give you a brownout error there, but it sometimes can't even fail cleanly if the current is insufficient.

Esp8266-hax0r, 2019-07-08:
OK. Will check it out.

closed 2019-09-20 by stale[bot]