espressif/arduino-esp32 #2970: assertion "heap != NULL && "free() target pointer is outside heap areas""

GeorgeFlorian, 2019-07-08:
OS: Linux 19.1
Board: ESP32-EVB

Hello !

I am trying to save values from a POST using ESPAsyncWebServer into a file.

```
  server.on("/staticIP", HTTP_POST, [](AsyncWebServerRequest * request){
    if(request->hasArg("saveStatic")){
      int params = request->params();
      String values_static[6];
      for(int i=0;i<params;i++){
        AsyncWebParameter* p = request->getParam(i);
        if(p->isPost()){
            logOutput((String)"POST[" + p->name().c_str() + "]: " + p->value().c_str() + "\n");
            values_static[i] = p->value();            
          } else {
              logOutput((String)"GET[" + p->name().c_str() + "]: " + p->value().c_str() + "\n");
            }
      } // for(int i=0;i<params;i++)

      if(values_static[0] != NULL &&
        values_static[0].length() != 0 &&
        values_static[1] != NULL &&
        values_static[1].length() != 0 &&
        values_static[2] != NULL &&
        values_static[2].length() != 0 &&
        values_static[3] != NULL &&
        values_static[3].length() != 0 &&
        values_static[4] != NULL &&
        values_static[4].length() != 0 &&
        values_static[5] != NULL &&
        values_static[5].length() != 0) {
            File inputsWrite = SPIFFS.open("/network.txt", "w");
            if(!inputsWrite) logOutput((String)"ERROR_INSIDE_POST ! Couldn't open file to write Static IP credentials !"); 
            inputsWrite.println(values_static[0]);   // SSID
            inputsWrite.println(values_static[1]);   // Password
            inputsWrite.println(values_static[2]);   // Local IP
            inputsWrite.println(values_static[3]);   // Gateway
            inputsWrite.println(values_static[4]);   // Subnet
            inputsWrite.println(values_static[5]);   // DNS
            inputsWrite.close();
            logOutput("Configuration saved !");
            request->redirect("/logs");
            shouldReboot = true;
      } else request->redirect("/staticIP");
    } else {
      request->redirect("/staticIP");
    }
    }); // server.on("/staticLogin", HTTP_POST, [](AsyncWebServerRequest * request)
```

But I get the following error:
```
POST[networkName]: ssid

POST[networkPassword]: password

POST[ipAddress]: 192.168.1.55

POST[gateway]: 0.0.0.0

POST[subnet]: 255.255.255.0

POST[dns]: 8.8.8.8

POST[saveStatic]: Save and Reboot

assertion "heap != NULL && "free() target pointer is outside heap areas"" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/heap_caps.c", line 267, function: heap_caps_free
abort() was called at PC 0x400e6d6f on core 1

Backtrace: 0x4008c7e0:0x3ffcec20 0x4008ca11:0x3ffcec40 0x400e6d6f:0x3ffcec60 0x40084c9b:0x3ffcec90 0x4008624d:0x3ffcecb0 0x4000bec7:0x3ffcecd0 0x400e118e:0x3ffcecf0 0x400d8937:0x3ffced10 0x400def23:0x3ffcee10 0x400dcf25:0x3ffcee50 0x400dcfa1:0x3ffceea0 0x40134d89:0x3ffceec0 0x40134e01:0x3ffcef00 0x4013548a:0x3ffcef20 0x4008877d:0x3ffcef50

Rebooting...
```
The backtrace says:
```
Decoding stack results
0x400e6d6f: _vfprintf_r at ../../../.././newlib/libc/stdio/vfprintf.c line 1063
0x40084c9b: esp_rom_spiflash_read at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spi_flash/spi_flash_rom_patch.c line 544
0x4008624d: rtc_clk_cpu_freq_get_config at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/soc/esp32/rtc_clk.c line 699
0x400e118e: _svfprintf_r at ../../../.././newlib/libc/stdio/vfprintf.c line 1018
0x400d8937: spiffs_res_to_errno at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spiffs/esp_spiffs.c line 425
0x400def23: nvs::HashList::clear() at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/nvs_flash/src/intrusive_list.h line 211
0x400dcf25: spiffs_object_truncate at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spiffs/spiffs/src/spiffs_nucleus.c line 1770
0x400dcfa1: spiffs_object_truncate at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spiffs/spiffs/src/spiffs_nucleus.c line 1797
0x4008877d: uxTaskPriorityGet at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/tasks.c line 1576

```

stickbreaker, 2019-07-08:
This is your problem:
```c++
      String values_static[6];
      for(int i=0;i<params;i++){
        AsyncWebParameter* p = request->getParam(i);
          if(p->isPost()){
              logOutput((String)"POST[" + p->name().c_str() + "]: " + p->value().c_str() + "\n");
              values_static[i] = p->value();            
            } else {
                logOutput((String)"GET[" + p->name().c_str() + "]: " + p->value().c_str() + "\n");
            }
       } // for(int i=0;i<params;i++)

```
you create an array of 6 elements(0..5), but then you try use `params` of them. since `params` is not limited too 6,  It DIES when you try to any elements greater than the 6th.  Either increase you allocation (bad) or handle each element one at a time (good).

Chuck.

GeorgeFlorian, 2019-07-09:
@stickbreaker Yes, you are right !
I noticed that right after posting here.

closed 2019-07-09 by GeorgeFlorian