espressif/arduino-esp32 #2971: Wifi mode AP pings only few seconds

psykokwak-com, 2019-07-08:
Hi all,
I simply use the AP WiFi exemple.
A start a ping on 192.168.4.1 then I connect to the ESP32. The WiFi connection is OK and I receive my pings... But only the first 3 or 4. Then after the ping stops until y disconnect then I reconnect.
The ping test is only for test. I have no connectivity at all (the embedded webserver does not respond).
I tried with my smartphone too. Same.

I Use the latest Arduino version (1.0.2).

psykokwak-com, 2019-07-25:
Considering this code : 
```C
#include <WiFi.h>
#include <WiFiClient.h>
#include <WiFiAP.h>
#include <WebServer.h>

WebServer server(80);

void setup() {
  Serial.begin(115200);
  Serial.println();
  Serial.println("Configuring access point...");

  WiFi.softAP("MyAP", "");
  IPAddress myIP = WiFi.softAPIP();
  Serial.print("AP IP address: ");
  Serial.println(myIP);

  server.on("/", []() {
    server.send(200, "text/plain", "OK");
  });

  server.begin();
}

void loop() {
  // ping 192.168.4.1 from my laptop works until I uncomment this
  // line then I try to reach http://192.168.4.1/
  // Then the ping command stops.
  // server.handleClient();
}
```

With a fresh install, and a "erase_flash" command on the ESP32.
As soon as I try to connect to the webserver using my browser.

```
Configuring access point...
I (31) wifi: wifi driver task: 3ffb5700, prio:23, stack:3584, core=0
I (71) wifi: wifi firmware version: 6d404d4
I (71) wifi: config NVS flash: disabled
I (74) wifi: config nano formating: disabled
I (79) wifi: Init dynamic tx buffer num: 32
I (82) wifi: Init data frame dynamic rx buffer num: 10
I (87) wifi: Init management frame dynamic rx buffer num: 10
I (93) wifi: Init static rx buffer size: 1600
I (97) wifi: Init static rx buffer num: 8
I (100) wifi: Init dynamic rx buffer num: 10
I (163) wifi: mode : softAP (30:ae:a4:cc:4c:5d)
I (164) wifi: Init max length of beacon: 752/752
I (164) wifi: Init max length of beacon: 752/752
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 13 - AP_START
AP IP address: [D][WiFiGeneric.cpp:336] _eventCallback(): Event: 13 - AP_START
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 13 - AP_START
1
I (74080) wifi: n:1 0, o:1 0, ap:1 1, sta:255 255, prof:1
I (74080) wifi: station: 44:1c:a8:e3:26:ef join, AID=1, bgn, 20
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 15 - AP_STACONNECTED
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 17 - AP_STAIPASSIGNED
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 17 - AP_STAIPASSIGNED
```

Is there something I doing wrong ?

lbernstone, 2019-07-25:
Put a delay in your loop.  50ms should give the system some time to do other tasks.

psykokwak-com, 2019-07-25:
Nothing change!

lbernstone, 2019-07-25:
See if the sdkconfig.h changes in https://github.com/espressif/arduino-esp32/commit/b0d8d4dd442b2953e57b16d05d6bf2a893334f1d help.  You can copy the sdkconfig.h into your sketch folder and edit it there.

psykokwak-com, 2019-07-26:
Are you sure that if I put the sdkconfig.h file into my arduino sketch folder, the arduino toolchain will use it to rebuild the binary ?

closed 2019-10-08 by stale[bot]