espressif/arduino-esp32 #2975: Calling SPIFFS.begin causes "dangerous relocation" error

ssilverman, 2019-07-10:
### Hardware:
Board:							Adafruit ESP32 Feather
Core Installation version:			espressif32@1.9.0, tool-esptoolpy@1.20600.0, toolchain-xtensa32@2.50200.80, framework-arduinoespressif32@2.10002.190628, framework-espidf@3.30202.190627, tool-mkspiffs@2.230.0
IDE name:						PlatformIO (v4.0.0) on VSCode (v1.36.1)
Flash Frequency:					80MHz
PSRAM enabled:                                    Don't know
Upload Speed:					N/A
Computer OS:                                        Mac OSX

### Description:
In a fairly large program, when I #include <SPIFFS.h> and then call `SPIFFS.begin(true, "/spiffs")`, I get this error:

```
Linking .pioenvs/featheresp32/firmware.elf
/Users/shawn/.platformio/packages/framework-arduinoespressif32/tools/sdk/lib/libspiffs.a(esp_spiffs.o): In function `vfs_spiffs_update_mtime':
/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spiffs/esp_spiffs.c:270:(.text.vfs_spiffs_update_mtime+0x12): dangerous relocation: windowed longcall crosses 1GB boundary; return may fail: time
collect2: error: ld returned 1 exit status
```

This feels very similar to #385.

ssilverman, 2019-07-10:
Looking at esp_spiffs.c, I thought that undefining CONFIG_SPIFFS_USE_MTIME would fix the problem, so I tried commenting out this line in sdkconfig.h:

```cpp
#define CONFIG_SPIFFS_USE_MTIME 1
```

This did not work.

atanisoft, 2019-08-02:
@ssilverman do you have any functions named "time" in the rest of your project? This seems like an overlapping function declaration.

closed 2019-10-15 by stale[bot]