espressif/arduino-esp32 #2977: WiFiClient.cpp  read "connection reset by peer  at line 409"

WangYifa, 2019-07-11:
int WiFiClient::read(uint8_t *buf, size_t size)
{
    int res = -1;
    res = _rxBuffer->read(buf, size);
    if(_rxBuffer->failed()) {
        log_e("fail on fd %d, errno: %d, \"%s\"", fd(), errno, strerror(errno));
        stop();
    }
    return res;
}

This error caused the device to restart after a while.
I want to know why this problem appears and how to solve it.

WangYifa, 2019-07-11:
i am using the "arduinoespressif32\libraries\HTTPClient\examples\StreamHttpClient"

FlyingWraptor, 2019-07-23:
The error gets thrown due to it attempting to read from a timed out or blocking connection.
I'm having the same issue for the write function when the target host is blocking the connection. My best guess is that the ESP32 Arduino wrapper made some changes to the interaction with the IDF library causing this freshly created bug/unhandled exception

My errors:
```
[E][WiFiClient.cpp:282] setOption(): fail on fd 54, errno: 22, "Invalid argument"
[E][WiFiClient.cpp:365] write(): fail on fd 54, errno: 104, "Connection reset by peer"
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x40141c8c  PS      : 0x00060a30  A0      : 0x80142039  A1      : 0x3ffb1d30  
A2      : 0x00000000  A3      : 0x3ffb1d8f  A4      : 0x00000001  A5      : 0x00000001  
A6      : 0x00000000  A7      : 0xffffffa7  A8      : 0x00000000  A9      : 0x00000000  
A10     : 0x00000000  A11     : 0x3ffc88d4  A12     : 0x00000000  A13     : 0x00000001  
A14     : 0x00060020  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000008  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x40141c8c:0x3ffb1d30 0x40142036:0x3ffb1d50 0x40144aaa:0x3ffb1d80 0x4011f103:0x3ffb1db0 0x4011ed92:0x3ffb1dd0 0x400d2e32:0x3ffb1f10 0x400d2cc1:0x3ffb1f50 0x400d1ab3:0x3ffb1f70 0x400d4a7b:0x3ffb1fb0 0x40088b49:0x3ffb1fd0
```

Using the ArduinoWebSockets library which already kinda covered (but not solved)  the issue a here:
https://github.com/gilmaimon/ArduinoWebsockets/issues/10

atanisoft, 2019-08-02:
Do you have a reproducible test case? Any decoded backtrace?

Also please use the issue template for future issues.

technorainbows, 2019-09-17:
I had a similar issue that occurred only when using the http.post function of the httpclient library  (but the error was in the wificlient library, same as with OP). 

After considerable debugging, I found two things: 

1) the issue was my error -- caused because I was sending a payload with http.post (the library requires I send something), even though my API wasn't set up to receive a payload. Since the API didn't receive the payload, my guess is it was left in the buffer and caused regular crashes when reading from the buffer later. Once I fixed this in the API, no more crashing.  

2) That said, I think additional error handling would be very helpful for this library as this error shouldn't cause the board to reset. It appears that when fillBuffer returns a 0 it causes things to crash. (So, when I played around with  modifying the WiFiClient library and had it return 1 on line 79, instead of a 0, the crashing stopped. It did mean occasionally garbley-gook was read into the http.post response, but it didn't bring the whole device down at least.)

Original:
```
          int res = recv(_fd, _buffer + _fill, _size - _fill, MSG_DONTWAIT);
          if(res < 0) {
                if(errno != EWOULDBLOCK) {
                    log_e("failed = true");
                    _failed = true;
                }
                return 0;
           }
```

Edit for debugging purposes (that stopped board from crashing):

            int res = recv(_fd, _buffer + _fill, _size - _fill, MSG_DONTWAIT);
            if(res < 0) {
                // if(errno != EWOULDBLOCK) {
                //     log_e("failed = true");
                //     _failed = true;
                // }
                return 1;
            }

Here's the error message and decoded backtrace from my problem:

Debugging message from a bit earlier in debugging process (so line numbers might not match up as I added several additional debugging log messages in the library), but I don't seem to have a saved later copy of this error msg:
>  [E][WiFiClient.cpp:436] read(): fail on fd 63, errno: 104, "Connection reset by peer"
> 16:46:16.774 -> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

...

> Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
> 18:19:42.288 -> Core 1 register dump:
> 18:19:42.288 -> PC      : 0x400d4b54  PS      : 0x00060030  A0      : 0x800d4f61  A1      : 0x3ffb1cd0  
> 18:19:42.288 -> A2      : 0x00000000  A3      : 0x3ffb1d2f  A4      : 0x00000001  A5      : 0x00000001  
> 18:19:42.288 -> A6      : 0x3ffc2308  A7      : 0x00000068  A8      : 0x00000000  A9      : 0x00000000  
> 18:19:42.288 -> A10     : 0x00000000  A11     : 0x40086924  A12     : 0x00000050  A13     : 0x00000000  
> 18:19:42.325 -> A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
> 18:19:42.325 -> EXCVADDR: 0x00000008  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  
> 18:19:42.325 -> 
> 18:19:42.325 -> Backtrace: 0x400d4b54:0x3ffb1cd0 0x400d4f5e:0x3ffb1cf0 0x4016d582:0x3ffb1d20 0x400dd051:0x3ffb1d50 0x400dd0e9:0x3ffb1d70 0x400d7ba5:0x3ffb1d90 0x400d8697:0x3ffb1e30 0x400d8703:0x3ffb1e70 0x400d8727:0x3ffb1e90 0x400d4022:0x3ffb1eb0 0x400de28d:0x3ffb1fb0 0x40088c45:0x3ffb1fd0
> 18:19:42.361 -> 

...

> PC: 0x400d4b54: WiFiClientRxBuffer::read(unsigned char*, unsigned int) at ~/Library/Arduino15/packages/esp32/hardware/esp32/1.0.3/libraries/WiFi/src/WiFiClient.cpp line 107
> EXCVADDR: 0x00000008
> 
> Decoding stack results
> 0x400d4b54: WiFiClientRxBuffer::read(unsigned char*, unsigned int) at ~/Library/Arduino15/packages/esp32/hardware/esp32/1.0.3/libraries/WiFi/src/WiFiClient.cpp line 107
> 0x400d4f5e: WiFiClient::read(unsigned char*, unsigned int) at ~/Library/Arduino15/packages/esp32/hardware/esp32/1.0.3/libraries/WiFi/src/WiFiClient.cpp line 434
> 0x4016d582: WiFiClient::read() at ~/Library/Arduino15/packages/esp32/hardware/esp32/1.0.3/libraries/WiFi/src/WiFiClient.cpp line 345
> 0x400dd051: Stream::timedRead() at ~/Library/Arduino15/packages/esp32/hardware/esp32/1.0.3/cores/esp32/Stream.cpp line 36
> 0x400dd0e9: Stream::readStringUntil(char) at ~/Library/Arduino15/packages/esp32/hardware/esp32/1.0.3/cores/esp32/Stream.cpp line 287
> 0x400d7ba5: HTTPClient::handleHeaderResponse() at ~/Library/Arduino15/packages/esp32/hardware/esp32/1.0.3/libraries/HTTPClient/src/HTTPClient.cpp line 1118
> 0x400d8697: HTTPClient::sendRequest(char const*, unsigned char*, unsigned int) at ~/Library/Arduino15/packages/esp32/hardware/esp32/1.0.3/libraries/HTTPClient/src/HTTPClient.cpp line 573
> 0x400d8703: HTTPClient::POST(unsigned char*, unsigned int) at ~/Library/Arduino15/packages/esp32/hardware/esp32/1.0.3/libraries/HTTPClient/src/HTTPClient.cpp line 493
> 0x400d8727: HTTPClient::POST(String) at ~/Library/Arduino15/packages/esp32/hardware/esp32/1.0.3/libraries/HTTPClient/src/HTTPClient.cpp line 498
> 0x400d4022: loop() at /Users/ashliana/code/arduino-sketches/LED-IOT-API/LED-IOT-http-client/LED-IOT-http-client.ino line 273
> 0x400de28d: loopTask(void*) at ~/Arduino15/packages/esp32/hardware/esp32/1.0.3/cores/esp32/main.cpp line 19
> 0x40088c45: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143
>

janainascal, 2019-10-30:
I have a similar error, when I send the wifi credentions by a Portal.
I need to debug a little more, but I agree with @technorainbows, considering the error described, it shouldn't reset the board.

My log before the crash:
```
mg_rpc.c:293                    Portal.Set via HTTP 192.168.4.2:48025
mgos_captive_portal:252         Captive Portal WiFi Setup Testing SSID: Vinicius PASS: 147258369
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
```

closed 2020-01-12 by stale[bot]