espressif/arduino-esp32 #2978: esp32 arduino enviroment espressif32 1.9.0 update

zekageri, 2019-07-11:
I was used espressif 1.8.0  in VSCODE with PlatformIO. My OTA upload was working fine until i updated to the latest espressif32 release which is 1.9.0. Now i get the following Terminal 
OUTPUT:

09:35:49 [DEBUG]: Options: {'timeout': 10, 'esp_ip': '10.255.0.200',
'host_port': 43495, 'image': '.pioenvs\\esp32dev\\firmware.bin', 'host_ip': '0.0.0.0', 'auth': '', 'esp_port': 8266, 'spiffs': False, 'debug': True, 'progress': True}
09:35:49 [INFO]: Starting on 0.0.0.0:43495
09:35:49 [INFO]: Upload size: 1370960
Sending invitation to 10.255.0.200 ..........
09:35:49 [ERROR]: No response from the ESP
*** [upload] Error 1

Nothing more.

If i install 1.8.0, all is working fine again. However i can upload it via COM port and all part of my sketch is working.
My OTA update function is from the OTA examples:

```
  ArduinoOTA.onStart([]() {
    olvasunk_e = false;
    events.send("Update Start", "ota");
  });
  ArduinoOTA.onEnd([]() {
    olvasunk_e = true;
    events.send("Update End", "ota");
  });
  ArduinoOTA.onProgress([](unsigned int progress, unsigned int total) {
    char p[32];
    sprintf(p, "Progress: %u%%\n", (progress / (total / 100)));
    events.send(p, "ota");
  });
  ArduinoOTA.onError([](ota_error_t error) {
    if (error == OTA_AUTH_ERROR) events.send("Auth Failed", "ota");
    else if (error == OTA_BEGIN_ERROR) events.send("Begin Failed", "ota");
    else if (error == OTA_CONNECT_ERROR) events.send("Connect Failed", "ota");
    else if (error == OTA_RECEIVE_ERROR) events.send("Recieve Failed", "ota");
    else if (error == OTA_END_ERROR) events.send("End Failed", "ota");
  });
  
  ArduinoOTA.setHostname(hostName);
  ArduinoOTA.begin();

  MDNS.addService("http","tcp",80);
```

I'am using the latest from everything. I updated the platformIO core and every aspects of the PlatformIO extension as possible.

lbernstone, 2019-07-11:
Make sure you have your ports matching.  Looks like it is trying to connect to port 8266, but the OTA example uses 3232.

zekageri, 2019-07-12:
> Make sure you have your ports matching. Looks like it is trying to connect to port 8266, but the OTA example uses 3232.

I don't modify the port. It is the default. I just download the new espressif 1.9.0. Where can i modify it?
If i write to the platformIO.ini that:
```
build_flags = 
    --port             =   3232
```
I got a compiler error.

lbernstone, 2019-07-12:
[RTFM](https://docs.platformio.org/en/latest/platforms/espressif32.html#over-the-air-ota-update) or change it on the esp side:
`ArduinoOTA.setPort(8266);`

zekageri, 2019-07-15:
> [RTFM](https://docs.platformio.org/en/latest/platforms/espressif32.html#over-the-air-ota-update) or change it on the esp side:
> `ArduinoOTA.setPort(8266);`

Nop. That was not the issue. I applied the line that you mentioned and i get the esp no response error immidiately when i try to upload my program with the latest 1.9.0.
Last serial monitor:

```
esptool.py v2.6
Configuring upload protocol...
AVAILABLE: esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = espota
Uploading .pio\build\esp32dev\firmware.bin
08:41:20 [DEBUG]: Options: {'timeout': 10, 'esp_ip': '10.255.0.200', 'host_port': 45317, 'image': '.pio\\build\\esp32dev\\firmware.bin', 'host_ip': '0.0.0.0', 'auth': '', 'esp_port': 8266, 'spiffs': False, 'debug': True, 'progress': True}
08:41:20 [INFO]: Starting on 0.0.0.0:45317
08:41:20 [INFO]: Upload size: 1366240
Sending invitation to 10.255.0.200 ..........
08:41:20 [ERROR]: No response from the ESP
*** [upload] Error 1
====================== [ERROR] Took 21.77 seconds ======================
The terminal process terminated with exit code: 1
```

zekageri, 2019-07-15:
Okay. That actuallly solved my problem. I had to put that line 
`ArduinoOTA.setPort(8266); `
to my setup() and do a manual update on COM port, and than i restarted everything and now it is working.
Thank you!

closed 2019-07-15 by zekageri