espressif/arduino-esp32 #2979: ESP32 ASYNC-TCP. ack timeout 4

zekageri, 2019-07-11:
I'am using the latest Async_TCP lib with espressif 1.8.0 on VSCODE with PlatformIO.
With the latest ASYNC_TCP lib update all my timeout and corrupt heap poisoning issues are gone in Ethernet mode or STA mode. But if am setting a softAP like this:

```
WiFi.setAutoConnect(false);
  WiFi.disconnect(true);
  WiFi.enableAP(true);
  WiFi.setSleep(false);

  WiFi.softAP(ssid,password, 12, 0, 1);
  delay(100);
  Serial.print("WifiCH: ");
  Serial.println(WiFi.channel());
```

And connect to my ESP's acces point with an android phone, after a while i get these error message when build_flags = -DCORE_DEBUG_LEVEL=5.

```
[D][WiFiClient.cpp:482] connected(): Disconnected: RES: 0, ERR: 128
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 15 - AP_STACONNECTED
dhcps: send_offer>>udp_sendto result 0
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 17 - AP_STAIPASSIGNED
[W][AsyncTCP.cpp:717] _poll(): pcb is NULL
dhcps: send_offer>>udp_sendto result 0
dhcps: send_nak>>udp_sendto result 0
[D][WiFiClient.cpp:482] connected(): Disconnected: RES: 0, ERR: 128
[W][AsyncTCP.cpp:717] _poll(): pcb is NULL
[W][AsyncTCP.cpp:717] _poll(): pcb is NULL
[D][WiFiClient.cpp:482] connected(): Disconnected: RES: 0, ERR: 128
[W][AsyncTCP.cpp:717] _poll(): pcb is NULL
[W][AsyncTCP.cpp:717] _poll(): pcb is NULL
[W][AsyncTCP.cpp:717] _poll(): pcb is NULL
[D][WiFiClient.cpp:482] connected(): Disconnected: RES: 0, ERR: 128
[D][WiFiClient.cpp:482] connected(): Disconnected: RES: 0, ERR: 128
[W][AsyncTCP.cpp:717] _poll(): pcb is NULL
[W][AsyncTCP.cpp:717] _poll(): pcb is NULL
[D][WiFiClient.cpp:482] connected(): Disconnected: RES: 0, ERR: 128
[D][WiFiClient.cpp:482] connected(): Disconnected: RES: 0, ERR: 128
[W][AsyncTCP.cpp:717] _poll(): pcb is NULL
[W][AsyncTCP.cpp:717] _poll(): pcb is NULL
[D][WiFiClient.cpp:482] connected(): Disconnected: RES: 0, ERR: 128
[W][AsyncTCP.cpp:730] _poll(): ack timeout 4
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 16 - AP_STADISCONNECTED
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_nak>>udp_sendto result 0
dhcps: send_nak>>udp_sendto result 0
dhcps: send_nak>>udp_sendto result 0
```

As you can see, after a while i get some PCB NULL errors and some disconnect, it is because iam switching pages. And then i get the ack timeout 4 and all of the ap mode is stuck. My code works and i can reach the web via ethernet interface, while the AP mode is present but stucked.

I tried my pages from both SPIFFS and Compiled into progmem. Nothing changed.
I tried the AP mode channels and with SSID and without it. The problem is still there. 
But just in AP mode.

bertmelis, 2019-07-12:
> With the latest ASYNC_TCP lib update
When I look at the repo, line 717 doesn't generate this message. Are you really using the latest AsyncTCP?

zekageri, 2019-07-12:
> > With the latest ASYNC_TCP lib update
> 
> When I look at the repo, line 717 doesn't generate this message. Are you really using the latest AsyncTCP?

Yes:O I upgraded yesterday. Or two days ago maybe.
Via PIO lib Upgrade.

bertmelis, 2019-07-12:
The error you're seeing is on line 893: https://github.com/me-no-dev/AsyncTCP/blob/90715ae6b3ee72e9e40cd3dd2f9609217bf3ee02/src/AsyncTCP.cpp#L893

4 commits since...

zekageri, 2019-07-15:
> The error you're seeing is on line 893: https://github.com/me-no-dev/AsyncTCP/blob/90715ae6b3ee72e9e40cd3dd2f9609217bf3ee02/src/AsyncTCP.cpp#L893
> 
> 4 commits ago it was on 717

Okay you were right. I did not used the latest because i forget to copy ot to my project's lib folder. 
The case is that the ack timeout issue was gone. But if iam in AP mode and i going trought my web server with my android phone, Eventually i get that error and reboot:

```
dhcps: send_offer>>udp_sendto result 0
assertion "new_rcv_ann_wnd <= 0xffff" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/tcp.c", line 779, function: tcp_update_rcv_ann_wnd
abort() was called at PC 0x40102abb on core 0

Backtrace: 0x4008d86c:0x3ffb3d70 0x4008da9d:0x3ffb3d90 0x40102abb:0x3ffb3db0 0x401296da:0x3ffb3de0 0x40129769:0x3ffb3e00 0x401557dd:0x3ffb3e20 0x40126108:0x3ffb3e40 0x40089789:0x3ffb3e70
```
I'am going to decode it but my PlatformIO can not build my sketch because it can not find espressif32 builder. I assume that is because the new 1.9.0 update. Coming back soon

zekageri, 2019-07-15:
Okay, with the latest async tcp lib the multiheap problem is back again.

CORRUPT HEAP: Bad head at 0x3ffdc504. Expected 0xabba1234 got 0x00000000
assertion "head != NULL" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/multi_heap_poisoning.c", line 214, function: multi_heap_free
abort() was called at PC 0x40102abb on core 1

Backtrace: 0x4008d86c:0x3ffd5620 0x4008da9d:0x3ffd5640 0x40102abb:0x3ffd5660 0x4008d4e1:0x3ffd5690 0x40085a62:0x3ffd56b0 0x40087249:0x3ffd56d0 0x4000bec7:0x3ffd56f0 0x40107c51:0x3ffd5710 0x40156265:0x3ffd5740 0x40089789:0x3ffd5770

zekageri, 2019-07-15:
Okay. I was wrong. With the latest Async TCP And latest ESP ASYNC webserver libs almost everything is okay. The problem is with the 1.9.0 arduino-espressif32 framwork. If i upgrade from 1.8.0 , the Corrupt heap problem is back again, and all my web things are slow A.F. But if i revert back to 1.8.0, the async web is fast and no heap corruption, but the ACK timeout is back again. In AP mode, the esp stopped working after some time, and i get ack timeout 4 error on serial monitor from async web.
[W][AsyncTCP.cpp:730] _poll(): ack timeout 4
I have to rejoin to the esp32's wifi network to be able to undo the freezee.

closed 2019-07-17 by zekageri

MkLHX, 2019-07-17:
Hi there!  Getting same error and crash after PIO code, ESP32 Framework and python updates.

```
[W][AsyncTCP.cpp:717] _poll(): pcb is NULL
```
and sometimes get 
```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.     
Core 0 register dump:
PC      : 0x40116214  PS      : 0x00060b30  A0      : 0x80111dda  A1      : 0x3ffaf210 

A2      : 0x3ffd3390  A3      : 0x22657261  A4      : 0x0000716b  A5      : 0x00000000 

A6      : 0x3ffb631c  A7      : 0x00000000  A8      : 0x43081433  A9      : 0x706d8be0 

A10     : 0x2d656c75  A11     : 0x00001670  A12     : 0x706d7570  A13     : 0x00000b38 

A14     : 0x00060b20  A15     : 0x00000000  SAR     : 0x00000019  EXCCAUSE: 0x0000001c 

EXCVADDR: 0x00000004  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff 

Backtrace: 0x40116214:0x3ffaf210 0x40111dd7:0x3ffaf240 0x400d8c49:0x3ffaf260 0x4010db2c:0x3ffaf280 0x400887f9:0x3ffaf2b0
```

or 

```
assertion "new_rcv_ann_wnd <= 0xffff" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/tcp.c", line 779, function: tcp_update_rcv_ann_wnd
abort() was called at PC 0x400f0d17 on core 0

Backtrace: 0x4008d264:0x3ffaf1b0 0x4008d495:0x3ffaf1d0 0x400f0d17:0x3ffaf1f0 0x40114c12:0x3ffaf220 0x40114ca1:0x3ffaf240 0x400d9919:0x3ffaf260 0x401116a4:0x3ffaf280 0x40089029:0x3ffaf2b0
```

Both exception decode give me something about lwip/tcp.c 
Guru Meditation:
```

Decoding stack results
0x40116214: tcp_output at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/tcp_out.c line 1027
0x40111dd7: tcp_recved at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/tcp.c line 828
0x4010db2c: tcpip_thread at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/api/tcpip.c line 124
0x400887f9: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143
```
assertion "new_rcv_ann_wnd" decodded:
```
Decoding stack results
0x4008d264: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 155
0x4008d495: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 170
0x400f0d17: __assert_func at ../../../.././newlib/libc/stdlib/assert.c line 63
0x40114c12: tcp_update_rcv_ann_wnd at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/tcp.c line 779
0x40114ca1: tcp_recved at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/tcp.c line 820
0x401116a4: tcpip_thread at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/api/tcpip.c line 124
0x40089029: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143
```

So just rollback esp32-arduino framework give you correct fix?

CircuitSetup, 2019-07-24:
I'm getting a very similar issue. I've tested with 1.0.2 and 1.0.3 rc1, and IDF 3.2 & 3.2.2. ESPAsyncWebServer and AsyncTCP are up to date. I've also tried with brand new ESP32s.

I noted that EEPROM was converted to work with NVS in the latest version, but that doesn't seem to make a difference. Sometimes I got a random error and sometimes it works without issue. Slightly different errors occur each time, making it harder to debug.

MkLHX, 2019-07-24:
Still waiting for a miracle cause i can't debug more at this time

johnnytolengo, 2020-10-31:
Here the same issue: 

[W][AsyncTCP.cpp:949] _poll(): pcb is NULL

with last versions.
could someone solve it?