espressif/arduino-esp32 #2980: About log_printf

ducalex, 2019-07-11:
### Hardware:
Board:							
Core Installation version:			d5fdd715efddfe5838996eeee7840223f81f52bd
IDE name:						IDF component
Flash Frequency:					80Mhz
PSRAM enabled:                                      no
Upload Speed:					115200
Computer OS:                                          Windows 8.1

### Description:
I'm investigating a silent freeze on a board and I stumbled upon log_printf. At this point I do not think my issue is related to this function, but the code left me curious nonetheless.

I understand the uart lock part, but I don't get all the va_copy/malloc/buffer/vsnprintf+ets_printf.
I know ets_printf is missing types and that's why there is a buffered vsnprintf, but why not just call printf directly at this point?

Thanks

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp
int log_printf(const char *format, ...)
{
    if(s_uart_debug_nr < 0){
        return 0;
    }
    static char loc_buf[64];
    char * temp = loc_buf;
    int len;
    va_list arg;
    va_list copy;
    va_start(arg, format);
    va_copy(copy, arg);
    len = vsnprintf(NULL, 0, format, arg);
    va_end(copy);
    if(len >= sizeof(loc_buf)){
        temp = (char*)malloc(len+1);
        if(temp == NULL) {
            return 0;
        }
    }
    vsnprintf(temp, len+1, format, arg);
#if !CONFIG_DISABLE_HAL_LOCKS
    if(_uart_bus_array[s_uart_debug_nr].lock){
        xSemaphoreTake(_uart_bus_array[s_uart_debug_nr].lock, portMAX_DELAY);
        ets_printf("%s", temp);
        xSemaphoreGive(_uart_bus_array[s_uart_debug_nr].lock);
    } else {
        ets_printf("%s", temp);
    }
#else
    ets_printf("%s", temp);
#endif
    va_end(arg);
    if(len >= sizeof(loc_buf)){
        free(temp);
    }
    return len;
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

stickbreaker, 2019-07-11:
@ducalex The `log_printf()` has a 64byte stack based buffer the first call,` len = vsnprintf(NULL, 0, format, arg);` is just to verify the resulting output line will fit into the 64 byte buffer.  If it won't, it then allocates a heap buffer to hold the output line.  

But, the `va_copy()`, `va_start()` and `va_end()` seem to be mixed up?

My understanding of `va_` is that the data structures are destroyed when `va_arg()` uses them.  to me, it would seem that the sequence of calls is wrong? 

It should be this way:
 ```c++
    va_start(arg, format);
    va_copy(copy, arg);
    len = vsnprintf(NULL, 0, format, copy); // use the copy because vsnprintf() mangles it?
    va_end(copy); // done with copy, dispose of it
//cut
  allocate heap buffer if needed
//cut
    vsnprintf(temp, len+1, format, arg);
    va_end(arg);
 ```
instead of:
 ```c++
    va_start(arg, format);
    va_copy(copy, arg);
    len = vsnprintf(NULL, 0, format, arg); //I think 'copy' should be used
    va_end(copy); // done with copy? dispose of it? where was it ever used?
//cut
  allocate heap buffer if needed
//cut
    vsnprintf(temp, len+1, format, arg);
    va_end(arg);
 ```
Chuck.

ducalex, 2019-07-11:
@stickbreaker  But calling printf/vprintf wouldn't need a local buffer at all.

Consider:
`````cpp
int log_printf(const char *format, ...)
{
    if(s_uart_debug_nr < 0){
        return 0;
    }
    int len = 0;
    va_list arg;
    va_start(arg, format);
#if !CONFIG_DISABLE_HAL_LOCKS
    if(_uart_bus_array[s_uart_debug_nr].lock){
        xSemaphoreTake(_uart_bus_array[s_uart_debug_nr].lock, portMAX_DELAY);
        len = vprintf(format, arg);
        xSemaphoreGive(_uart_bus_array[s_uart_debug_nr].lock);
    } else {
        len = vprintf(format, arg);
    }
#else
    len = vprintf(format, arg);
#endif
    va_end(arg);
    return len;
}
`````

Is there a difference in practice?

stickbreaker, 2019-07-11:
@ducalex `vprintf()` prints to console? 
`ets_printf()` prints to the current debug output, which can change. (uart0..2,Flash,network).
Also, the esp32 doesn't have a console?

Chuck.

ducalex, 2019-07-11:
That could make sense. However the default macros do use vprintf: 

https://github.com/espressif/esp-idf/blob/master/components/log/log.c#L94

closed 2019-07-13 by ducalex