espressif/arduino-esp32 #2981: delay() function doesn't work for less than 10ms

ducalex, 2019-07-11:
### Hardware:
Board:							Lolin32
Core Installation version:			d5fdd715efddfe5838996eeee7840223f81f52bd
IDE name:						IDF component
Flash Frequency:					40Mhz
PSRAM enabled:                                     no
Upload Speed:					115200
Computer OS:                                         Windows 8.1

### Description:
delay() doesn't work for periods smaller than one tick(10ms) (it is also rounded down if not divisible by 1 tick). 

It is a small change but I can create a pull request if it is the proper way to do things also, just tell me.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp
// This implementation should work:
void delay(uint32_t ms)
{
    uint32_t start = millis();
    vTaskDelay(ms / portTICK_PERIOD_MS);
    int remains = ms - (millis() - start); // Can be negative because vTaskDelay time isn't guaranteed
    if (remains > 0) delayMicroseconds(remains * 1000);
}
```

stickbreaker, 2019-07-11:
@ducalex 
Why do you think `portTICK_PERIOD_MS` is 10?

Chuck.

stickbreaker, 2019-07-11:
@ducalex if you are worried the `portTICK_PERIOD_MS` will change in the future, then you could change `delay()` [current Delay coding](https://github.com/espressif/arduino-esp32/blob/7dbda4988b60f83ecfc887c7f92233b2fb3f63dd/cores/esp32/esp32-hal-misc.c#L142)
to:
```c++
void delay(uint32_t ms)
{
    vTaskDelay(ms / portTICK_PERIOD_MS);
    uint32_t remainderUS = (ms % portTICK_PERIOD_MS)*1000;
    if(remainderUS) delayMicroseconds(remainderUS);
}
```

ducalex, 2019-07-11:
> @ducalex
> Why do you think `portTICK_PERIOD_MS` is 10?
> 
> Chuck.

````
#define CONFIG_FREERTOS_HZ 100
#define configTICK_RATE_HZ				( CONFIG_FREERTOS_HZ )
#define portTICK_PERIOD_MS			( ( TickType_t ) 1000 / configTICK_RATE_HZ )\
````

vTaskDelay's argument is a tick number. Therefore it works correctly only on multiples of 10ms.

Your code would also resolve the problem indeed.

ducalex, 2019-07-11:
You can see the problem with this code:
````cpp
void setup()
{
    for (int i = 0; i < 20; i++) {
        int start = millis();
        delay(i);
        printf("i = %d, actual sleep = %dms\n", i, (int)(millis() - start));
    }
}
````

Output:
````
i = 0, actual sleep = 0ms
i = 1, actual sleep = 0ms
i = 2, actual sleep = 0ms
i = 3, actual sleep = 0ms
i = 4, actual sleep = 0ms
i = 5, actual sleep = 0ms
i = 6, actual sleep = 0ms
i = 7, actual sleep = 0ms
i = 8, actual sleep = 0ms
i = 9, actual sleep = 0ms
i = 10, actual sleep = 7ms
i = 11, actual sleep = 10ms
i = 12, actual sleep = 10ms
i = 13, actual sleep = 10ms
i = 14, actual sleep = 10ms
i = 15, actual sleep = 10ms
i = 16, actual sleep = 10ms
i = 17, actual sleep = 10ms
i = 18, actual sleep = 10ms
i = 19, actual sleep = 10ms
````

negativekelvin, 2019-07-12:
https://github.com/espressif/arduino-esp32/blob/b0d8d4dd442b2953e57b16d05d6bf2a893334f1d/tools/sdk/include/config/sdkconfig.h#L229

So I assume you are using arduino-esp32 as a component with your own sdkconfig settings?

stickbreaker, 2019-07-12:
@ducalex with current core [7dbda4](https://github.com/espressif/arduino-esp32/commit/7dbda4988b60f83ecfc887c7f92233b2fb3f63dd)
```
to console i = 0, actual sleep = 0ms
to console i = 1, actual sleep = 1ms
to console i = 2, actual sleep = 2ms
to console i = 3, actual sleep = 3ms
to console i = 4, actual sleep = 4ms
to console i = 5, actual sleep = 5ms
to console i = 6, actual sleep = 6ms
to console i = 7, actual sleep = 7ms
to console i = 8, actual sleep = 8ms
to console i = 9, actual sleep = 9ms
to console i = 10, actual sleep = 10ms
to console i = 11, actual sleep = 11ms
to console i = 12, actual sleep = 12ms
to console i = 13, actual sleep = 13ms
to console i = 14, actual sleep = 14ms
to console i = 15, actual sleep = 15ms
to console i = 16, actual sleep = 16ms
to console i = 17, actual sleep = 17ms
to console i = 18, actual sleep = 18ms
to console i = 19, actual sleep = 19ms
```

Chuck.

ducalex, 2019-07-12:
> So I assume you are using arduino-esp32 as a component with your own sdkconfig settings?
Correct, the default esp-idf settings.

> Let's say portTICK_PERIOD_MS = 10 and you call delay(9). Do you really want to block and spin for 9ms?

I do. That is how other arduino boards work and a lot of code and libraries expect delay to work as per [the documentation](https://www.arduino.cc/reference/en/language/functions/time/delay/).

That is also how esp-idf's own [usleep()](https://github.com/espressif/esp-idf/blob/master/components/newlib/time.c#L325) works.

As you pointed out delay works fine in Arduino IDE, if you consider Arduino as component to be unsupported then please ignore this issue!

Thanks

ducalex, 2019-07-12:
Yes, I can see it works fine in Arduino IDE now. As pointed out by @negativekelvin, it is only a problem when using arduino-esp32 as a component because freertos frequency is 100hz (by default).

stickbreaker, 2019-07-12:
I missed you were using arduino as a component.  I don't know all the rules when using arduino as a component.  I do know, I2C is not shared between `Wire()` and `idf`.

Chuck.

negativekelvin, 2019-07-12:
>I do. That is how other arduino boards work and a lot of code and libraries expect delay to work as per the documentation.
That is also how esp-idf's own usleep() works. 

Yes usleep, emphasis on the u.  It is not good design to use it for ms delays.  Is there a reason you don't want to set tick hz to 1000?

ducalex, 2019-07-12:
>  Is there a reason you don't want to set tick hz to 1000?

None, I didn't know that Arduino-esp32 IDE version wasn't using the default 100Hz.

I will change my project to 1000 but perhaps it could be worth mentioning this necessary step in the setup guide here https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md.

Thanks

romansavrulin, 2019-07-19:
@ducalex Please fix that doc to  and make a PR

ducalex, 2019-07-19:
Done. https://github.com/espressif/arduino-esp32/pull/3014

closed 2019-10-01 by stale[bot]