espressif/arduino-esp32 #2990: could not get host from dns

zuqualla, 2019-07-16:
Hi, 

I get 
`[E][WiFiUdp.cpp:170] beginPacket(): could not get host from dns: 5` 
and
`[E][WiFiUdp.cpp:170] beginPacket(): could not get host from dns: 11`

i guess the numbers at the end stand for:
'''
#define	EIO 5		/* I/O error */
#define	EAGAIN 11	/* No more processes */
'''
Anyone an idea what this means?

I only get this error with my router, if i go via hot spot on my phone i don't get it. So maybe it goes to wrong DNS server? How can i get the DNS server ip used by the STA when using 'WiFi.mode(WIFI_AP_STA);'?

thx!

zuqualla, 2019-07-16:
I check on my pc, and the DNS for the router is set the same as the default gateway. It does resolve DNS though.

zuqualla, 2019-07-16:
So simplified example hereafter, works on my hotspot, doesn't via my router:

´´´
#include <WiFi.h>
#include <netdb.h>

const char* ssid = "yournetwork";
const char* password = "yourpassword";

void setup() {

    Serial.begin(115200);
  
    // delete old config
    WiFi.disconnect(true);

    //start WIFI AP
    WiFi.mode(WIFI_AP_STA);  //Both hotspot and client are enabled
    IPAddress apIP(192, 168, 0, 1);
    String WifiAPName = "Air-Quality-Sensor-AQS12345678";
    WiFi.softAP("AQS12345678", "password"); 
    WiFi.softAPConfig(apIP, apIP, IPAddress(255, 255, 255, 0));   

    delay(1000);
    
    Serial.println("  Acces Point name: " + WifiAPName);
    Serial.print("  Acces Point IP address: ");
    Serial.println(WiFi.softAPIP()); // kernel panics if not printed seperatly.
    Serial.println("");
  
    // start wifi STA
    Serial.print("Connecting to local wifi network (SSID:AQS12345678):"); 

    WiFi.begin(ssid, password);
    float startTime = millis(); 
    while (WiFi.status() != WL_CONNECTED ) {
       if(millis()-startTime > 10000) // 10 s
       {
         WiFi.disconnect (true);
         Serial.println("\n  Cound't connect to local Wifi network. Check Wifi credentials and Wifi Network status.");
         break;
       } else {
        delay(250); Serial.print(".");
       }
     }

    if (WiFi.status() == WL_CONNECTED ){
      Serial.print("\n  Connected to wifi network: "); Serial.println(WiFi.SSID());
      Serial.print("  with IP address: "); Serial.println(WiFi.localIP());          
    } 
}

void loop() {

  Serial.println("------------");
  int i;
  struct hostent *he;
  struct in_addr **addr_list;
  struct in_addr addr;
  
  he = gethostbyname("airquality.addisabeba.info");
  if (he == NULL) {
      Serial.println("ERROR: hostname not resolved by DNS");
  } else {
      // print information about this host:
      printf("Official name is: %s\n", he->h_name);
      printf("IP address: %s\n", inet_ntoa(*(struct in_addr*)he->h_addr));
      printf("All addresses: ");
      addr_list = (struct in_addr **)he->h_addr_list;
      for(i = 0; addr_list[i] != NULL; i++) {
          printf("%s ", inet_ntoa(*addr_list[i]));
      }
      printf("\n");
  }
  delay(2000);
}
´´´

lbernstone, 2019-07-16:
```
#include <WiFi.h>
void setup() {
  Serial.begin(115200);
  WiFi.begin(); //Put ssid, passwd in here if needed
  WiFi.waitForConnectResult();
  Serial.println(WiFi.dnsIP());
  WiFi.config(WiFi.localIP(), WiFi.gatewayIP(), WiFi.subnetMask(), IPAddress(8,8,8,8)); 
  delay(10);
  Serial.println(WiFi.dnsIP());
}
void loop() {}
```

zuqualla, 2019-07-16:
@lbernstone, it works! thank you so much! 

I was digging too deep, trying with `inet_pton(AF_INET, "8.8.8.8", &dnsserver); dns_setserver(0, &dnsserver);` But that didn't work. 

If you pass in Addis Abeba, you get a beer from me! :) There's all kind of weird DNS blocking and MAC filtering over here. So changing it to 8.8.8.8 was really necessary.

closed 2019-07-16 by zuqualla