espressif/arduino-esp32 #2993: http update needs authentication

fabltd, 2019-07-16:
It appears there is no way to added authentication to http/https update?

Suggest an optional for a token / secret.

lbernstone, 2019-07-16:
The WebServer manages authentication.  Examples are included in the [library](https://github.com/espressif/arduino-esp32/tree/master/libraries/WebServer/examples).

fabltd, 2019-07-16:
This is for the t_httpUpdate_return ret = httpUpdate.update(client, server); method. There is no way to specify authentication.

collin80, 2019-07-18:
What kind of auth do you need? Basic where it sends in plaintext? Or digest perhaps? There are examples out there for doing auth but you're right, they don't seem to be integrated into the core library HTTPClient like you'd want/hope for them to be. In particular, digest is a pain but code does exist to do it externally of the HTTPClient library.

fabltd, 2019-07-24:
I was thinking JWT. Currently I have a work around by passing a JWT in the current version param. There should be an extra option when declaring the instance for example: String &authKey

atanisoft, 2019-08-02:
@fabaff feel free to submit a PR to add auth to the HTTPUpdate library..

closed 2019-10-15 by stale[bot]