espressif/arduino-esp32 #2994: ESP32 ASYNC WEB + SPIFFS + WIFI SOFTAP

zekageri, 2019-07-17:
### I have a few problems with my **_BARE ESP32 WROOM-32 chip_**, and i don't know if these are related to each other in some way.
 
First of all this is a bare esp chip with 16mb of flash. 
I use a custom partition table for it.
My first problem is in the partitions. If i use more than 1 or 1,5 mb SPIFFS partition, the flash is very very slow in all cases. The write, the read and everything. Now i have this partition table, and it is seems stable to me.

```
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x700000,
app1,     app,  ota_1,   0x710000,0x700000,
eeprom,   data, 0x99,    0xE10000,0x1000,
spiffs,   data, spiffs,  0xE11000,0xFADC0,
```
I have got some space that i don't use right now.
As you can see there is some nvs data and some otadata and eeprom space.
My app space is 7MB 2x because of the OTA.
And i have near 1MB of SPIFFS.
If i decrease both of the app to for example 5mb, then i got +4mb for my flash.
If i add it to the SPIFFS partition, my flash is slow. 
I'am using the default PlatformIO esptool.py for flashing my chip.

My second problem is maybe related to this.
When the chip is booting, everytime i get this message on my  serial monitor:

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
```
However it seems stable and every aspect of my flash is working if it is in the 1MB partition table.
I think my flash encryption is disabled. If it is not disabled by default then maybe doesnt. I dont know how to look deeper into this.
Iam mounting my SPIFFS on the setup with a 1000 delay after powerup:

```
void setup() {
  Serial.begin(115200);
  delay(1000);
  SPIFFS.begin() ? Serial.println("SPIFFS.OK") : Serial.println("SPIFFS.FAIL");
}
```

My third problem is the WiFI softap.
Iam using Ethernet connection by default with a LAN8720 like this:
[](https://www.waveshare.com/lan8720-eth-board.htm)
The ethernet connection is pretty cool and fast as it should be. I'am using it with a very old ethernet lib from Me_no_Dev back in the day where they created it and working for the first time.
So when i turn on my ap mode  in the loop like this:

```
  if(WifiOn == true){
    WiFi.setAutoConnect(false);
    WiFi.disconnect(true);
    WiFi.enableAP(true);
    WiFi.setSleep(false);
    WiFi.softAP(ssid,password, 10, 0, 1);
    Serial.println("Szerintem itt akad el! 1");
    tft.drawBitmap(120, 90, wifi_icon, 20, 15, WHITE);
    delay(15);
    WifiOn = false;
  }
```

Sometimes my esp is restarting with no error, sometimes its working. There is impossible to be a power issue because my board is powered by a big Power Supply. An EMG18144, and i have got capacitors everywhere. But that case is like one or two times from 20.
The real problem with the softAP is that its Wifi signal is sometimes weak A.F and sometimes it is good. I tried it on every channel. (Maybe because of the cables around the antenna?)
And it doesn't matter if the signal is good or bad, if i connect to it, it is going dead after some time if iam surfing the webpages. Sometimes it is restarting with this error:

```
dhcps: send_offer>>udp_sendto result 0
assertion "new_rcv_ann_wnd <= 0xffff" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/tcp.c", line 779, function: tcp_update_rcv_ann_wnd
abort() was called at PC 0x40102abb on core 0

Backtrace: 0x4008d86c:0x3ffb3d70 0x4008da9d:0x3ffb3d90 0x40102abb:0x3ffb3db0 0x401296da:0x3ffb3de0 0x40129769:0x3ffb3e00 0x401557dd:0x3ffb3e20 0x40126108:0x3ffb3e40 0x40089789:0x3ffb3e70
```

But sometimes with heap poisoning. ( I know it is a common problem with the async lib )
But the heap poisoning is never happening if iam on desktop and i investigate the webserver throught the ethernet interface. Just on mobile. But on mobile it is common on both ethernet and softAP.

My webpages are in the SPIFFS in separate compressed files. 
My biggest page size is 6KB.
And the pages pulling one biggerish JS and a CSS file. These are compressed to hell too.
I also tried it if these two are on the SPIFFS. Result is the same.

Maybe all these issues are coming from my wrong partition table and this flash read error, 1000 problem. In my original plan i wanted like 4 mb app space and the rest is for SPIFFS.

Here is some pictures from the bare esp chip setup:
I'am using a few things with it. Two I/O expander, two Digital Potentiometer, a Key matrix and an RTC module. All of these in one i2c bus. A display, an ethernet modul, an RFID reader, two little power supplyes for testing purposes and two supply delay chip because of the ethernet modul.

[image]
[image]

I will be very glad if someone can help me with something!
Thank you!

boarchuz, 2019-07-17:
I'd still have my money on a power supply issue. No matter how good your power supply is, dupont wires in a breadboard setup can often cause problems due to all the contact resistance and so forth going on there. Especially if those white wires with dodgy connections heading off-screen to the left are the power supply...
I've solved similar issues by simply doubling up the dupont wires on power connections before.
Do you have access to a scope?

zekageri, 2019-07-17:
Yap. I have a scope. I measured the voltages over the breadboard with a refined multimeter. The voltages seems to be good. One of my little power supply is broken. I have two of them because the RFID reader is an 5volt device. And from that little power supply i'am going to another to make 3v3 to every other thing. And the first little power supply is get the power from a big stable one. But the powers seems to be fine with a multimeter. :/

lbernstone, 2019-07-17:
I would agree with @boarchuz that the most likely problem is power.  Put a 470-1000 uF cap between the ground and 3v3 pins on the esp32 to help stabilize while breadboarding.  
As far as the SPIFFS slowness, I would stick to even numbers on the partition sizes.  The sector size is 0x1000, so any fractions of that is likely to cause some slowness just to get a few extra bytes.  Depending on what you are doing with the filesystem, FFat may be more appropriate than SPIFFS.
If you upgrade to arduino-esp32 v1.0.3, you will get back the sector from eeprom.  The repo version currently has some PRs in place that should help with tcp buffer issues.

zekageri, 2019-07-18:
Do you think that the SPIFFS error issue is because the power or anything else? There are capacitors on the ESP supply pin and on the display as well. These are two 16v470uF caps. 

The SPIFFS slowness is coming if i remove the eeprom partition and counting the spiffs from there. :/ I'am logging to the SPIFFS when there is tag detected by the RFID reader. I log the logins and the logouts. And also all the tag numbers are in a txt file on the SPIFFS and when a tag comes, i search that tag from the file, and if it is there i see if some tag is already logged in(this is a flag and not in spiffs) and if there aren't any tags logged in currently than log that tag in, and write that tag and the corresponding tag name wich is also in SPIFFS in a separate file to an other file to the SPIFFS as logged in tag with name and the time when this tag is logged in.

zekageri, 2019-07-18:
I have tree levels for the tags. And if a tag comes to one of the levels, i "save" it to the SPIFFS as level1 tag. All other tags that are level1 too i "save" them to the level1 tag txt to the SPIFFS as well. When the esp turns on, i read these tag numbers to separate variables for further processing. And if the read is succesfull. In the loop i wait for an interrupt from the RFID. If there are tag, i compare that tag from the RFID to all of the variables that i readed from the saved txt file. The "save to variable from spiffs" function on boot is looks like that:

```
static const inline void handleMentes() {
  if ( yo == true) {
    File f = SPIFFS.open(elsoszint, "r");
    if (!f) {
      Serial.println("Elsőszint.F");
    } else {
      int xCnt = 0;
      Lista = "";
      while ( xCnt < 1) {
        if (ment == 0) {
          f.seek(0, SeekCur);
          Lista = f.readStringUntil('\n');
          fu = 0;
          fi = 0;
          fa = 0;
          for (int i = 0; i < 5; i++) {
            if (fu == 0 && i == 0) {data4 = Lista.substring(0, 1).toInt();Lista.remove(0, 1);fu++;}
            if (fi == 0 && i == 1) {data4 = Lista.substring(0, 2).toInt();Lista.remove(0, 2);fi++;}
            if (fa <= 3 && i >= 2) {data4 = Lista.substring(0, 3).toInt();Lista.remove(0, 3);fa++;}
            level1[0][i] = data4;
          }
          ment++;

        } if (ment == 1) {
          fu = 0;
          fi = 0;
          fa = 0;
          f.seek(('\n' - 10), SeekCur);
          Lista = f.readStringUntil('\n');
          for (int i = 0; i < 5; i++) {
            if (fu == 0) {data4 = Lista.substring(0, 1).toInt();Lista.remove(0, 1);fu++;}
            if (fi == 0 && i == 1) {data4 = Lista.substring(0, 2).toInt();Lista.remove(0, 2);fi++;}
            if (fa <= 3 && i >= 2) {data4 = Lista.substring(0, 3).toInt();Lista.remove(0, 3);fa++;}
            level1[1][i] = data4;
          }
          ment++;
        } if (ment == 2) {
          fu = 0;
          fi = 0;
          fa = 0;
          f.seek(('\n' - 10), SeekCur);
          Lista = f.readStringUntil('\n');

          for (int i = 0; i < 5; i++) {
            if (fu == 0) {data4 = Lista.substring(0, 1).toInt();Lista.remove(0, 1);fu++;}
            if (fi == 0 && i == 1) {data4 = Lista.substring(0, 2).toInt();Lista.remove(0, 2);fi++;}
            if (fa <= 3 && i >= 2) {data4 = Lista.substring(0, 3).toInt();Lista.remove(0, 3);fa++;}
            level1[2][i] = data4;
          }
          ment++;
        } if (ment == 3) {
          fu = 0;
          fi = 0;
          fa = 0;
          f.seek(('\n' - 10), SeekCur);
          Lista = f.readStringUntil('\n');

          for (int i = 0; i < 5; i++) {
            if (fu == 0) {data4 = Lista.substring(0, 1).toInt();Lista.remove(0, 1);fu++;}
            if (fi == 0 && i == 1) {data4 = Lista.substring(0, 2).toInt();Lista.remove(0, 2);fi++;}
            if (fa <= 3 && i >= 2) {data4 = Lista.substring(0, 3).toInt();Lista.remove(0, 3);fa++;}
            level1[3][i] = data4;
          }
          ment++;
        } if (ment == 4) {
          fu = 0;
          fi = 0;
[... 140 lines cut ...]
          if (!f) {Serial.println("Levelegy.F");} 
          else {
            f.print(ujtag + '\n');
            f.print("<br>");
          }
          hol++;
          f.close();
          File k = SPIFFS.open("/egyes.txt", "w");
          if (!k) {
            Serial.println("Egyes.F");
          } else {
            k.println("Van egyes!");
          } k.close();
          if (SPIFFS.exists("/formatComplete.txt")) {
            SPIFFS.remove("/formatComplete.txt");
          }
          yo = true;
          handleMentes();
        }
      }
  }
}
```

As you can see i watch if the tag is "egyesjott". if it is "egyesjott" then i have to save its name from a user input and the tag number from the rfid reader. After that i open "elsoszint" wich is "first level" ( that holds the names for the first level tags) and after that i open "levelegy" wich means "levelone" (that holds the tag numbers). Both of the functions are slow if i have more than one MB SPIFFS partition, even if i don't have eeprom.

atanisoft, 2019-08-02:
@zekageri The reason your SPIFFS access is slow likely is due to the way that the esp32 handles caching of flash memory. Only 4MB can be mapped at one time, see section 3.2 of the WROOM-32D spec sheet (https://www.espressif.com/sites/default/files/documentation/esp32-wroom-32d_esp32-wroom-32u_datasheet_en.pdf).

SPIFFS uses *ONE* large blob in flash for the entire filesystem and every access to it will require reading the entire file up to the point of the file you are trying to access. if the SPIFFS partition crosses a 4MB barrier it will be *VERY* slow as it pages in the segments that cross the 4MB barrier.

I'd suggest switching to FFat instead of SPIFFS as it doesn't suffer the same monolithic blob storage problem as it has a central directory block with pointers to where file data is stored. It should speed up the performance considerably, you can try this out with the default_ffat partition.

zekageri, 2019-10-02:
> @zekageri The reason your SPIFFS access is slow likely is due to the way that the esp32 handles caching of flash memory. Only 4MB can be mapped at one time, see section 3.2 of the WROOM-32D spec [...]

I did tried the FFAT example. 
It did work properly and reasonably fast, but the thing is i really really need to upload my files to the filesystem via web and/or via an easy plugin from PIO or some IDE.

Nevermind, i close this now because these are almost obvious issues. 
Thank you for replying!

closed 2019-10-02 by zekageri