espressif/arduino-esp32 #2995: assertion "don't call tcp_recved for listen-pcbs"

MkLHX, 2019-07-17:
### Hardware:
Board:							ESP32 Dev Module/MH-ET live esp32 mini kit 
Core Installation version:			        1.9.0
IDE name:						Platform.io home 2.2/core 4.0
Flash Frequency:					80Mhz
PSRAM enabled:                                       no
Upload Speed:					        921600
Computer OS:                                          Windows 10

### Description:
Since Espressif 1.90, PIO core 4.0 and Python3.7 updates i have some troubles with AsyncTCP connection.(i'm on windows 10). Before, everythings works perfectly.

I'm using 1 ESP32 board (mhetesp32minikit) as server and 2 ESP32 (mhetesp32minikit and lolin32) as clients.

Every 3sec, clients sent json data on the server through wifi. when data are received server make some treatments and then he send json data with new values.
My debug level is on 5 (verbose).
```
[main]... Firmware version : 0.0.1
[EEPROM MANAGER] ... EepromVirginState : 127
[EEPROM MANAGER] ... EEPROM is not virgin
[MAC Address] ... 80:7D:3A:B7:72:74
[MDNS]... start : water-sensor-274.local
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 13 - AP_START
[Wifi AP]... started
[Wifi AP]... ssid : water-sensor-274[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 3 - STA_STOP
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 3 - STA_STOP
[Wifi AP]... ip : 192.168.4.1
[Wifi check credentials]... start
[Wifi check credentials]... stored ssid : hub-67c/EOL
[Wifi check credentials]... stored password : greenponik/EOL
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
[Wifi STA]... trying to connect to : hub-67c

.-.-.-.-[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:379] _eventCallback(): STA IP: 192.168.4.3, MASK: 255.255.255.0, GW: 192.168.4.1

[Wifi STA...  connected to : hub-67c
[Wifi STA]... ip : 192.168.4.3
####----Booting - Setup end----####
[txCommunication]... data sent to 192.168.4.1
[getWaterTemperature]... temperature : 24.31
[ECRead]... ecVoltage: 430.00
[pH Voltage]... phVoltage: 11.00
[readPH]... phValue 15.71
[txCommunication]... data received from 192.168.4.1
[txCommunication]... data sent to 192.168.4.1
[ECRead]... ecVoltage: 430.00
[getWaterTemperature]... temperature : 24.37
[pH Voltage]... phVoltage: 11.00
[readPH]... phValue 15.71
[getWaterTemperature]... temperature : 24.37
[txCommunication]... data received from 192.168.4.1
[txCommunication]... data sent to 192.168.4.1
[getWaterTemperature]... temperature : 24.37
[ECRead]... ecVoltage: 429.00
[getWaterTemperature]... temperature : 24.37
[pH Voltage]... phVoltage: 11.00
[readPH]... phValue 15.71
[getWaterTemperature]... temperature : 24.37
[txCommunication]... data received from 192.168.4.1
assertion "don't call tcp_recved for listen-pcbs" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/tcp.c", line 803, function: tcp_recvedabort() was called at PC 0x400f0e07 on core 0

Backtrace: 0x4008d28c:0x3ffaf1d0 0x4008d4bd:0x3ffaf1f0 0x400f0e07:0x3ffaf210 0x40114d2a:0x3ffaf240 0x400d9a09:0x3ffaf260 0x40111794:0x3ffaf280 0x40089051:0x3ffaf2b0

Rebooting...
```
the backtrace decoder give me: 
```
Decoding stack results
0x4008d28c: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 155
0x4008d4bd: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 170
0x400f0e07: __assert_func at ../../../.././newlib/libc/stdlib/assert.c line 63
0x40114d2a: tcp_recved at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/tcp.c line 802
0x40111794: tcpip_thread at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/api/tcpip.c line 124
0x40089051: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143
```

i cannot put the sketch here.

lbernstone, 2019-07-18:
AsyncTCP has had updates recently to address buffer issues.  Please post issues with 3rd party libraries on the [library's issues board](https://github.com/me-no-dev/AsyncTCP/issues?q=is%3Aissue+is%3Aopen+sort%3Aupdated-desc).
The lwip stack is compiled into the [sdk libraries](https://github.com/espressif/arduino-esp32/tree/master/tools/sdk/lib) included in arduino-esp32.  The [source](https://github.com/espressif/esp-lwip/blob/3ed39f27981e7738c0a454f9e83b8e5164b7078b/src/core/tcp.c) is part of the esp-idf (nee lwip).  If you want to recompile it yourself, [the docs](https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md) explain how that can be done.

MkLHX, 2019-07-18:
Hi @lbernstone , sorry if i posted the issue in bad thread but with this error i cannot say if its provided by sdk or third party lib, i just need help to understand what's happened on my TCP connection who works fine last few days ago

closed 2019-09-30 by stale[bot]

valkyrie54, 2023-04-21:
Hi, I was just testing my debugger and stepped into some esp32 system code and got the following message, which seems to be looking for files from a user called ficeto on his desktop from google play for arduiostudio app with android system not windows.

DaeMonSxy, 2023-10-24:
did anyone solved the tcp_update_rcv_ann_wnd issue? Im fighting with that last weeks...