espressif/arduino-esp32 #2996: Cannot compile under windows with latest IDF update

luc-github, 2019-07-17:
With latest idf update under windows :
https://github.com/espressif/arduino-esp32/pull/2992
cannot compile anything anymore, the error with wificlient.ino example

```
Linking everything together...
"C:\\Users\\user\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc" -nostdlib "-LC:\\Users\\user\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/lib" "-LC:\\Users\\user\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/ld" -T esp32_out.ld -T esp32.common.ld -T esp32.rom.ld -T esp32.peripherals.ld -T esp32.rom.libgcc.ld -T esp32.rom.spiram_incompatible_fns.ld -u ld_include_panic_highint_hdl -u call_user_start_cpu0 -Wl,--gc-sections -Wl,-static -Wl,--undefined=uxTopUsedPriority -u __cxa_guard_dummy -u __cxx_fatal_exception -Wl,--start-group "C:\\Users\\user\\AppData\\Local\\Temp\\arduino_build_726089\\sketch\\WiFiClient.ino.cpp.o" "C:\\Users\\user\\AppData\\Local\\Temp\\arduino_build_726089\\libraries\\WiFi\\ETH.cpp.o" "C:\\Users\\user\\AppData\\Local\\Temp\\arduino_build_726089\\libraries\\WiFi\\WiFi.cpp.o" "C:\\Users\\user\\AppData\\Local\\Temp\\arduino_build_726089\\libraries\\WiFi\\WiFiAP.cpp.o" "C:\\Users\\user\\AppData\\Local\\Temp\\arduino_build_726089\\libraries\\WiFi\\WiFiClient.cpp.o" "C:\\Users\\user\\AppData\\Local\\Temp\\arduino_build_726089\\libraries\\WiFi\\WiFiGeneric.cpp.o" "C:\\Users\\user\\AppData\\Local\\Temp\\arduino_build_726089\\libraries\\WiFi\\WiFiMulti.cpp.o" "C:\\Users\\user\\AppData\\Local\\Temp\\arduino_build_726089\\libraries\\WiFi\\WiFiSTA.cpp.o" "C:\\Users\\user\\AppData\\Local\\Temp\\arduino_build_726089\\libraries\\WiFi\\WiFiScan.cpp.o" "C:\\Users\\user\\AppData\\Local\\Temp\\arduino_build_726089\\libraries\\WiFi\\WiFiServer.cpp.o" "C:\\Users\\user\\AppData\\Local\\Temp\\arduino_build_726089\\libraries\\WiFi\\WiFiUdp.cpp.o" "C:\\Users\\user\\AppData\\Local\\Temp\\arduino_build_726089\\core\\core.a" -lgcc -lopenssl -lbtdm_app -lfatfs -lwps -lcoexist -lwear_levelling -lesp_http_client -lprotobuf-c -lhal -lnewlib -ldriver -lbootloader_support -lpp -lfreemodbus -lmesh -lsmartconfig -ljsmn -lwpa -lethernet -lphy -lesp-face -lapp_trace -lconsole -lulp -lwpa_supplicant -lfreertos -lbt -lmicro-ecc -lesp32-camera -lcxx -lxtensa-debug-module -ltcp_transport -lmdns -lvfs -lesp_ringbuf -lsoc -lcore -lfb_gfx -lsdmmc -llibsodium -lcoap -ltcpip_adapter -lprotocomm -lesp_event -limage_util -lc_nano -lesp-tls -lasio -lrtc -lspi_flash -lwpa2 -lwifi_provisioning -lesp32 -lface_recognition -lapp_update -lnghttp -lspiffs -lface_detection -lespnow -lnvs_flash -lesp_adc_cal -llog -ldl_lib -lsmartconfig_ack -lexpat -lm -lfr -lmqtt -lc -lheap -lmbedtls -llwip -lnet80211 -lesp_http_server -lpthread -ljson -lesp_https_ota -lfd -lstdc++ -Wl,--end-group -Wl,-EL -o "C:\\Users\\user\\AppData\\Local\\Temp\\arduino_build_726089/WiFiClient.ino.elf"
c:/users/user/documents/arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe:C:\Users\user\Documents\Arduino\hardware\espressif\esp32/tools/sdk/lib\libesp-face.a: file format not recognized; treating as linker script

c:/users/user/documents/arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe:C:\Users\user\Documents\Arduino\hardware\espressif\esp32/tools/sdk/lib\libesp-face.a:1: syntax error

collect2.exe: error: ld returned 1 exit status
```

No issue under Linux, I have updated tools with get.exe but same error

bdring, 2019-07-17:
I can confirm the same problem is occurring for me. I can go back to a previous commit and it works. 

BTW: When I did get.exe on the latest commit, it did not appear to "get" anything.

I use Windows 10 and Arduino IDE 1.8.9

luc-github, 2019-07-17:
fixed by https://github.com/espressif/arduino-esp32/commit/548f712df2487e5a8790001e2f088a426324a04f

Thank you @me-no-dev  ^_^

closed 2019-07-17 by luc-github