espressif/arduino-esp32 #3000: [E][vfs_api.cpp:238] VFSFileImpl(): fopen(/spiffs/relay_status1.html) failed

GeorgeFlorian, 2019-07-18:
OS: Linux Mint 19.1
Board: ESP32-EVB
Platform = https://github.com/platformio/platform-espressif32.git

Hello !

Today I was stress testing the ESPAsyncWebServer from my project and I got the following error for the first time ever.

```
[E][vfs_api.cpp:238] VFSFileImpl(): fopen(/spiffs/relay_status1.html) failed
[E][vfs_api.cpp:238] VFSFileImpl(): fopen(/spiffs/relay_status1.html) failed
[E][vfs_api.cpp:238] VFSFileImpl(): fopen(/spiffs/relay_status1.html) failed
```

Followed by:
```
[W][AsyncTCP.cpp:883] _poll(): pcb is NULL
[W][AsyncTCP.cpp:883] _poll(): pcb is NULL
[W][AsyncTCP.cpp:883] _poll(): pcb is NULL
[W][AsyncTCP.cpp:883] _poll(): pcb is NULL
[W][AsyncTCP.cpp:896] _poll(): ack timeout 4
```
I've added an `ESP.restart()` to the `ack timeout 4` so the ESP restarted afterwards.

I got the above error while opening a web-page. Nothing crazy. Just `http://IP_ADDR/home` which is represented by `index.html`.

`relay_status1.html` only has this : `%PLACEHOLDER_Status1%`.

I am using a script inside `index.html` to constantly refresh the respective placeholder:
```
            <div class = "text_box_STA">
                <p>%PLACEHOLDER_Relay1%: <span class = "relay_status1">%PLACEHOLDER_Status1%</span></p>
                <p>Timer 1: <span>%PLACEHOLDER_Timer1%</span>(s)</p>
                <p>%PLACEHOLDER_Relay2%: <span class = "relay_status2">%PLACEHOLDER_Status2%</span></p>
                <p>Timer 2: <span>%PLACEHOLDER_Timer2%</span>(s)</p>
                <p class = "auth">/ON URL for %PLACEHOLDER_Relay1% : <span>http://%PH_Auth%%PH_IP_Addr%/relay1/on</span></p>                
                <p class = "auth">/OFF URL for %PLACEHOLDER_Relay1% : <span>http://%PH_Auth%%PH_IP_Addr%/relay1/off</span></p>
                <p class = "auth">/ON URL for %PLACEHOLDER_Relay2% : <span>http://%PH_Auth%%PH_IP_Addr%/relay2/on</span></p>
                <p class = "auth">/OFF URL for %PLACEHOLDER_Relay2% : <span>http://%PH_Auth%%PH_IP_Addr%/relay2/off</span></p>
            </div>
<script>
    $(document).ready( function(){
    $('.text_box').load('events_placeholder.html');
    $('.relay_status1').load('relay_status1.html');
    $('.relay_status2').load('relay_status2.html');
    refresh();
    });
    
    function refresh(){
        setTimeout( function() {
            $('.text_box').load('events_placeholder.html');
            $('.relay_status1').load('relay_status1.html');
            $('.relay_status2').load('relay_status2.html');
            refresh();
        }, 1000);
    }
</script>
```

This is the handler I am using to GET the `/home` page:
```
      server.on("/home", HTTP_GET, [](AsyncWebServerRequest* request){
        if(userFlag) {
          if(!request->authenticate(value_login[0].c_str(), value_login[1].c_str()))
            return request->requestAuthentication(NULL,false);
          // request->send(SPIFFS, "/index.html", "text/html", false, processor);
            AsyncWebServerResponse *response = request->beginResponse(SPIFFS, "/index.html", "text/html", false, processor);
            File file = SPIFFS.open("/index.html");
            response->addHeader("Content-Length", (String) file.size ());
            request->send(response);
            file.close();
        } else {          
          // request->send(SPIFFS, "/index.html", "text/html", false, processor);
          AsyncWebServerResponse *response = request->beginResponse(SPIFFS, "/index.html", "text/html", false, processor);
          File file = SPIFFS.open("/index.html");
          response->addHeader("Content-Length", (String) file.size ());
          request->send(response);
          file.close();
        }        
      });
```

And the processor method is straightforward:
```
```

lbernstone, 2019-07-18:
One of the parameters for SPIFFS.begin() is maxOpenFiles.  This is 10 by default.  If you will serve a large number of clients regularly, you can increase this.  IIRC the buffers will use about 2K for each additional client.

GeorgeFlorian, 2019-07-18:
> One of the parameters for SPIFFS.begin() is maxOpenFiles. This is 10 by default. If you will serve a large number of clients regularly, you can increase this.

This should be done like this ?
```
SPIFFS.begin(true, "/", 20)
```

> IIRC the buffers will use about 2K for each additional client.

I didn't fully understand this.
If I serve 20 pages how will those 2K be calculated ?

lbernstone, 2019-07-18:
I know the documentation here sucks, but the [headers](https://github.com/espressif/arduino-esp32/blob/master/libraries/SPIFFS/src/SPIFFS.h#L26) are a form of documentation.
default is:
`SPIFFS.begin(false,"/spiffs",10);`
I meant 2k per file lock, so you need an additional 20k to bump it to 20 locks.

GeorgeFlorian, 2019-07-19:
> I know the documentation here sucks, but the [headers](https://github.com/espressif/arduino-esp32/blob/master/libraries/SPIFFS/src/SPIFFS.h#L26) are a form of documentation.
> default is: [...]

Ok. Thanks !
I will experiment with this.

closed 2019-07-19 by GeorgeFlorian

mrWheel, 2020-05-30:
I have the same issue stress testing a simple webserver!
Changing maxOpenFiles for SPIFFS does not help!

name=AsyncTCP version=1.1.1
name=ESP Async WebServer version=1.2.3
<pre>
[12:10:12][ 212932|113792] restAPI     (  29): from[192.168.12.20] URI[/api/v0/devinfo] method[GET] 
[12:10:15][ 212932|113792] restAPI     (  29): from[192.168.12.20] URI[/api/v0/devinfo] method[GET] 
[E][vfs_api.cpp:238] VFSFileImpl(): fopen(/spiffs/index.js) failed
[E][vfs_api.cpp:238] VFSFileImpl(): fopen(/spiffs/FSexplorer.png) failed
[E][vfs_api.cpp:238] VFSFileImpl(): fopen(/spiffs/index.css) failed
[E][vfs_api.cpp:238] VFSFileImpl(): fopen(/spiffs/settings.png) failed
[W][AsyncTCP.cpp:949] _poll(): pcb is NULL
[W][AsyncTCP.cpp:949] _poll(): pcb is NULL
[12:10:17][ 212932|113792] restAPI     (  29): from[192.168.12.20] URI[/api/v0/devinfo] method[GET] 
[12:10:20][ 212932|113792] restAPI     (  29): from[192.168.12.20] URI[/api/v0/devinfo] method[GET] 
</pre>