espressif/arduino-esp32 #3001: Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Chedeshri, 2019-07-18:
@chegewara I am trying to create BLE client using ESP32  
IDE - Visual studio code (Platform.io)
Client is getting created but when I try to read characteristic value I am  getting following error
### Hardware:
Board:							ESP32 
Core Installation version:			1.0.1
IDE name:						Visual studio code (Platform.io)
Flash Frequency:					40Mhz
PSRAM enabled:                                    no
Upload Speed:					115200
Computer OS:                                        Mac OS 
Error : - 
 - Created client
 - Connected to server
 - Found our service
[E][BLERemoteCharacteristic.cpp:280] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[E][BLERemoteCharacteristic.cpp:280] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[E][BLERemoteCharacteristic.cpp:280] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[E][BLERemoteCharacteristic.cpp:280] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[E][BLERemoteCharacteristic.cpp:280] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[E][BLERemoteCharacteristic.cpp:280] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
 - Found our characteristic
The characteristic value was: 
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4008dfba  PS      : 0x00060b30  A0      : 0x800d6bdc  A1      : 0x3ffca210  
A2      : 0x10008000  A3      : 0x00000000  A4      : 0xffffffff  A5      : 0x00000000  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x80186e9b  A9      : 0x3ffca220  
A10     : 0x3ffca2a8  A11     : 0x3f402512  A12     : 0x00000005  A13     : 0x0000ff00  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x10008040  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0xffffffff  

Backtrace: 0x4008dfba:0x3ffca210 0x400d6bd9:0x3ffca250 0x400d549e:0x3ffca290 0x400d5a1a:0x3ffca2e0 0x400d1c3e:0x3ffca320 0x400d86a1:0x3ffca390 0x40090461:0x3ffca3b0

Rebooting...
ets Jun  8 2016 00:22:57

Thanks in anticipation!

chegewara, 2019-07-18:
https://github.com/me-no-dev/EspExceptionDecoder

wakwak-koba, 2019-07-19:
I use BLEClient at the core and find many bugs.
Would you like to try my Fork ?
https://github.com/wakwak-koba/arduino-esp32

- many bugs fixed.
- multi connection supports.
- reconnect after disconnection supports.

example:
https://github.com/wakwak-koba/arduino-esp32/tree/master/libraries/BLE/examples/BLE_clients_catch_notifies

Chedeshri, 2019-07-19:
> https://github.com/me-no-dev/EspExceptionDecoder

Decoding stack results
0x40091658: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 155
0x40091889: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 170
0x401494c7: __assert_func at ../../../.././newlib/libc/stdlib/assert.c line 63
0x4011e447: transmit_data at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/hci/hci_hal_h4.c line 145
0x400db3e9: transmit_fragment at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/hci/hci_layer.c line 356
0x400dc251: fragment_and_dispatch at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/hci/packet_fragmenter.c line 81
0x400db627: event_packet_ready at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/hci/hci_layer.c line 347
0x400dccd7: fixed_queue_process at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/osi/fixed_queue.c line 289
0x400db553: hci_host_thread_handler at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/hci/hci_layer.c line 238
0x40090155: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143

chegewara, 2019-07-19:
@Chedeshri Its strange backtrace, because its pointing straight to esp-idf bluedroid library. 
If you could use BLE library from this repository, maybe it will fix your issue. There is few bugfixes that should help with stability.

I dont know if changes are merged to master only or to some new sub-version, maybe @me-no-dev can help with this.

Chedeshri, 2019-07-19:
> @Chedeshri Its strange backtrace, because its pointing straight to esp-idf bluedroid library.
> If you could use BLE library from this repository, maybe it will fix your issue. There is few bugfixes [...]

Thank you for your feed back: 
I tried same code on visual studio code, I am pasting the decode messege hope it gives some leads : 

PC: 0x4008dfba: xQueueGenericReceive at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/queue.c line 1571
EXCVADDR: 0x10008040

Decoding stack results
0x4008dfba: xQueueGenericReceive at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/queue.c line 1571
0x400d6bd9: FreeRTOS::Semaphore::give() at /Users/shrianthchede/Documents/Arduino/hardware/espressif/esp32/libraries/BLE/src/FreeRTOS.cpp line 123
0x400d549e: BLEScan::BLEScan() at /Users/shrianthchede/Documents/Arduino/hardware/espressif/esp32/libraries/BLE/src/BLEScan.cpp line 30
0x400d5a1a: BLEScan::handleGAPEvent(esp_gap_ble_cb_event_t, esp_ble_gap_cb_param_t*) at /Users/shrianthchede/Documents/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/bits/stl_function.h line 387
0x400d1c3e: setup() at /Users/shrianthchede/Documents/Arduino/BLE_client/BLE_client.ino line 125
0x400d86a1: _Unwind_RaiseException at /Volumes/build/idf/crosstool-NG/.build/src/gcc-5.2.0/libgcc/unwind.inc line 99
0x40090461: uxPortCompareSetExtram at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/portmux_impl.inc.h line 106

chegewara, 2019-07-19:
Its really hard to say. That decoded stack result leads to the end of ble scan. So, if you could paste setup function from your code or at least the part you are performing ble init and scan that may be helpful.

wakwak-koba, 2019-07-20:
chegewara, Do you remember the following exchanges?
https://github.com/nkolban/esp32-snippets/issues/765

I think it is just the same problem as at that time.
The current version does not reflect the PR at the time.

chegewara, 2019-07-20:
@wakwak-koba You are correct, in arduino-esp32 library this bug has not been fixed:
https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/src/BLERemoteService.cpp#L167
and seems that it is similar in this code:
https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/src/BLERemoteCharacteristic.cpp#L259

Im not sure it is related, but @Chedeshri could you change both values to 1 and test again? Thanks

Chedeshri, 2019-07-22:
> @wakwak-koba You are correct, in arduino-esp32 library this bug has not been fixed:
> https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/src/BLERemoteService.cpp#L167
> and seems that [...]

@chegewara I had already kept it to '1' - 
while(true) {
		uint16_t count = 1;
		esp_gatt_status_t status = ::esp_ble_gattc_get_all_descr(
			getRemoteService()->getClient()->getGattcIf(),
			getRemoteService()->getClient()->getConnId(),
			getHandle(),
			&result,
			&count,
			offset
		);

wakwak-koba, 2019-07-22:
It was not the latest 1.0.2 of arduino-esp32, but it was 1.0.1 one before !?

Chedeshri, 2019-07-22:
> Its really hard to say. That decoded stack result leads to the end of ble scan. So, if you could paste setup function from your code or at least the part you are performing ble init and scan that may be helpful.

[text.txt](https://github.com/espressif/arduino-esp32/files/3416282/text.txt)

wakwak-koba, 2019-07-22:
Once connection succeeds, does reconnection fail?

At the time of MyClientCallback.onDisconnect (), it seems that it is too early to 'BLEDevice :: getScan ()-> start (0)' in the middle of disconnection process.

Chedeshri, 2019-07-22:
> Once connection succeeds, does reconnection fail?
> 
> At the time of MyClientCallback.onDisconnect (), it seems that it is too early to 'BLEDevice :: getScan ()-> start (0)' in the middle of disconnection process.

Yes reconnection fails after sucessfull connection! 
So do you suggest to add delay before the next scan ?

Chedeshri, 2019-07-22:
> Once connection succeeds, does reconnection fail?
> 
> At the time of MyClientCallback.onDisconnect (), it seems that it is too early to 'BLEDevice :: getScan ()-> start (0)' in the middle of disconnection process.

Yes it esp resets during reconnection!

chegewara, 2019-07-22:
Maybe here is the problem:
```
    if(pRemoteCharacteristic->canRead()) {
      std::string value = pRemoteCharacteristic->readValue();
      Serial.print("The characteristic value was: ");
      Serial.println(value.c_str());
    }
```
Is it possible that characteristic value is not set and value length is 0? Im just guessing now, because i dont know how this line will work in such case, but it looks like it crashed exactly here:
`Serial.println(value.c_str());`

Chedeshri, 2019-07-23:
> Maybe here is the problem:
>
> ```
>     if(pRemoteCharacteristic->canRead()) {
>       std::string value = pRemoteCharacteristic->readValue();
>       Serial.print("The characteristic value was: "); [...]

The sensor is throwing values because I can see them on nrf nordic connect, however I also tried debugging `Serial.println(value.c_str());` the error still persist without the read and write  characteristic value. 
My application is simple I have an device (server) which has some measurements and I am trying to configuring esp32 as bluetooth client, connection between measuring device and esp32 is successful but could not read the value.

bertdd, 2019-08-30:
I am getting LoadProhibited issues when reading data from 4 Voltcraft SEM6000 Power meters. The crashes happen randomly, and are less frequent when I add a couple of milliseconds of delay() between the measurements. Another variant is that the program just "hangs".
Could this be related to what wakwak_koba is describing? 
Would it be wiser to revert to the espressif BLE api instead of the Arduino one? 
Are all the calls blocking? , even with callbacks? 
What would be best practice when it comes to reading multiple values quickly from a multitude of BLE servers? The individual functions work just fine when just executed once. 
I could not really find any good examples with "heavy workload" for BLE.. Ideas anyone ?

H3ndrik-Jan, 2019-11-11:
> Once connection succeeds, does reconnection fail?

Hi, I encounter about the same problem with my ESP32 wroom 32 board. I get the LoadProhibited error and the InstrFetchProhibited error alternately whenever I try to do a new http request. The first http request is always succesfull though, so the ESP doesn't reset until a new http request is called.

I am still not quiet sure about the reason for this, and what I should do to fix this. Does anyone have an answer for me?

I am using the arduino IDE with arduino-esp32 version 1.0.4.

lbernstone, 2019-11-11:
Please do not hijack threads.  Your issue is unrelated.  Open a new issue and follow the template.

closed 2020-01-25 by stale[bot]