espressif/arduino-esp32 #3002: Not able to enable the flash encryption in development mode

chhotu, 2019-07-18:
Note : I have refer the steps as per ESP flash encryption(https://docs.espressif.com/projects/esp-idf/en/latest/security/flash-encryption.html)
################Enabling Flash Encryption :: 1st test case################
I have build flash_encryption example which is provided in esp-idf where, enabled only flash encryption in Development Mode.

Build and flash the complete image including bootloader, partition table, and app. These partitions are initially written to the flash unencrypted.
I have flash the esp using 'make -j4 flash monitor' command 

I can easily identify from the logs that the flash is encrypted.

Sample program to check Flash Encryption
This is ESP32 chip with 2 CPU cores, WiFi/BT/BLE, silicon revision 1, 4MB external flash
Flash encryption feature is enabled
Flash encryption mode is DEVELOPMENT
Flash in encrypted mode with flash_crypt_cnt = 1
Halting...

And also tried the following scenarios 
1)update the modified plaintext application image to flash in an encrypted format
	idf.py encrypted-app-flash monitor
2)all partitions need to be updated in an encrypted format
	idf.py encrypted-flash monitor

Up-to this stage everything is working as expected.

################Disabling Flash Encryption :: 2nd test case################
As I have enabled flash encryption in Development mode I can disable flash encryption again by writing FLASH_CRYPT_CNT eFuse (only in Development mode):

As per the steps of disabling flash Encryption I have burn the efuse using below command.

$ python espefuse.py -b 115200 -p COM150 burn_efuse FLASH_CRYPT_CNT
espefuse.py v2.7-dev
Connecting.....
Burning efuse FLASH_CRYPT_CNT (Flash encryption mode counter) 0x1 -> 0x3. This is an irreversible operation.
Type 'BURN' (all capitals) to continue.
BURN

After disabling flash encryption I have build and flash the esp and it will give output as expected
ESP logs after disabling the flash encryption

Sample program to check Flash Encryption
This is ESP32 chip with 2 CPU cores, WiFi/BT/BLE, silicon revision 1, 4MB external flash
FLASH_CRYPT_CNT eFuse value is 3
Flash encryption feature is disabled

Halting...

################Enabling Flash Encryption :: 3rd test case################

Now I want to enable flash encryption again so for that I have followed the same steps as I had done in a 1st test case

I'm getting following things to continue

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

lbernstone, 2019-07-18:
https://www.merriam-webster.com/dictionary/irreversible

chhotu, 2019-07-19:
> https://www.merriam-webster.com/dictionary/irreversible

Is it any possible way to enable the flash encryption again?

closed 2019-10-01 by stale[bot]