espressif/arduino-esp32 #3005: Potential bug in WiFi library for ESP32

Muller59, 2019-07-18:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.2
IDE name:						Arduino IDE 1.8.9
Flash Frequency:					80Mhz
PSRAM enabled:                                    no
Upload Speed:				      921600
Computer OS:                                        Windows 7

### Description:
Greetings,

I am trying to work with the WiFi.SoftAP class in arduino on an ESP32.

the problem lies in the IP address for the softIP access point.
i am requesting 192.168.**2**.1 via WiFi.softAPConfig()
- IPAddress ap_local_IP(192, 168, **2**, 1);
- IPAddress ap_gateway(192, 168, 1, 254);
- IPAddress ap_subnet(255, 255, 255, 0);

WiFi.softAPConfig(ap_local_IP, ap_gateway, ap_subnet);  is returning 192.168.**4**.1 as the IP address

the program and output below illustrates that behavior

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

/*
   Program: ESSP32-Wifi-Test
   Purpose: test wifi connectvity for esp32
*/

#include <WiFi.h>

#define DEBUG true      // Set to false to suppress debug output on serial

// Define placeholders for ssid and password. The contents are irrelevant, will be filled from /config.txt
char* ssid     = (char *) "YourWiFiNetworkName";      // The SSID (name) of the Wi-Fi network
char* password = (char *) "YourWifiPassword";         // The password of the Wi-Fi network

// Configuration Variables for the SoftAP name and IP.
const char* ap_ssid = "CaptivePortal";
const char* ap_password = "qwerty123";
IPAddress ap_local_IP(192, 168, 2, 1);
IPAddress ap_gateway(192, 168, 1, 254);
IPAddress ap_subnet(255, 255, 255, 0);

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);//Starting serial comunication
  Serial.println();

  WiFi.mode(WIFI_OFF);        //Prevents reconnection issue (taking too long to connect)
  delay(1000);
  WiFi.mode(WIFI_STA);

  WiFi.begin(ssid, password);

  Serial.print("Connecting to ");
  Serial.print(ssid);
  Serial.println(" ...");

  int i = 0;
  while (WiFi.status() != WL_CONNECTED  && i < 20) { // Wait 20 seconds for the Wi-Fi to connect
    delay(1000);  // wait 1 sec
    Serial.print(++i);
    Serial.print(' ');
  }
  Serial.println(' ');

  if (WiFi.status() == WL_CONNECTED ) {
    Serial.print("Local IP address = ");
    Serial.println(WiFi.localIP());
    Serial.print("Local Gateway address= ");
    Serial.println(WiFi.gatewayIP());
    Serial.print("Local DNS server address = ");
    Serial.println(WiFi.dnsIP(0));
    Serial.print("Local Subnet mask = ");
    Serial.println(WiFi.subnetMask());
  }

  // Spin up a local AP if not connected.
  if (WiFi.status() != WL_CONNECTED ) {
    WiFi.mode(WIFI_AP);
    Serial.println("");
    Serial.println("Configuring access point...");
    Serial.println("Setting soft-AP configuration ... ");
    WiFi.softAPConfig(ap_local_IP, ap_gateway, ap_subnet);
    if (DEBUG) Serial.println(WiFi.softAPIP());  //@@@

    Serial.println("Setting soft-AP ... ");
    WiFi.softAP(ap_ssid, ap_password);
    Serial.print("Soft-AP name = ");
    Serial.println(ap_ssid);
    Serial.print("Soft-AP password = ");
    Serial.println(ap_password);
    Serial.print("Soft-AP IP address = ");
    Serial.println(WiFi.softAPIP());
  }
}

void loop() {
  // put your main code here, to run repeatedly:

}

```

### Debug Messages:
```
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:9232
load:0x40080400,len:6400
entry 0x400806a8
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:9232
load:0x40080400,len:6400
entry 0x400806a8

[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 14 - AP_STOP
Connecting to YourWiFiNetworkName ...
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
1 2 [D][WiFiGeneric.cpp:336] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:351] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
3 4 [D][WiFiGeneric.cpp:336] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:351] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
5 6 7 [D][WiFiGeneric.cpp:336] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:351] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
8 9 [D][WiFiGeneric.cpp:336] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:351] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
10 11 12 [D][WiFiGeneric.cpp:336] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:351] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
13 14 [D][WiFiGeneric.cpp:336] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:351] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
15 16 [D][WiFiGeneric.cpp:336] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:351] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
17 18 19 [D][WiFiGeneric.cpp:336] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:351] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
20  

Configuring access point...
Setting soft-AP configuration ... 
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 13 - AP_START
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 13 - AP_START
192.168.4.1
Setting soft-AP ... 
Soft-AP name = CaptivePortal
Soft-AP password = qwerty123
Soft-AP IP address = 192.168.4.1
```

lbernstone, 2019-07-18:
I am unable to reproduce.  A minimal sketch gives me the right address (even if I put in your invalid gateway).
```
#include <WiFi.h>
void setup() {
  Serial.begin(115200);
  IPAddress apip(192,168,2,1);
  IPAddress apgw(192,168,1,254);
  IPAddress apsub(255,255,255,0);
  if(WiFi.softAPConfig(apip,apgw,apsub))
    Serial.println("config success");
  WiFi.softAP("esp_test", "testing12345");
  Serial.println(WiFi.softAPIP());
}

void loop() {}
```

Muller59, 2019-07-19:
thanks for your rely, much appreciated.
you are right, your code runs as advertised, and sorry about the class C netmask, that was supposed to be class B. 

So, i did some further testing on my code above.
i found that if I comment out the offending code block trying to connect to the Wifi network (the piece below) my code also gives the correct AP IP address. 

As soon as that code runs, the error returns. I think (and i am often wrong) there may bee some corruption going on in the wifi stack.

I did find a workaround. 
If i use  WiFi.mode(WIFI_OFF);   delay(1000);  before   WiFi.mode(WIFI_AP); the code runs as expected.

offending code block
``  WiFi.mode(WIFI_STA);

  WiFi.begin(ssid, password);

  Serial.print("Connecting to ");
  Serial.print(ssid);
  Serial.println(" ...");

  int i = 0;
  while (WiFi.status() != WL_CONNECTED  && i < 20) { // Wait 20 seconds for the Wi-Fi to connect
    delay(1000);  // wait 1 sec
    Serial.print(++i);
    Serial.print(' ');
  }
  Serial.println(' ');

  if (WiFi.status() == WL_CONNECTED ) {
    Serial.print("Local IP address = ");
    Serial.println(WiFi.localIP());
    Serial.print("Local Gateway address= ");
    Serial.println(WiFi.gatewayIP());
    Serial.print("Local DNS server address = ");
    Serial.println(WiFi.dnsIP(0));
    Serial.print("Local Subnet mask = ");
    Serial.println(WiFi.subnetMask());
  }``

again, thanks for your time,

regards, 
Rob

closed 2019-10-01 by stale[bot]