espressif/arduino-esp32 #3008: Cannot Connect to WiFi

kmmaran, 2019-07-19:
### Hardware:
Board:							Wemos Lolin and Custom Board
Core Installation version:			Not sure on this. My apology.
IDE name:						Arduino IDE
Flash Frequency:					40MHz (Crystal Freq)
PSRAM enabled:                                     Not sure on this. My apology.
Upload Speed:					921600
Computer OS:                                        Windows 10

### Description:
Hi Experts,
I am RF/Hardware Engineer trying to get into embedded world, my apology for any error on below request. 
I am trying to connect my board to WiFi access point. I have built my own custom board, everything working well, but the WiFi can't connect to access point. I tried on Wemos Lolin Light and seems to be working on same code. I have checked the hardware, connect up to spectrum analyser and board is transmitting around same RF level but can't connect to access point. I have attached here my board and Wemos board details from monitor. Also checked, clock, voltages, decoupling. 

The different from serial messages from custom board and Wemos is 
rtcwdt_rtc_reset not appearing in custom board.

### Sketch:  
```cpp

//Change the code below by your sketch
#include "WiFi.h"
 
const char* ssid = "Not Shown";
const char* password =  "Not Shown";

void setup() {
 
  Serial.begin(115200);
 
  WiFi.begin(ssid, password);
 
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.println("Connecting to WiFi..");
  }
 
  Serial.println("Connected to the WiFi network");
 
}
 
void loop() {}
```

### Debug Messages:
```
**Message from custom board**

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:8424
ho 0 tail 12 room 4
load:0x40080400,len:5868
entry 0x4008069c
Connecting to WiFi..
Connecting to WiFi..
Connecting to WiFi..

**Message from Wemos board**

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:8424
ho 0 tail 12 room 4
load:0x40080400,len:5868
entry 0x4008069c
Connecting to WiFi..
Connecting to WiFi..
Connected to the WiFi network

```
Thank you for your help.

lbernstone, 2019-07-20:
Crashes during WiFi init are typically power related.  Make sure you provide a stable voltage, and can supply at least 0.5A during the boot sequence. I typically put a 4.7uF capacitor b/w 3v3 and gnd.

kmmaran, 2019-07-23:
@ibernstone thank you for you reply. I found the problem, is related to crystal. Crystal able to produce close 40MHz but lower amplitude because the pad not soldered properly.

closed 2019-07-23 by kmmaran