espressif/arduino-esp32 #3011: Disconnect all existing connections in multirole BLE mode (server and client at the same time) when only one device disconnected

MegaGarynych, 2019-07-19:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			       1.0.3-rc1
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
Computer OS:                                        Windows 10

### Description:
I have combined two examples BLE_client and BLE_server in one sketch to have ability connect to several BLE devices and control my esp32 board via smartphone application. All works excellent. Smartphone can connect to esp32, read/write characteristics and get notifications. Also, esp32 can connect to several BLE devices (like BLE button) and read/write characteristics and get notifications. But if one device is disconnect (smartphone or only one BLE button) all existed connections was disconnected too (all BLE buttons and smartphone). I think it is wrong behavior. Only one connection should be break, all other should be continue keep.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp
#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEServer.h>

#define SERVICE_UUID        "4fafc201-1fb5-459e-8fcc-c5c9c331914b"
#define CHARACTERISTIC_UUID "beb5483e-36e1-4688-b7f5-ea07361b26a8"

// The remote service we wish to connect to.
//static BLEUUID serviceUUID("4fafc201-1fb5-459e-8fcc-c5c9c331914b");
static BLEUUID serviceUUID("a3aa");
static BLEUUID charUUID("d368");
// The characteristic of the remote service we are interested in.
//static BLEUUID    charUUID("beb5483e-36e1-4688-b7f5-ea07361b26a8");

static boolean doConnect = false;
static BLERemoteCharacteristic* pRemoteCharacteristic;
static BLEAdvertisedDevice* myDevice;

uint8_t mac[6] = {0xXX, 0x81, 0xF9, 0xXX, 0xA3, 0xXX};

BLECharacteristic *pCharacteristic;
 

class MyClientCallback : public BLEClientCallbacks {
  void onConnect(BLEClient* pclient) {
    Serial.println("onСonnect");
  }

  void onDisconnect(BLEClient* pclient) {
    Serial.println("onDisconnect");
  }
};

class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
 /**
   * Called for each advertising BLE server.
   */
  void onResult(BLEAdvertisedDevice advertisedDevice) {
    Serial.print("BLE Advertised Device found: ");
    Serial.println(advertisedDevice.toString().c_str());

    // We have found a device, let us now see if it contains the service we are looking for.
    uint8_t* addr = (uint8_t*)advertisedDevice.getAddress().getNative(); 
    if (memcmp(addr, mac, 6) == 0) {
      Serial.print("This is our device: ");
      BLEDevice::getScan()->stop();
      myDevice = new BLEAdvertisedDevice(advertisedDevice);
      doConnect = true;
    } // Found our server
  } // onResult
}; // MyAdvertisedDeviceCallbacks

bool connectToServer() {
    Serial.print("Forming a connection to ");
    Serial.println(myDevice->getAddress().toString().c_str());
    
    BLEClient*  pClient  = BLEDevice::createClient();
    Serial.println(" - Created client");
[... 262 lines cut ...]
[D][BLEDevice.cpp:562] getAdvertising(): get advertising
[D][BLEAdvertising.cpp:491] handleGAPEvent(): handleGAPEvent [event no: 1]
[D][BLEDevice.cpp:108] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEDevice.cpp:562] getAdvertising(): get advertising
[D][BLEAdvertising.cpp:189] start(): - advertising service: 4fafc201-1fb5-459e-8fcc-c5c9c33191
[D][BLEDevice.cpp:154] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 5] ... Unknown
[D][BLEDevice.cpp:154] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 5] ... Unknown
[D][BLEClient.cpp:458] handleGAPEvent(): BLEClient ... handling GAP event!
[D][BLEDevice.cpp:562] getAdvertising(): get advertising
[D][BLEAdvertising.cpp:491] handleGAPEvent(): handleGAPEvent [event no: 0]
[D][BLEClient.cpp:458] handleGAPEvent(): BLEClient ... handling GAP event!
[D][BLEDevice.cpp:562] getAdvertising(): get advertising
[D][BLEAdvertising.cpp:491] handleGAPEvent(): handleGAPEvent [event no: 1]
[D][BLEClient.cpp:458] handleGAPEvent(): BLEClient ... handling GAP event!
[D][BLEDevice.cpp:562] getAdvertising(): get advertising
[D][BLEAdvertising.cpp:491] handleGAPEvent(): handleGAPEvent [event no: 6]
[D][FreeRTOS.cpp:157] take(): Semaphore taking: name: SetValue (0x3ffddb24), owner: <N/A> for <Unknown>
[D][FreeRTOS.cpp:166] take(): Semaphore taken:  name: SetValue (0x3ffddb24), owner: <Unknown>
[D][FreeRTOS.cpp:157] take(): Semaphore taking: name: SetValue (0x3ffddb24), owner: <N/A> for <Unknown>
[D][FreeRTOS.cpp:166] take(): Semaphore taken:  name: SetValue (0x3ffddb24), owner: <Unknown>
[D][FreeRTOS.cpp:157] take(): Semaphore taking: name: SetValue (0x3ffddb24), owner: <N/A> for <Unknown>
[D][FreeRTOS.cpp:166] take(): Semaphore taken:  name: SetValue (0x3ffddb24), owner: <Unknown>
[D][FreeRTOS.cpp:157] take(): Semaphore taking: name: SetValue (0x3ffddb24), owner: <N/A> for <Unknown>
[D][FreeRTOS.cpp:166] take(): Semaphore taken:  name: SetValue (0x3ffddb24), owner: <Unknown>
```

chegewara, 2019-07-19:
Hi,
from logs we can see that only 1 device is disconnected:
```
onDisconnect
[I][BLEDevice.cpp:610] removePeerDevice(): remove: 1, GATT role client
```
Could you add custom GAP and GATT handlers to see full logs from esp-idf library during disconnecting?
Here is example:
https://github.com/espressif/arduino-esp32/issues/2401#issuecomment-458777266

MegaGarynych, 2019-07-20:
> 
> 
> Hi,
> from logs we can see that only 1 device is disconnected:
> 
> ```
> onDisconnect
> [I][BLEDevice.cpp:610] removePeerDevice(): remove: 1, GATT role client
> ```
> 
Yes :) but it is wrong Disconnecting event. I am only disconnect smartphone application, but BLE buttons disconnected too.
I have change some sketch parts and found strange behavior, please see attachements.
I have separate client and server callbacks.

> Could you add custom GAP and GATT handlers to see full logs from esp-idf library during disconnecting?
> Here is example:
> [#2401 (comment)](https://github.com/espressif/arduino-esp32/issues/2401#issuecomment-458777266)
Ok, thanks I will test soon.

```cpp
#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEServer.h>

#define SERVICE_UUID        "4fafc201-1fb5-459e-8fcc-c5c9c331914b"
#define CHARACTERISTIC_UUID "beb5483e-36e1-4688-b7f5-ea07361b26a8"

// The remote service we wish to connect to.
//static BLEUUID serviceUUID("4fafc201-1fb5-459e-8fcc-c5c9c331914b");
static BLEUUID serviceUUID("a3aa");
static BLEUUID charUUID("d368");
// The characteristic of the remote service we are interested in.
//static BLEUUID    charUUID("beb5483e-36e1-4688-b7f5-ea07361b26a8");

static boolean doConnect = false;
static BLERemoteCharacteristic* pRemoteCharacteristic;
static BLEAdvertisedDevice* myDevice;

uint8_t mac[6] = {0x00, 0x81, 0xF9, 0x60, 0xA3, 0x08};

BLECharacteristic *pCharacteristic;
 
class MyServerCallback: public BLEServerCallbacks {
    void onConnect(BLEServer* pServer) {
      Serial.println("MyServerCallback.onConnect");
    };

    void onDisconnect(BLEServer* pServer) {
      Serial.println("MyServerCallback.onDisconnect");
    }
};

class MyClientCallback : public BLEClientCallbacks {
  void onConnect(BLEClient* pclient) {
    Serial.println("MyClientCallback.onСonnect");
  }

  void onDisconnect(BLEClient* pclient) {
    Serial.println("MyClientCallback.onDisconnect");
  }
};

class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
 /**
   * Called for each advertising BLE server.
   */
  void onResult(BLEAdvertisedDevice advertisedDevice) {
    Serial.print("BLE Advertised Device found: ");
    Serial.println(advertisedDevice.toString().c_str());

[... 507 lines cut ...]
[D][BLEDevice.cpp:562] getAdvertising(): get advertising
[D][BLEAdvertising.cpp:491] handleGAPEvent(): handleGAPEvent [event no: 0]
[D][BLEClient.cpp:458] handleGAPEvent(): BLEClient ... handling GAP event!
[D][BLEDevice.cpp:562] getAdvertising(): get advertising
[D][BLEAdvertising.cpp:491] handleGAPEvent(): handleGAPEvent [event no: 1]
[D][BLEClient.cpp:458] handleGAPEvent(): BLEClient ... handling GAP event!
[D][BLEDevice.cpp:562] getAdvertising(): get advertising
[D][BLEAdvertising.cpp:491] handleGAPEvent(): handleGAPEvent [event no: 6]
[D][FreeRTOS.cpp:157] take(): Semaphore taking: name: SetValue (0x3ffddb1c), owner: <N/A> for <Unknown>
[D][FreeRTOS.cpp:166] take(): Semaphore taken:  name: SetValue (0x3ffddb1c), owner: <Unknown>
[D][FreeRTOS.cpp:157] take(): Semaphore taking: name: SetValue (0x3ffddb1c), owner: <N/A> for <Unknown>
[D][FreeRTOS.cpp:166] take(): Semaphore taken:  name: SetValue (0x3ffddb1c), owner: <Unknown>
[D][FreeRTOS.cpp:157] take(): Semaphore taking: name: SetValue (0x3ffddb1c), owner: <N/A> for <Unknown>
[D][FreeRTOS.cpp:166] take(): Semaphore taken:  name: SetValue (0x3ffddb1c), owner: <Unknown>
[D][FreeRTOS.cpp:157] take(): Semaphore taking: name: SetValue (0x3ffddb1c), owner: <N/A> for <Unknown>
[D][FreeRTOS.cpp:166] take(): Semaphore taken:  name: SetValue (0x3ffddb1c), owner: <Unknown>
[D][FreeRTOS.cpp:157] take(): Semaphore taking: name: SetValue (0x3ffddb1c), owner: <N/A> for <Unknown>
[D][FreeRTOS.cpp:166] take(): Semaphore taken:  name: SetValue (0x3ffddb1c), owner: <Unknown>
```
But In all cases my behavior is the same
1. Esp32 start and automatically connect to  BLE button
2. Write to BLE button characteristic
2. Then I am connect to esp32 via smartphone   
3. Read several times esp32 characteristic
4. Disconnect smartphone from esp32

MegaGarynych, 2019-07-20:
I have add custom GAP and GATT handlers to see full logs 

My testings steps is the same in my previous messages

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:8424
ho 0 tail 12 room 4
load:0x40080400,len:5868
entry 0x4008069c
Starting BLE work!
[D][FreeRTOS.cpp:157] take(): Semaphore taking: name: RegisterAppEvt (0x3ffdd160), owner: <N/A> for registerApp
[D][FreeRTOS.cpp:166] take(): Semaphore taken:  name: RegisterAppEvt (0x3ffdd160), owner: registerApp
[D][BLEDevice.cpp:108] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... Unknown
[W][ble_multirole_test.ino:113] my_gatts_event_handler(): custom gatts event handler, event: 0
[D][FreeRTOS.cpp:157] take(): Semaphore taking: name: CreateEvt (0x3ffdd3b8), owner: <N/A> for createService
[D][FreeRTOS.cpp:166] take(): Semaphore taken:  name: CreateEvt (0x3ffdd3b8), owner: createService
[D][FreeRTOS.cpp:157] take(): Semaphore taking: name: CreateEvt (0x3ffdd61c), owner: <N/A> for executeCreate
[D][FreeRTOS.cpp:166] take(): Semaphore taken:  name: CreateEvt (0x3ffdd61c), owner: executeCreate
[D][BLEDevice.cpp:108] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... Unknown
[W][ble_multirole_test.ino:113] my_gatts_event_handler(): custom gatts event handler, event: 7
[D][BLEService.cpp:225] addCharacteristic(): Adding characteristic: uuid=beb5483e-36e1-4688-b7f5-ea07361b26 to service: UUID: 4fafc201-1fb5-459e-8fcc-c5c9c33191, handle: 0x0028
[D][FreeRTOS.cpp:157] take(): Semaphore taking: name: SetValue (0x3ffddfe8), owner: <N/A> for <Unknown>
[D][FreeRTOS.cpp:166] take(): Semaphore taken:  name: SetValue (0x3ffddfe8), owner: <Unknown>
[D][BLECharacteristic.cpp:89] executeCreate(): Registering characteristic (esp_ble_gatts_add_char): uuid: beb5483e-36e1-4688-b7f5-ea07361b26, service: UUID: 4fafc201-1fb5-459e-8fcc-c5c9c33191, handle: 0x0028
[D][FreeRTOS.cpp:157] take(): Semaphore taking: name: CreateEvt (0x3ffddb1c), owner: <N/A> for executeCreate
[D][FreeRTOS.cpp:166] take(): Semaphore taken:  name: CreateEvt (0x3ffddb1c), owner: executeCreate
[D][BLEDevice.cpp:108] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... Unknown
[W][ble_multirole_test.ino:113] my_gatts_event_handler(): custom gatts event handler, event: 9
[D][FreeRTOS.cpp:157] take(): Semaphore taking: name: StartEvt (0x3ffdd6dc), owner: <N/A> for start
[D][FreeRTOS.cpp:166] take(): Semaphore taken:  name: StartEvt (0x3ffdd6dc), owner: start
[D][BLEDevice.cpp:108] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... Unknown
[W][ble_multirole_test.ino:113] my_gatts_event_handler(): custom gatts event handler, event: 12
[I][BLEDevice.cpp:560] getAdvertising(): create advertising
[D][BLEDevice.cpp:562] getAdvertising(): get advertising
[D][BLEDevice.cpp:562] getAdvertising(): get advertising
[D][BLEAdvertising.cpp:189] start(): - advertising service: 4fafc201-1fb5-459e-8fcc-c5c9c33191
Characteristic defined! Now you can read it in your phone!
[D][FreeRTOS.cpp:157] take(): Semaphore taking: name: ScanEnd (0x3ffde2b4), owner: <N/A> for start
[D][BLEDevice.cpp:562] getAdvertising(): get advertising
[D][FreeRTOS.cpp:166] take(): Semaphore taken:  name: ScanEnd (0x3ffde2b4), owner: start
[D][BLEAdvertising.cpp:491] handleGAPEvent(): handleGAPEvent [event no: 0]
[W][ble_multirole_test.ino:105] my_gap_event_handler(): custom gap event handler, event: 0
[D][BLEDevice.cpp:562] getAdvertising(): get advertising
[D][BLEAdvertising.cpp:491] handleGAPEvent(): handleGAPEvent [event no: 1]
[W][ble_multirole_test.ino:105] my_gap_event_handler(): custom gap event handler, event: 1
[D][BLEDevice.cpp:562] getAdvertising(): get advertising
[D][BLEAdvertising.cpp:491] handleGAPEvent(): handleGAPEvent [event no: 6]
[W][ble_multirole_test.ino:105] my_gap_event_handler(): custom gap event handler, event: 6
[D][BLEDevice.cpp:562] getAdvertising(): get advertising
[D][BLEAdvertising.cpp:491] handleGAPEvent(): handleGAPEvent [event no: 2]
[W][ble_multirole_test.ino:105] my_gap_event_handler(): custom gap event handler, event: 2
[D][BLEDevice.cpp:562] getAdvertising(): get advertising
[D][BLEAdvertising.cpp:491] handleGAPEvent(): handleGAPEvent [event no: 7]
[W][ble_multirole_test.ino:105] my_gap_event_handler(): custom gap event handler, event: 7
[D][BLEAdvertisedDevice.cpp:418] setRSSI(): - setRSSI(): rssi: -19
[D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0x01 (), length: 1, data: 06
[D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0x09 (), length: 20, data: 424c452044656d6f204c6967687426537065616b
[D][BLEAdvertisedDevice.cpp:407] setName(): - setName(): name: BLE Demo Light&Speak
BLE Advertised Device found: Name: BLE Demo Light&Speak, Address: 00:81:f9:60:a3:08
This is our device: [D][BLEDevice.cpp:562] getAdvertising(): get advertising
[D][FreeRTOS.cpp:157] take(): Semaphore taking: name: SetValue (0x3ffddfe8), owner: <N/A> for <Unknown>
[... 139 lines cut ...]
[D][BLEDevice.cpp:562] getAdvertising(): get advertising
[D][BLEAdvertising.cpp:189] start(): - advertising service: 4fafc201-1fb5-459e-8fcc-c5c9c33191
[W][ble_multirole_test.ino:113] my_gatts_event_handler(): custom gatts event handler, event: 15
[... cut ...]
[D][BLEAdvertising.cpp:491] handleGAPEvent(): handleGAPEvent [event no: 6]
[W][ble_multirole_test.ino:105] my_gap_event_handler(): custom gap event handler, event: 6
[D][FreeRTOS.cpp:157] take(): Semaphore taking: name: SetValue (0x3ffddfe8), owner: <N/A> for <Unknown>
[D][FreeRTOS.cpp:166] take(): Semaphore taken:  name: SetValue (0x3ffddfe8), owner: <Unknown>
[D][FreeRTOS.cpp:157] take(): Semaphore taking: name: SetValue (0x3ffddfe8), owner: <N/A> for <Unknown>
[D][FreeRTOS.cpp:166] take(): Semaphore taken:  name: SetValue (0x3ffddfe8), owner: <Unknown>

```

closed 2019-10-02 by stale[bot]

AndreasHobby, 2021-04-23:
I  have the same issue with the latest version. All BLE devices are disconnecting when one is loosing the connection (e.g. if I switch the device off or bringing it outside the range)