espressif/arduino-esp32 #3012: ESP32-EYE CameraWebserver- Controlling the stream on/off from my program NOT from webserver

bmillier, 2019-07-19:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do.
Describe your system( Hardware, computer, O/S, core version, environment)
Describe what is failing
Show the shortest possible code that will duplicate the error
Show the EXACT error message(it doesn't work is not enough)
Then if someone is interested and knowledgeable you might get a answer.  All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ESP32 Dev Module? ?node32? ?ttgo_lora?
Core Installation/update date:			?11/jul/2017?
IDE name:							?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
PSRAM enabled:                                          ?no?
Upload Speed:						?115200?
Computer OS:                                            ?Windows 10? ?Mac OSX? ?Ubuntu?

### Description:
Describe your problem here

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

bmillier, 2019-07-19:
This question is the same as issue 2674, but I couldn't see a solution there. 
I want to be able to start stop the camera stream from a button press on the ESP32 EYE- instead of the webserver  START STREAM button. I can see where all of the other controls sliders are handled in the "app_httpd.cpp" file, but can't see any code that handles either the "start/stop stream" or "start still" buttons. The button ID is "toggle-stream", but I find no reference to that in the ESP32 C code. What file contains the ESP32 C code that turns the stream on/off, and what does the function look like?
  Once I start a stream from the web client, I can turn it on/off by adding code in the "stream_handler" function (in app_httpd.c)  to basically skip over all of the code in this function, when I want it to stop. But, I can't start the stream this way.
  I don't know enough about any code that might be running on the client side, to know how the web server could be controlling the stream, apart from its sending a command of some sort.
Thanks in advance for any help

waldorffsmama, 2019-09-26:
Try this: 

http://[your_cams_ipaddress]:81/stream

bmillier, 2019-09-26:
@waldorffsmama- Thanks for the response. I wanted to start the stream from a trigger to the ESP32-EYE board itself- i.e. via a switch or motion detector  connected to a GPIO pin. I assume this web request would work from another computer- but are you saying that I could also issue it from the ESP32-EYE itself, and have it respond to a request that it basically is sending out itself? I've not tried doing this before, but if that is what you meant, I can give it a try.

waldorffsmama, 2019-09-27:
@bmillier, maybe I did not understand correctly what you want to do. You want to show the stream with the ESP32-EYE dashboard in a browser tab? So if a switch connected to the ESP32-EYE is pressed, a tab on your browser opens and displays the dashboard and starts a stream? Or do you assume you always have that tab open on your browser and the gpio-button just takes over your click on the stream-button in your browser?
What if the ESP-EYE dashboard is not open on your browser? What if it is not in focus? And also, the ESP-EYE allows only one client. Those are things that need to be sorted out on your machine but where the ESP32 coding has little influence.

bmillier, 2019-09-27:
@waldorffsmama : Yes that is basically correct. I don't want it to be streaming all of the time- just if a motion sensor (for example) triggers a GPIO pin. Then the ESP32-EYE would send an email message or notification of some sort to the user, as well as start the stream. The user would then open the browser, or alternately, a client on a PC could record the stream, once it started coming in. Of course, the user can start the stream manually in the web-browser, but I really wanted to figure out how to do this at the ESP32-EYE end.
  I've seen where the ESP-EYE only allows one client, by trying to look at the stream on both my phone and PC at the same time, but this isn't a big issue for me.
  I find it strange that the Arduino example program "CameraWebServer" contains all the code needed to send the various user-adjustable camera parameters from incoming web requests to the camera chip itself. But, the stream on/off command must be buried in the "compressed" web server code that is contained in the camera_index.h file. That is the part I haven't figured out yet.
Thank you.

waldorffsmama, 2019-09-28:
What you are looking for is a cam-server. Stream constantly to your PC and record on movement only or execute commands or call webhooks. Consider Contacam.

closed 2019-12-11 by stale[bot]