espressif/arduino-esp32 #3017: OTA dual core question

atlask1, 2019-07-20:
Hi,
I have a question: what happen if I permorm OTA and I'm using dual core or the new sketch from OTA use dual core? 
OTA need one core free? 
thanks

lbernstone, 2019-07-20:
OTA uses cpu cycles from whatever transport mechanism you are using (wifi, bluetooth, uart, etc).  The OTA firmware update process itself is simply an MD5 checksum, and marking the new firmware as active.  It behaves essentially the same with single core or dual.

atlask1, 2019-07-20:
Ok, thanks .
I’ve tested now. It’s ok . 
But it’s possible also via BLE?

atanisoft, 2019-08-02:
@atlask1 OTA over BLE is possible. One example which uses ESP-IDF APIs: https://github.com/chegewara/esp32-ota-with-ble-setup. There are very likely other examples to be found via GitHub search...

closed 2019-10-10 by atlask1