espressif/arduino-esp32 #3020: HTTP_UPDATE_FAILED Error (-104): Wrong HTTP Code

DaveCalaway, 2019-07-21:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.2
IDE name:						Arduino IDE and Platform.io
Flash Frequency:					80Mhz
PSRAM enabled:                                      no
Upload Speed:					115200
Computer OS:                                          Mac OSX

### Description: 
My test repository link: 
https://github.com/DaveCalaway/ESP8266-32/blob/master/Arduino/ESP32/OTA/OTA_git/Blink.ino.esp32.bin

After that post https://is.gd/hFXBtI i edited the link to the report like this:

http://raw.githubusercontent.com/DaveCalaway/ESP8266-32/master/Arduino/ESP32/OTA/OTA_git/Blink.ino.esp32.bin

but the problem remain.

### Sketch:  
```
/**
   httpUpdate.ino

    Created on: 27.11.2015

*/

#include <Arduino.h>

#include <WiFi.h>
#include <WiFiMulti.h>

#include <HTTPClient.h>
#include <HTTPUpdate.h>

#define LED_BUILTIN 2
WiFiMulti WiFiMulti;

void setup() {

  Serial.begin(115200);
  Serial.setDebugOutput(true);

  Serial.println();
  Serial.println();
  Serial.println();

  for (uint8_t t = 4; t > 0; t--) {
    Serial.printf("[SETUP] WAIT %d...\n", t);
    Serial.flush();
    delay(1000);
  }

  WiFi.mode(WIFI_STA);
  WiFiMulti.addAP("aaa", "bbb");

}

void loop() {
  // wait for WiFi connection
  if ((WiFiMulti.run() == WL_CONNECTED)) {

    WiFiClient client;

    // The line below is optional. It can be used to blink the LED on the board during flashing
    // The LED will be on during download of one buffer of data from the network. The LED will
    // be off during writing that buffer to flash
    // On a good connection the LED should flash regularly. On a bad connection the LED will be
    // on much longer than it will be off. Other pins than LED_BUILTIN may be used. The second
    // value is used to put the LED on. If the LED is on with HIGH, that value should be passed
    httpUpdate.setLedPin(LED_BUILTIN, LOW);

    t_httpUpdate_return ret = httpUpdate.update(client, "http://raw.githubusercontent.com/DaveCalaway/ESP8266-32/master/Arduino/ESP32/OTA/OTA_git/Blink.ino.esp32.bin");
    // Or:
    //t_httpUpdate_return ret = httpUpdate.update(client, "server", 80, "file.bin");

    switch (ret) {
      case HTTP_UPDATE_FAILED:
        Serial.printf("HTTP_UPDATE_FAILED Error (%d): %s\n", httpUpdate.getLastError(), httpUpdate.getLastErrorString().c_str());
        break;

      case HTTP_UPDATE_NO_UPDATES:
        Serial.println("HTTP_UPDATE_NO_UPDATES");
        break;

      case HTTP_UPDATE_OK:
        Serial.println("HTTP_UPDATE_OK");
        break;
    }
  }
}

```
Debug:
```
[D][HTTPClient.cpp:265] beginInternal(): host: raw.githubusercontent.com port: 80 url: /DaveCalaway/ESP8266-32/master/Arduino/ESP32/OTA/OTA_git/Blink.ino.esp32.bin
[D][HTTPClient.cpp:1002] connect():  connected to raw.githubusercontent.com:80
[D][HTTPClient.cpp:1127] handleHeaderResponse(): code: 301
[D][HTTPUpdate.cpp:224] handleUpdate(): Header read fin.

[D][HTTPUpdate.cpp:225] handleUpdate(): Server header:

[D][HTTPUpdate.cpp:226] handleUpdate():  - code: 301

[D][HTTPUpdate.cpp:227] handleUpdate():  - len: 0

[D][HTTPUpdate.cpp:233] handleUpdate(): ESP32 info:

[D][HTTPUpdate.cpp:234] handleUpdate():  - free Space: 1310720

[D][HTTPUpdate.cpp:235] handleUpdate():  - current Sketch Size: 739456

[E][HTTPUpdate.cpp:358] handleUpdate(): HTTP Code is (301)

[D][WiFiClient.cpp:482] connected(): Disconnected: RES: 0, ERR: 128
[D][HTTPClient.cpp:369] disconnect(): tcp is closed

HTTP_UPDATE_FAILED Error (-104): Wrong HTTP Code 
```

stickbreaker, 2019-07-21:
https://en.wikipedia.org/wiki/HTTP_301

Chuck.

DaveCalaway, 2019-07-21:
> https://en.wikipedia.org/wiki/HTTP_301
> 
> Chuck.

The wiki says: "The 301 redirect is considered a best practice for upgrading users from HTTP to HTTPS."

So i MUST use the HTTPS example? When can i use the HTTP?

stickbreaker, 2019-07-21:
The server you are connecting to refuses to use the method you requested.  If you want to actually retrieve the file you want, you have to use it's methods.

>The wiki says: "The 301 redirect is considered a best practice for upgrading users from HTTP to HTTPS."

Actually look at the `Location` field in the returned header.  In the Wiki example:
>HTTP/1.1 301 Moved Permanently
Location: http://www.example.org/index.asp

It tells you both the location and the protocol.  In this example: `HTTP` and `www.example.org/index.asp`.  a 301 does not always mean upgrade to `HTTPS`.

Chuck.

lbernstone, 2019-07-21:
You can set up a http server on your workstation to test unencrypted OTA.
https://appdividend.com/2019/02/06/python-simplehttpserver-tutorial-with-example-http-request-handler/

DaveCalaway, 2019-07-21:
@stickbreaker thx you, now i understand the problem.
I have not experience in HTTP, so i do not know how to reproduce the problem.
The Igor solution https://is.gd/7aiSNN doesn't works.

I'm very curious to know if i using the HTTPS with GitHub, i don't fall in the redirect problem.

@lbernstone good solution for a simple test. I will try this solution, with a simple upload button https://is.gd/n43zLT .

atanisoft, 2019-08-02:
@DaveCalaway GitHub only serves content via https, any http request will be "upgraded" to https via a 301 http redirect. You might be able to just replace http with https and have it work.

DaveCalaway, 2019-08-03:
If I use the "Chrome Developer Tools", f12 button, I can see the directly lick to the file.

I use the https example, with ce certification of the page that gave me the error 301.
But the code gave me an error about the ce certification.

Is it normal? I supposed that the ce certification is univocal of the github web site.

atanisoft, 2019-08-03:
@DaveCalaway switch from WifiClient to WifiClientSecure and it should be able to process the certs. You may need to configure the WifiClientSecure for CA certs to validate etc. I'm not certain there is an automated way to validate/ignore certs.

DaveCalaway, 2019-08-04:
@atanisoft yes i did it.
Now if i run the [example](https://is.gd/54d3ah) with HTTPS secure, with the direct link to my file:
https://raw.githubusercontent.com/DaveCalaway/ESP8266-32/master/Arduino/ESP32/OTA/Blink.ino.esp32.bin

and the .crt file from the github.com: https://ibb.co/yF2z6pN

Now i copied it to the Arduino sketch, using the same struct for the [example](https://is.gd/SxDsrg):
```
const char *rootCACertificate = \
"-----BEGIN CERTIFICATE-----\n" \
"MIIEtjCCA56gAwIBAgIQDHmpRLCMEZUgkmFf4msdgzANBgkqhkiG9w0BAQsFADBs\n" \
"MQswCQYDVQQGEwJVUzEVMBMGA1UEChMMRGlnaUNlcnQgSW5jMRkwFwYDVQQLExB3\n" \
"d3cuZGlnaWNlcnQuY29tMSswKQYDVQQDEyJEaWdpQ2VydCBIaWdoIEFzc3VyYW5j\n" \
"ZSBFViBSb290IENBMB4XDTEzMTAyMjEyMDAwMFoXDTI4MTAyMjEyMDAwMFowdTEL\n" \
"MAkGA1UEBhMCVVMxFTATBgNVBAoTDERpZ2lDZXJ0IEluYzEZMBcGA1UECxMQd3d3\n" \
"LmRpZ2ljZXJ0LmNvbTE0MDIGA1UEAxMrRGlnaUNlcnQgU0hBMiBFeHRlbmRlZCBW\n" \
"YWxpZGF0aW9uIFNlcnZlciBDQTCCASIwDQYJKoZIhvcNAQEBBQADggEPADCCAQoC\n" \
"ggEBANdTpARR+JmmFkhLZyeqk0nQOe0MsLAAh/FnKIaFjI5j2ryxQDji0/XspQUY\n" \
"uD0+xZkXMuwYjPrxDKZkIYXLBxA0sFKIKx9om9KxjxKws9LniB8f7zh3VFNfgHk/\n" \
"LhqqqB5LKw2rt2O5Nbd9FLxZS99RStKh4gzikIKHaq7q12TWmFXo/a8aUGxUvBHy\n" \
"/Urynbt/DvTVvo4WiRJV2MBxNO723C3sxIclho3YIeSwTQyJ3DkmF93215SF2AQh\n" \
"cJ1vb/9cuhnhRctWVyh+HA1BV6q3uCe7seT6Ku8hI3UarS2bhjWMnHe1c63YlC3k\n" \
"8wyd7sFOYn4XwHGeLN7x+RAoGTMCAwEAAaOCAUkwggFFMBIGA1UdEwEB/wQIMAYB\n" \
"Af8CAQAwDgYDVR0PAQH/BAQDAgGGMB0GA1UdJQQWMBQGCCsGAQUFBwMBBggrBgEF\n" \
"BQcDAjA0BggrBgEFBQcBAQQoMCYwJAYIKwYBBQUHMAGGGGh0dHA6Ly9vY3NwLmRp\n" \
"Z2ljZXJ0LmNvbTBLBgNVHR8ERDBCMECgPqA8hjpodHRwOi8vY3JsNC5kaWdpY2Vy\n" \
"dC5jb20vRGlnaUNlcnRIaWdoQXNzdXJhbmNlRVZSb290Q0EuY3JsMD0GA1UdIAQ2\n" \
"MDQwMgYEVR0gADAqMCgGCCsGAQUFBwIBFhxodHRwczovL3d3dy5kaWdpY2VydC5j\n" \
"b20vQ1BTMB0GA1UdDgQWBBQ901Cl1qCt7vNKYApl0yHU+PjWDzAfBgNVHSMEGDAW\n" \
"gBSxPsNpA/i/RwHUmCYaCALvY2QrwzANBgkqhkiG9w0BAQsFAAOCAQEAnbbQkIbh\n" \
"hgLtxaDwNBx0wY12zIYKqPBKikLWP8ipTa18CK3mtlC4ohpNiAexKSHc59rGPCHg\n" \
"4xFJcKx6HQGkyhE6V6t9VypAdP3THYUYUN9XR3WhfVUgLkc3UHKMf4Ib0mKPLQNa\n" \
"2sPIoc4sUqIAY+tzunHISScjl2SFnjgOrWNoPLpSgVh5oywM395t6zHyuqB8bPEs\n" \
"1OG9d4Q3A84ytciagRpKkk47RpqF/oOi+Z6Mo8wNXrM9zwR4jxQUezKcxwCmXMS1\n" \
"oVWNWlZopCJwqjyBcdmdqEU79OX2olHdx3ti6G8MdOu42vi/hw15UJGQmxg7kVkn\n" \
"8TUoE6smftX3eg==\n" \
"-----END CERTIFICATE-----\n" ;
```
I received this error:
```
Waiting for NTP time sync: .
Current time: Sun Aug  4 09:36:07 2019
[E][WiFiClient.cpp:261] setSocketOption(): 1006 : 9
[E][ssl_client.cpp:33] handle_error(): X509 - Certificate verification failed, e.g. CRL, CA or signature check failed
[E][ssl_client.cpp:35] handle_error(): MbedTLS message code: -9984
[E][WiFiClientSecure.cpp:132] connect(): start_ssl_client: -9984
[E][HTTPUpdate.cpp:217] handleUpdate(): HTTP error: connection refused

HTTP_UPDATE_FAILED Error (-1): HTTP error: connection refused
```

If i try to see the ce with this site, i received a good result: https://is.gd/UQm92D

Someone have my problem?

DaveCalaway, 2019-08-17:
Part of the solution is here: https://github.com/espressif/arduino-esp32/issues/846#issuecomment-345442453
The page https://github.com has a certificate, but the direct link  ( https://raw.githubusercontent.com ) has another certificate.

closed 2019-08-17 by DaveCalaway