espressif/arduino-esp32 #3025: Reset error PLEASE HELP ME.

DoTTa00, 2019-07-22:
### Hardware:
Board:							TTGO-ESP 32 SP-WROOM-32 board with OLED and battery holder.

Core Installation/update date:			
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                          no
Upload Speed:						115200
Computer OS:                                            Windows 10

### Description:
I am doing a test to get data in JSON, but when I try to access the IP in the browser there is a reset that appears on the Serial monitor and I can not open the IP page.

In my search in the forums, I already did the erase_flash command in the IDF, I changed the partition and the power is being made by the PC.

PS: I'm new to the world of ESP32.
PS2: I can not post the code, because it does not just involve me but a matter of work.

Please help me, I'm desperate.

### Debug Messages:
```
dhcps: send_offer>>udp_sendto result 0
New Client.
E (12566) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (12566) task_wdt:  - IDLE0 (CPU 0)
E (12567) task_wdt: Tasks currently running:
/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/tasks.c:2331 (pcTaskGetTaskName)- assert failed!
abort() was called at PC 0x4008a116 on core 0

Backtrace: 0x4008cbac:0x3ffbc2c0 0x4008cddd:0x3ffbc2e0 0x4008a116:0x3ffbc300 0x400dbae6:0x3ffbc320 0x40081b59:0x3ffbc340 0x4013fd07:0x3ffbc400 0x400da1cb:0x3ffbc420 0x4008aa49:0x3ffbc440 0x40088b49:0x3ffbc460

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:9232
load:0x40080400,len:6400
entry 0x400806a8
Sistema inicializado...
Setting AP (Access Point)…AP IP address: 192.168.4.1
Server is ready.
Please connect to http://0.0.0.0
Inicializado array oscilografia...

```
### Decoding stack results:
```
0x4008cbac: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 155
0x4008cddd: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 170
0x4008a116: pcTaskGetTaskName at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/tasks.c line 2331
0x400dbae6: task_wdt_isr at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/task_wdt.c line 166
0x4013fd07: esp_pm_impl_waiti at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/pm_esp32.c line 487
0x400da1cb: esp_vApplicationIdleHook at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/freertos_hooks.c line 63
0x4008aa49: prvIdleTask at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/tasks.c line 3412
0x40088b49: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143
```

lbernstone, 2019-07-22:
The watchdog timer is called because something in your code is blocking the cpu.  Without seeing the code it is impossible to know what that is.  Try to make a minimal sketch that demonstrates the problem.  The most basic thing to check is that any loop that might have a lot of processing involved calls a delay() so that other tasks can get some processing time.

DoTTa00, 2019-07-22:
Thank you for habilitate to help me. @lbernstone 

the delay() wich you say i put in the loop() function or in logical of the calculo and loading JSON?

### The code simplyfied

```
#include <WiFi.h>
#include <ArduinoJson.h>

WiFiServer server(80);
DynamicJsonDocument doc(1000);

void setup(void){
  
  Serial.begin(115200);
  Serial.println("Sistema inicializado...");

  //-------------------------------------------------------
  // JSON
  //-------------------------------------------------------
  Serial.print("Setting AP (Access Point)…");
  WiFi.softAP(ssid, password);

  IPAddress IP = WiFi.softAPIP();
  Serial.print("AP IP address: ");
  Serial.println(IP);
  

  server.begin();

  Serial.println(F("Server is ready."));
  Serial.print(F("Please connect to http://"));
  Serial.println(WiFi.localIP());
}

void loop(void){
	
	WiFiClient client = server.available(); 
 	if (!client) return;

 	Serial.println(F("New Client."));
 	while (client.available()) client.read();
 

  
  	timeAutomaticReadStart(); //It's a function wich realize the calculos and load JSON data
	
	// Write JSON document
	serializeJsonPretty(doc, client);

  	// Disconnect
  	client.stop();

 }

// This below it's my logical to load the JSON data and this function is called after the calculos
void mostraValoresJson(){

  doc["FaseA"]=timeA;
  doc["FaseB"]=timeB;
  doc["FaseC"]=timeC;
  
  JsonArray oscilografia_A = doc.createNestedArray("oscilografia");
   
    for(int j=0; j < OSCILOGRAFIA_SIZE; j++){
    JsonObject oscilografiaObj = oscilografia_A.createNestedObject();
     oscilografiaObj["TimeStamp"]=oscilografia[j].timestamp;
     oscilografiaObj["FaseA"]=oscilografia[j].faseA;
     oscilografiaObj["FaseB"]=oscilografia[j].faseB;
     oscilografiaObj["FaseC"]=oscilografia[j].faseC;
     
    
    if(oscilografia[j].faseA == 9999 && oscilografia[j].faseB == 9999 && oscilografia[j].faseC == 9999){
      break;
    }

   
  }
  Serial.println(oscilografia_A.size());
  
}
```

closed 2019-10-04 by stale[bot]