espressif/arduino-esp32 #3028: boot crash: /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/queue.c:621 (xQueueTakeMutexRecursive)- assert failed!

alorbach, 2019-07-25:
I am new to Esp32 dev, but I have succesffully developed for ESP8266 already. 
So I am trying to get ESp32 AI Thinker module with OV2640 camera module working. 

The sample sketches I tried are working, and I have started to adapt the sample code into my own existing project. 

It compiles fine but crashed with the following output: 
`ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:9232
load:0x40080400,len:6400
entry 0x400806a8
/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/queue.c:621 (xQueueTakeMutexRecursive)- assert failed!
abort() was called at PC 0x400894db on core 0

Backtrace: 0x4008c7e0:0x3ffe39e0 0x4008ca11:0x3ffe3a00 0x400894db:0x3ffe3a20 0x400e6700:0x3ffe3a40 0x400829ea:0x3ffe3a60 0x40081c31:0x3ffe3a80 0x40081eab:0x3ffe3ad0 0x400e6339:0x3ffe3b00 0x400e64fa:0x3ffe3b30 0x400e654b:0x3ffe3b50 0x400d7818:0x3ffe3b70 0x400d4c70:0x3ffe3b90 0x400d271a:0x3ffe3bb0 0x400e69ef:0x3ffe3bd0 0x40082cfb:0x3ffe3bf0 0x40082f00:0x3ffe3c20 0x40078f2b:0x3ffe3c40 0x40078f91:0x3ffe3c70 0x40078f9c:0x3ffe3ca0 0x40079165:0x3ffe3cc0 0x400806da:0x3ffe3df0 0x40007c31:0x3ffe3eb0 0x4000073d:0x3ffe3f20

Rebooting...
`
A sample Sketch (https://github.com/yoursunny/esp32cam/blob/795e3a615f5076a1c5acdcd20f31d14bdf2bfe59/examples/WifiCam/WifiCam.ino) I used works fine with the same settings: 
`ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:9232
load:0x40080400,len:6400
entry 0x400806a8

[E][camera.c:1049] camera_probe(): Detected camera not supported.
[E][camera.c:1249] esp_camera_init(): Camera probe failed with error 0x20004
..http://172.21.0.184
  /cam.bmp
  /cam.jpg
  /cam.mjpeg`

This are my board settings. 
[image]

alorbach, 2019-07-25:
Well I found out it was caused by static object of a helper class I made to control eeprom. 
The constructor called "EEPROM.begin" before setup() was called. 

I found this using EspExceptionDecoder found here: 
https://github.com/me-no-dev/EspExceptionDecoder

Very usefull tool

alorbach, 2019-07-25:
To close, got the next ASSERT!
`/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/queue.c:1442 (xQueueGenericReceive)- assert failed!
abort() was called at PC 0x4008bc69 on core 1
`

This is the stacktrace: 

`0x4008f0cc: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 155
0x4008f2fd: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 170
0x4008bc69: xQueueGenericReceive at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/queue.c line 1442
0x40115fd6: sys_mutex_lock at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/port/esp32/freertos/sys_arch.c line 78
0x401162b2: sys_arch_protect at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/port/esp32/freertos/sys_arch.c line 469
0x40105540: do_memp_malloc_pool at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/memp.c line 302
0x401055a9: memp_malloc at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/memp.c line 398
0x4010d2ad: udp_new at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/udp.c line 1145
0x400da52e: AsyncUDP::_init() at ...\packages\esp32\hardware\esp32\1.0.2\libraries\AsyncUDP\src\AsyncUDP.cpp line 469
0x400da5f4: AsyncUDP::listen(ip_addr const*, unsigned short) at ...\packages\esp32\hardware\esp32\1.0.2\libraries\AsyncUDP\src\AsyncUDP.cpp line 536
0x400da7d0: AsyncUDP::listen(unsigned short) at ...\packages\esp32\hardware\esp32\1.0.2\libraries\AsyncUDP\src\AsyncUDP.cpp line 697
0x400d8149: UdpServer::StartServer() at ...\arduino_build_538703\sketch\udpserver.cpp line 17
0x400d27fa: setup() at ..Controller.ino line 71`

These assrtes are really annoying. 
I have to say that WIFI has not been started at this point when my UDP Server starts, is that the problem?

alorbach, 2019-07-25:
Any chance to disable configASSERT() ?

ChristosPalisidis, 2019-07-25:
> Well I found out it was caused by static object of a helper class I made to control eeprom.
> The constructor called "EEPROM.begin" before setup() was called.
> 
> I found this using EspExceptionDecoder found here:
> https://github.com/me-no-dev/EspExceptionDecoder
> 
> Very usefull tool

Hello,
I have the same crashing allert in my code. My code has to do with wifi setup configuration via BLE . I managed to run it properly one time, but from the time i reset my esp32, this problem started to occur. I tried to download the EspExceptionDecoder but i cannot see the ESP Exception Decoder menu item. Can you guide me on how to install it properly?
Also, any further possible solutions to my problem are well acceptable :)

ChristosPalisidis, 2019-07-25:
My code is actually that of Mr Bernd Giesecke ,only upgrating from ArduinoJson 5 to 6 :
https://bitbucket.org/beegee1962/esp32_wifi_ble_esp32/src/master/src/main.cpp

alorbach, 2019-07-25:
Well it appears that this xQueueGenericReceive assert happens for various reasons. 
I am not sure yet if it happens because I am starting network services before WIFI is connected. 
This and all my other code worked perfectly fine on ESP8266 before. 

Regarding EspExceptionDecoder, I just following the Installation instructions and copied the extracted release to "<home_dir>/Arduino/tools/", I am on Windows. 

I really would like to disable those ASSERTS

atanisoft, 2019-07-25:
@alorbach you can not disable the ASSERTS, they are there for a reason! You are attempting to access/use functionality prior to it being configured/initialized. There global constructors will be invoked before app_main (IDF side) is invoked so you should only use the constructors for storing of values in the class instance and save actual init/config until a begin() like method is called from setup().

alorbach, 2019-07-25:
@atanisoft : I understand, but in my case setup() is finished and I have moved the UDP Server Start to top of my loop() function. 

Still I can the ASSERT. However WIFI has not been started at this point, is this the problem?

atanisoft, 2019-07-25:
> However WIFI has not been started at this point, is this the problem?

If you are attempting to use WiFi (UDP would!) then yes it is a problem in *YOUR* code. Since you haven't shared your code it is impossible to assist in tracking down what went wrong.

alorbach, 2019-07-25:
Well thats all I want to know. 
I am using many classes and includes, so it isn't so easy to share my code. 

I have used this code on ESP8266 without any problems, so I will look into changing the code that all network services are started after WIFI.Begin Call is made.

alorbach, 2019-07-26:
@ChristosPalisidis you need JAVA Runtime installed for ESPExceptiondecoder

VigneshTheLearner, 2019-08-11:
i have the same rebooting prob on esp32 cam got from aliexp... i need help... thanks in advance the example code which i used to run(worked before) now it causes SW_CPU_RESRET and reboot.
exception decoder gives
`Decoding stack results
0x40091ae0: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 155
0x40091d11: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 170
0x400dbbeb: __assert_func at ../../../.././newlib/libc/stdlib/assert.c line 63
0x400e3145: nvs::Storage::readMultiPageBlob(unsigned char, char const*, void*, unsigned int) at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/nvs_flash/src/nvs_storage.cpp line 438
0x400e31b0: nvs::Storage::readItem(unsigned char, nvs::ItemType, char const*, void*, unsigned int) at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/nvs_flash/src/nvs_storage.cpp line 455
0x400e227a: nvs_get_str_or_blob(nvs_handle, nvs::ItemType, char const*, void*, size_t*) at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/nvs_flash/src/nvs_api.cpp line 515
0x400e27ad: nvs_get_blob(nvs_handle, char const*, void*, size_t*) at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/nvs_flash/src/nvs_api.cpp line 525
0x4008da7d: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143
`

atanisoft, 2019-08-11:
@VigneshTheLearner your issue appears to be unrelated to this issue. Your issue appears to be related to a corrupt NVS, it failed to read the expected amount of data from NVS. You likely can fix the issue by erasing the flash and reuploading the sketch. If this does not fix your issue please open a new issue to track your crash.

VigneshTheLearner, 2019-08-11:
> @VigneshTheLearner your issue appears to be unrelated to this issue. Your issue appears to be related to a corrupt NVS, it failed to read the expected amount of data from NVS. You likely can fix the [...]

works after earsing the flash and reupload thank u bro @atanisoft

SalmaG98, 2019-08-23:
I also have a similar problem with my code: https://pastebin.com/797gDj6b  , I am using a Node mcu 32s and a pulse sensor SEN-11574 
I get the following error:
[image]

This is my first time working on an ESP32 and trying multiprocessing.
Any help or suggestions would be much appreciated!

lbernstone, 2019-08-23:
@SalmaG98 This is also an unrelated issue.  Please do not hijack the thread.  Post your backtrace in a new issue if you want help.

closed 2019-11-05 by stale[bot]

D3vl0per, 2019-12-03:
Hi Community!
I have same problem when I added WiFiMulti.h.
```
#include <Arduino.h>
#include <NTPClient.h>
#include <WiFi.h>
#include <WiFiUdp.h>
#include <PubSubClient.h>
#include <HTTPClient.h>
#include <ArduinoJson.h>
#include <config.h>
#include <gpio.h>
#include <WiFiMulti.h>

WiFiMulti wifiMulti;

extern "C"
{
	#include "freertos/FreeRTOS.h"
	#include "freertos/timers.h"
}

#include <AsyncMqttClient.h> // https://github.com/marvinroger/async-mqtt-client/blob/master/docs/2.-API-reference.md

AsyncMqttClient mqttClient;
TimerHandle_t mqttReconnectTimer;
TimerHandle_t wifiReconnectTimer;

bool debug = true;

WiFiUDP ntpUDP;
NTPClient timeClient(ntpUDP, "europe.pool.ntp.org", 3600, 60000);

void isConnectToWifi()
{
    wifiMulti.addAP(WIFI_SSID_1, WIFI_PASS_1);
    wifiMulti.addAP(WIFI_SSID_2, WIFI_PASS_2);
    wifiMulti.addAP(WIFI_SSID_3, WIFI_PASS_3);
    wifiMulti.addAP(WIFI_SSID_4, WIFI_PASS_4);
}
```