espressif/arduino-esp32 #3030: There is a lack of Ethernet examples

GeorgeFlorian, 2019-07-25:
OS: Linux Mint 19.1
Board: Olimex ESP32-EVB

Hello !

I am trying to implement an Ethernet connection instead of an WiFi connection.

This is how I start a WiFi connection in STA Mode right now:
```
       if(!WiFi.config(local_IP_STA, gateway_STA, subnet_STA, local_IP_STA, primaryDNS)) {
         logOutput((String)"Couldn't configure STATIC IP ! Starting DHCP IP !");
       }
       delay(50);
       WiFi.begin(x[0].c_str(),x[1].c_str());
```
What is the equivalent for the above code when using Ethernet ?

Also, I have a couple of questions:
 * Can WiFi and Ethernet be usable in the same time and how would this work. Will the ESP32 be able to handle 2 different types of connections in the same time or you can code-in both of these but only use one at a time ?
 * Can Ethernet be used in AP Mode ?
 * How come the ETH example is in WiFi examples ? Doesn't ETH deserve a separate directory with its own examples ?

Thank you !

atanisoft, 2019-08-02:
I'm guessing you missed these: https://github.com/espressif/arduino-esp32/tree/master/libraries/WiFi/examples

The ETH_XXX ones are for Ethernet, they are present in the WiFi library as that library provides the necessary infrastructure.

GeorgeFlorian, 2019-08-02:
> I'm guessing you missed these: https://github.com/espressif/arduino-esp32/tree/master/libraries/WiFi/examples
> 
> The ETH_XXX ones are for Ethernet, they are present in the WiFi library as that library provides the necessary infrastructure.

Of course I've missed them then ! ETH is not WiFi, is it ?
I've managed to make it work since I've posted the question.

atanisoft, 2019-08-02:
On the ESP32 platform the ETH and WiFi systems share the same TCP stack and underlying PHY. So they are in a way coupled.

If your issue has been solved please consider closing this issue :)

GeorgeFlorian, 2019-08-02:
> On the ESP32 platform the ETH and WiFi systems share the same TCP stack and underlying PHY. So they are in a way coupled.
> 
> If your issue has been solved please consider closing this issue :)

How is anyone supposed to know that ?

closed 2019-08-02 by GeorgeFlorian

atanisoft, 2019-08-02:
> How is anyone supposed to know that ?

Good question. Most likely via the IDF docs but those are a bit terse or hard to follow in some areas.