espressif/arduino-esp32 #3035: dnsIP() returns localIP()

GeorgeFlorian, 2019-07-26:
OS: Linux Mint 19.1
Board: ESP32 Wroover DevKitV4

Hello !

I've noticed something strange today. After a WiFi connection was established I am printing the IP values:
```
        IP_copy = WiFi.localIP().toString();
        logOutput((String)"Connected to " + v[0] + " with IP addres: " + IP_copy);
        logOutput((String)"Gateway: " + WiFi.gatewayIP().toString());
        logOutput((String)"Subnet: " + WiFi.subnetMask().toString());
        logOutput((String)"DNS: " + WiFi.dnsIP().toString());
```
The thing is that `WiFi.dnsIP().toString()` is the same as `WiFi.localIP().toString()` and I did not set them like that. 
`WiFi.dnsIP().toString()` should return `8.8.8.8`.

lbernstone, 2019-07-26:
Unable to reproduce.  Please provide a minimal sketch and indicate what version of code you are running.

GeorgeFlorian, 2019-07-27:
platform = https://github.com/platformio/platform-espressif32.git
board = esp32dev
```
Updating tool-scons                      @ 2.20501.7      [Up-to-date]
Updating tool-unity                      @ 1.20403.0      [Up-to-date]
Updating contrib-pysite                  @ 2.27.190418    [Up-to-date]
Updating contrib-piohome                 @ 2.2.0          [Up-to-date]
Updating tool-pioplus                    @ 2.5.2          [Up-to-date]

Platform Manager
================
Platform Espressif 32
--------
Updating espressif32                     @ 1.9.0          [Up-to-date]
Updating tool-esptoolpy                  @ 1.20600.0      [Up-to-date]
Updating toolchain-xtensa32              @ 2.50200.80     [Up-to-date]
Updating framework-arduinoespressif32    @ 2.10002.190628 [Up-to-date]
Updating tool-mkspiffs                   @ 2.230.0        [Up-to-date]

Platform Espressif 32
--------
Updating espressif32                     @ 2c1b430        [Up-to-date]
Updating tool-esptoolpy                  @ 1.20600.0      [Up-to-date]
Updating toolchain-xtensa32              @ 2.50200.80     [Up-to-date]
Updating framework-arduinoespressif32    @ 2.10002.190628 [Up-to-date]
Updating tool-mkspiffs                   @ 2.230.0        [Up-to-date]
```

```
//------------------------- fileReadLines()
void fileReadLines(File file, String x[]) {
  int i = 0;
    while(file.available()){
      String line= file.readStringUntil('\n');
      line.trim();
      x[i] = line;
      i++;
    }    
}

//------------------------- startSTA()
void startSTA(String x[]){
  if(x[0] != NULL &&      // SSID
     x[0].length() != 0 &&
     x[1] != NULL &&      // Password
     x[1].length() != 0 &&
     x[2] != NULL &&      //Local IP
     x[2].length() != 0 &&
     x[3] != NULL &&      // Gateway
     x[3].length() != 0 &&
     x[4] != NULL &&      // Subnet
     x[4].length() != 0 &&
     x[5] != NULL &&      // DNS
     x[5].length() != 0) {
       IPAddress local_IP_STA, gateway_STA, subnet_STA, primaryDNS;
       local_IP_STA.fromString(x[2]);
       gateway_STA.fromString(x[3]);
       subnet_STA.fromString(x[4]);
       primaryDNS.fromString(x[5]);
       if(!WiFi.config(local_IP_STA, gateway_STA, subnet_STA, local_IP_STA, primaryDNS))
          logOutput((String)"Couldn't configure STATIC IP ! Starting DHCP IP !");
       delay(50);
       WiFi.begin(x[0].c_str(),x[1].c_str());
       delay(50);   
       } else if (x[0] != NULL &&      // SSID
             x[0].length() != 0 &&
             x[1] != NULL &&           // Password
             x[1].length() != 0 &&
             x[2] == NULL &&           //Local IP
             x[2].length() == 0 &&
             x[3] == NULL &&           // Gateway
             x[3].length() == 0 &&
             x[4] == NULL &&           // Subnet
             x[4].length() == 0 &&
             x[5] == NULL &&           // DNS
             x[5].length() == 0) {
               WiFi.begin(x[0].c_str(),x[1].c_str());
               delay(50);
               }

  int k = 0;
  while (WiFi.status() != WL_CONNECTED && k<20) {
    k++;
    delay(1000);
    logOutput((String)"Attempt " + k + " - Connecting to WiFi..");
  }
}

void setup() {
Serial.begin(115200);
SPIFFS.begin();

  File networkRead = SPIFFS.open("/networkDisplay.txt");
  if(!networkRead) logOutput((String)"ERROR ! Couldn't open file to read !");
  String v[6];
  fileReadLines(networkRead,v);
  networkRead.close();
  IP_copy = WiFi.localIP().toString();
  logOutput((String)"Connected to " + v[0] + " with IP addres: " + IP_copy);
  logOutput((String)"Gateway: " + WiFi.gatewayIP().toString());
  logOutput((String)"Subnet: " + WiFi.subnetMask().toString());
  logOutput((String)"DNS: " + WiFi.dnsIP().toString());
}

void loop() {}
```

And `networkDisplay.txt` has the following lines:
```
ssid
password
192.168.1.35
1.1.1.1
255.255.255.0
8.8.8.8
```

GeorgeFlorian, 2019-07-28:
This also happens when using DHCP. The DNS returns the same address as the IP.

CelliesProjects, 2019-07-29:
Can not confirm. At least not on ESP32 Arduino and with DHCP.
Must be a PIO thing. 
With the following sketch I just get the correct IP.
````c++
#include <WiFi.h>

const char * wifissid = "pass";
const char * wifipsk = "psk";

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.println( wifissid );
  WiFi.begin( wifissid, wifipsk );
  while ( !WiFi.isConnected() )
    delay(100);
  Serial.println( "DNS:" + WiFi.dnsIP().toString());
}

void loop() {
  // put your main code here, to run repeatedly:

}
````

GeorgeFlorian, 2019-07-29:
Well, actually, I am an idiot and I can't count.

This is WiFi.config :
```
bool WiFiSTAClass::config(IPAddress local_ip, IPAddress gateway, IPAddress subnet, IPAddress dns1, IPAddress dns2)
```

And this is how I was initializing it:
```
if(!WiFi.config(local_IP_STA, gateway_STA, subnet_STA, local_IP_STA, primaryDNS))
```

I was using `local_IP_STA` as primary DNS and `primaryDNS` as secondary DNS.

Changing it accordingly made it work as it was supposed to:
```
if(!WiFi.config(local_IP_STA, gateway_STA, subnet_STA, primaryDNS))
```

closed 2019-07-29 by GeorgeFlorian