espressif/arduino-esp32 #3036: ESP32 keeps reboot

bensenq, 2019-07-27:
I have two ESP32 DEVKIT V1.  One has been runnig well since I purchased it.  The second one, that I just got in the mail, keeps rebooting even if I load the same code and all the same.  Even with empty code, keeps rebooting.
 
Here a copy of the serial interface:
rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:9232
load:0x40080400,len:6400
entry 0x400806a8
ets Jun 8 2016 00:22:57
 
It repeats forever.  I'm programming with Arduino IDE.  I do the same in both ESP32's, one runs fine, the other continuously resets.
On my second purchase, I bought 5 esp32.  When first plugged, they list all parameters on the Serial interface, and no rebooting.  Once I load a program with the Arduino IDE, they enter the continuous reboot regardless of the code uploaded. All the ones from the second purchase do the same. 
 
Your help will be much appreciated.

lbernstone, 2019-07-27:
Open a command prompt and run: `python esptool.py chip_id` from the esp32/tools folder.  
Make sure your image is smaller than the flash size listed.
Use the 'esp32 dev module' entry in boards, then set to the lowest common denominator flash settings (DIO/40MHz/no PSRAM/115200 baud).
Make sure your loop function has something in it, even if just a delay.

bensenq, 2019-07-28:
@lbernstone it works,  thank you very much. There must be some difference bettween two boards.
After using lowest setting (DIO/40MHz/no PSRAM/115200 baud), program is download successfully.

closed 2019-07-28 by bensenq

mmashraf, 2019-08-05:
I was unable to find chip_ID using commands. Secondly how and where to use this command on windows command prompt i.e esp32 dev module. please let mu guide further on 
how to check and find  "Make sure your image is smaller than the flash size listed"
and how to reduce the image size

. Connecting.....
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: 3c:71:bf:f1:1d:64
Uploading stub...
Running stub...
Stub running...
Warning: ESP32 has no Chip ID. Reading MAC instead.
MAC: 3c:71:bf:f1:1d:64
Hard resetting via RTS pin...

lbernstone, 2019-08-05:
DO NOT HIJACK THREADS!!!!
https://github.com/espressif/esptool#read-spi-flash-id-flash_id

wonnachai2, 2019-08-23:
@lbernstone Following #3129 I run on windows 10. I have searched but i dont found ESPtool.

wonnachai2, 2019-08-23:
@lbernstone Now I do it follow you but it still have same poblem.
[image]

thylive, 2019-08-30:
> @lbernstone Now I do it follow you but it still have same poblem.
> [image]

have you solved the problem?

thylive, 2019-09-01:
could you tell me how to slove the problem like this ?
and how to set " no PSRAM " ?

mmashraf, 2019-09-01:
go into tools and select PSRAM: "Disabled" for your ESP board

tonys0, 2019-10-15:
This actually helped me to solve the problem with continuous reboot.

In Arduino IDE: click Tools, Board, Board Manager. 
After the dialog is loaded type ESP32 and look for:
esp32 by Espressif Systems ...
if it is installed
Press Update button (in my case to version 1.0.4)
if it is not installed, I can not see how you got here to find the continuous reboot problem.
But do not worry, just press Install button.

Try this before you go for 160MHz CPU frequency, 40MHz flash, DIO flash mode, PSRAM disabled etc.
not to mention the resistor and capacitor on EN input.

Kashyapkoshti, 2020-11-19:
rst:0x10 (RTCWDT_RTC_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371

I am getting this error in my esp32 devkit v1
How to solve it?
I am using arduino IDE for programming the esp32.

quytran01, 2020-12-03:
i remove esp32 library and download version other . 
and it works

Kashyapkoshti, 2020-12-03:
@quytran01 can you provide which version you are using?

quytran01, 2020-12-04:
@Kashyapkoshti i am using esp32 version 1.0.3, it works but i think maybe the error is due to your hardware

quytran01, 2020-12-04:
@Kashyapkoshti Please try with other esp32  to be sure the fault is due hardware or software

lukbieli, 2021-10-20:
In my case reinstalling esp32 hardware library helped

pranjals94, 2024-01-05:
yap i tried in a different board it worked, but ESP32 seems to be power hungry... :D

Darren777, 2025-05-22:
I found 2 solutions to the problem which seems to be cause by a small amount of code being regarded as no code by the esp32 so it just stays in boot mode thinking that it has no code to run.

Firstly, holding the boot button on the ESP32 down prior to connecting the USB cable, and then only releasing it once the sketch starts uploading. this solved the problem.

Secondly, this solution only works in Arduino IDE, before uploading the sketch go to the upload options and choose the "large app" option