espressif/arduino-esp32 #3037: Can't connect to ESP32 AP mode

GeorgeFlorian, 2019-07-28:
OS: Linux Mint 19.1
Board: ESP32 Wroover DevKitV4

I am trying to connect to the ESP32 while it is in AP mode. Doing it with a Samsung Note 9 is easy peasy. Doing it with my PC however proves impossible. It will "connect" to the ESP but the webserver won't serve any page returning:

```
This site can’t be reached
109.108.112.114 took too long to respond.
```

```
//------------------------- logOutput(String)
void logOutput(String string1) {
	circle.push(string1);	
	Serial.println(string1);	
}

IPAddress local_IP_AP(109,108,112,114);
IPAddress gatewayAP(1,1,1,1);
IPAddress subnetAP(255,255,255,0);

 logOutput(WiFi.mode(WIFI_AP) ? "ESP32 went in AP Mode !" : "ESP couldn't go in AP_MODE. ESP_AP_STATION_MODE will start.");
  logOutput((String)"Starting AP ... ");
  logOutput(WiFi.softAP(ssidAP, passwordAP) ? "Ready" : "WiFi.softAP failed ! (Password must be at least 8 characters long )");
  delay(100);
  logOutput((String)"Setting AP configuration ... ");
  logOutput(WiFi.softAPConfig(local_IP_AP, gatewayAP, subnetAP) ? "Ready" : "Failed!");
  delay(100);
  logOutput((String)"Soft-AP IP address: ");
  logOutput(WiFi.softAPIP().toString());
```

These are the Verbose Logs:
```
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 0 - WIFI_READY
ESP32 went in AP Mode !
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 13 - AP_START
Starting AP ...
Ready
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 13 - AP_START
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 13 - AP_START
Setting AP configuration ...
Ready
Soft-AP IP address:
109.108.112.114
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 15 - AP_STACONNECTED
dhcps: send_offer>>udp_sendto result 0
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 17 - AP_STAIPASSIGNED
```

I have also tried without:
```
  logOutput((String)"Setting AP configuration ... ");
  logOutput(WiFi.softAPConfig(local_IP_AP, gatewayAP, subnetAP) ? "Ready" : "Failed!");
```

and it still did not work.

Also tried `IPAddress gatewayAP(0,0,0,0);` with no luck.

It just won't work on the PC.

What is happening ? Is this a bug or what ?

psykokwak-com, 2019-07-29:
Hi, Maybe #2971 ?

GeorgeFlorian, 2019-07-29:
> Hi, Maybe #2971 ?

Hello ! Maybe !
But I am using PlatformIO. 
I used both Stable and Development platforms and the problems are on both.
I don't really know if the PlatformIO ESP32 Core is using the same commits as this one.

GeorgeFlorian, 2019-07-29:
> Hi, Maybe #2971 ?

So, I've manually replaced all of the content of both sdkfiles with the ones I found here https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/sdkconfig and here https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/include/config/sdkconfig.h
and I have no idea if the changes are actually in place. I will continue testing.

GeorgeFlorian, 2019-07-29:
Update:

It appears as it works now ! I've only modified the sdkconfig files accordingly.

closed 2019-07-29 by GeorgeFlorian