espressif/arduino-esp32 #3039: SD_MMC

perf3ct1on666, 2019-07-29:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do?
Describe your system( Hardware, computer, O/S, core version, environment).
Describe what is failing.
Show the shortest possible code that will duplicate the error.
Show the EXACT error message(it doesn't work is not enough).
All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							ESP32 Dev board
Core Installation version:			1.0.3 rc1
IDE name:						arduino ide
Flash Frequency:					80mhz
PSRAM enabled:                                    no
Upload Speed:					921600
Computer OS:                                        win 10

// problem description
i meet the problem is when SD_MMC write file in loop(). if i put the delay more than 100, the file will still appended but after few time esp will hangging and the serial monitor show sdmmc_read_blocks failed. but if i put delay() 1 or 10 only the problem don't happen...
----------------------------------------------------------
void setup() {
    Serial.begin(115200);
    if(!SD_MMC.begin()){
        Serial.println("Card Mount Failed");
        return;
    }
    uint8_t cardType = SD_MMC.cardType();

    if(cardType == CARD_NONE){
        Serial.println("No SD_MMC card attached");
        return;
    }
}

void loop() {
  appendFile(SD_MMC, "/test.txt", "hello world\n");
  delay(1000);
}
```

### Debug Messages in Serial monitor
```
Appending to file: /test.txt
Message appended

Appending to file: /test.txt
Message appended

Appending to file: /test.txt
Message appended

Appending to file: /test.txt
E (2314) sdmmc_cmd: sdmmc_read_sectors_dma: sdmmc_send_cmd returned 0xffffffff
E (2318) diskio_sdmmc: sdmmc_read_blocks failed (-1)
Message appended

Appending to file: /test.txt
E (4324) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
E (4324) sdmmc_cmd: sdmmc_read_sectors_dma: sdmmc_send_cmd returned 0x107
E (4325) diskio_sdmmc: sdmmc_read_blocks failed (263)
```

perf3ct1on666, 2019-07-29:
And also i need help on uploading text file from SD_MMC to the phpmyadmin... i need upload the file not the text...
the research i found is need to use WiFiClient.h. it included the client.write(file) function to upload the file. but i hope someone can provide me a simple example...

perf3ct1on666, 2019-08-01:
any guys help?.....

closed 2019-08-09 by perf3ct1on666

Gornerrrrr, 2019-09-16:
Can you solve the problem?I have same problem.

Patriboom, 2023-10-25:
Here: https://rntlab.com/question/esp32-cam-saving-photo-on-sc-card/  to be easy to solve:  set the GPIO4 at LOW.
That's the way I'm working on mine hoping to solve the exact same problem.