espressif/arduino-esp32 #3043: esp32-1.0.3-rc1 release doesn't have the changes made to the sdkconfig files

GeorgeFlorian, 2019-07-29:
Hello !

I am trying to get the new commits:
https://github.com/espressif/arduino-esp32/commit/b0d8d4dd442b2953e57b16d05d6bf2a893334f1d and https://github.com/espressif/arduino-esp32/commit/9e32cec9a287f78f212ecdae72aa6d0912b87448
and so I've downloaded the pre-release, the latest available version but it doesn't have those commits...
I've checked the sdkconfig files and they have their old values...

atanisoft, 2019-07-29:
That would be because sdkconfig was modified *AFTER* rc1 was released...

GeorgeFlorian, 2019-07-29:
> That would be because sdkconfig was modified _AFTER_ rc1 was released...

That's what I thought also...

I ended up replacing all of the content of both sdkfiles from `.platformio\packages\framework-arduinoespressif32` with the ones I found here https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/sdkconfig and here https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/include/config/sdkconfig.h

I don't know if it's a good practice, but it did fix my AP issue: https://github.com/espressif/arduino-esp32/issues/3037

atanisoft, 2019-07-29:
You don't need to modify the internal packages from PIO to use the latest...
https://docs.platformio.org/en/latest/platforms/espressif32.html#upstream
By setting that value it will compile against the GIT version automatically.

GeorgeFlorian, 2019-07-29:
> You don't need to modify the internal packages from PIO to use the latest...
> https://docs.platformio.org/en/latest/platforms/espressif32.html#upstream
> By setting that value it will compile against the GIT version automatically.

Oh yeah, definitely . But I only need those bits. 
I am afraid that using the development or stage releases will break my code.

atanisoft, 2019-07-29:
You should test with it either way as once it is released your code is possibly broken anyway!

closed 2019-07-30 by GeorgeFlorian