espressif/arduino-esp32 #3044: Including <mutex> causes ctime "not been declared" errors

ssilverman, 2019-07-29:
### Hardware:
Board:							Adafruit ESP32 Feather
Core Installation version:			1.0.2
IDE name:						PlatformIO in VSCode
Flash Frequency:					N/A
PSRAM enabled:                                    N/A
Upload Speed:					N/A
Computer OS:                                        OSX

### Description:
Including `<mutex>` in a program causes the compiler to complain that a bunch of ctime things have not been declared.

I reference this issue for more details about the messages received and about possible causes from the esp-idf developers: https://github.com/espressif/esp-idf/issues/3698

lbernstone, 2019-07-29:
Please provide a minimal sketch and confirm that this fails in Arduino IDE.

ssilverman, 2019-07-29:
It doesn't fail in Arduino IDE, but it does fail in PlatformIO, and for non-simple sketches. Here's the error output when I include `<mutex>`:

```
In file included from /Users/shawn/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/chrono:41:0,
                 from /Users/shawn/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/mutex:39,
                 from src/ui/OSCHandler.h:11,
                 from src/ui/OSCHandler.cpp:4:
/Users/shawn/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/ctime:64:11: error: '::clock' has not been declared
   using ::clock;
           ^
/Users/shawn/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/ctime:65:11: error: '::difftime' has not been declared
   using ::difftime;
           ^
/Users/shawn/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/ctime:66:11: error: '::mktime' has not been declared
   using ::mktime;
           ^
/Users/shawn/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/ctime:67:11: error: '::time' has not been declared
   using ::time;
           ^
/Users/shawn/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/ctime:68:11: error: '::asctime' has not been declared
   using ::asctime;
           ^
/Users/shawn/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/ctime:69:11: error: '::ctime' has not been declared
   using ::ctime;
           ^
/Users/shawn/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/ctime:70:11: error: '::gmtime' has not been declared
   using ::gmtime;
           ^
/Users/shawn/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/ctime:71:11: error: '::localtime' has not been declared
   using ::localtime;
           ^
/Users/shawn/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/ctime:72:11: error: '::strftime' has not been declared
   using ::strftime;
```

I'll file a bug with the PlatformIO people...

ssilverman, 2019-08-01:
Turns out that I had `Time.h` and `Time.cpp` files in my root source directory (note the capitalization). I tried both renaming these and moving them to a subdirectory without renaming, and both approaches solved the problem.

closed 2019-08-01 by ssilverman