espressif/arduino-esp32 #3045: std::thread : please raise the main loop prio to >2 by default (instead of 1)

dsyleixa, 2019-07-30:
std::thread : please raise the main loop prio to >2 by default (instead of 1)

as  shown here the thread loop prios have to be increased to 0 to avoid blocking.
https://github.com/espressif/arduino-esp32/issues/2814
IMO it would be better to set the main loop to a higher prio by default (e.g., 3) to have a larger bandwidth fo thread prios to cooperate!

atanisoft, 2019-07-30:
You can call vTaskPrioritySet(NULL, 3); from setup and it will reset the current task's priority.

Or you can compile Arduino as an IDF component and adjust the settings however you like via sdkconfig.h

dsyleixa, 2019-07-30:
the latter opion is not clear, tbh.
I always use just the Arduino IDE, nothing different.

as to option 1:
here I was missing a documentation that 
vTaskPrioritySet(NULL, 3);
is supposed to work not just in std::threads but also in the main loop.
Thanks, I'll test that ASAP!

dsyleixa, 2019-07-30:
update:
no, that does not work:

```
// std::thread for ESP32, Arduino IDE

// ver 0.0.6 fibonacci

#include <Arduino.h>
#include <thread>
#include <freertos/task.h>

#ifndef LED_BUILTIN
#define LED_BUILTIN 13
#endif

const auto one_sec = std::chrono::seconds { 1 };

int fibbonacci(int n) {
   if(n == 0){
      return 0;
   } else if(n == 1) {
      return 1;
   } else {
      return (fibbonacci(n-1) + fibbonacci(n-2));
   }
}

void blinker_loop() {
    thread_local uint32_t counter = 0;
    Serial.println((String)"blinker_loop Current priority :" + uxTaskPriorityGet(NULL));
    vTaskPrioritySet(NULL,2);//set Priority
    Serial.println((String)"blinker_loop Current priority :" + uxTaskPriorityGet(NULL));
    while(true) {
        digitalWrite(LED_BUILTIN, HIGH);
        Serial.println((String)"blinker_loop (HIGH) counter: "+ counter);
        std::this_thread::sleep_for(one_sec);
        
        digitalWrite(LED_BUILTIN, LOW);
        Serial.println((String)"blinker_loop (LOW) counter: "+ counter);
        std::this_thread::sleep_for(std::chrono::milliseconds(500));
        
        
        counter++;
    }
}

void fibonacci_loop() {
    thread_local uint32_t counter = 0, i=0;
    Serial.println((String)"fibonacci_loop Current priority :" + uxTaskPriorityGet(NULL));
    vTaskPrioritySet(NULL,2);//set Priority
    Serial.println((String)"fibonacci_loop Current priority :" + uxTaskPriorityGet(NULL));
    while(true) {
        for(i=30; i<41; i++) {    // limits: test, debug
          Serial.println( (String)"Fibbonacci of "+i+"="+fibbonacci(i));            
        }        
        Serial.println((String)"fibonacci_loop counter: "+counter);  
        counter++;      
    }
}

std::thread *thread_1;
std::thread *thread_2;

void setup() {
  Serial.begin(115200);
  delay(1000);
  vTaskPrioritySet(NULL, 3);
  
  thread_1 = new std::thread(blinker_loop);
[... 150 lines cut ...]
main loop: 0
Fibbonacci of 32=2178309
blinker_loop (LOW) counter: 0
Fibbonacci of 33=3524578
blinker_loop (HIGH) counter: 1
Fibbonacci of 34=5702887
blinker_loop (LOW) counter: 1
blinker_loop (HIGH) counter: 2
blinker_loop (LOW) counter: 2
Fibbonacci of 35=9227465
blinker_loop (HIGH) counter: 3
E (12195) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (12195) task_wdt:  - IDLE0 (CPU 0)
E (12195) task_wdt: Tasks currently running:
E (12195) task_wdt: CPU 0: pthread
E (12195) task_wdt: CPU 1: IDLE1
E (12195) task_wdt: Aborting.
abort() was called at PC 0x400d6573 on core 0

Backtrace: 0x40089150:0x3ffbe160 0x4008937d:0x3ffbe180 0x400d6573:0x3ffbe1a0 0x40081671:0x3ffbe1c0 0x400ea4f9:0x3ffb93d0 0x400ea4f1:0x3ffb93f0 0x400ea4f1:0x3ffb9410 0x400ea4f1:0x3ffb9430 0x400ea4f1:0x3ffb9450 0x400ea4f1:0x3ffb9470 0x400ea4f1:0x3ffb9490 0x400ea4f1:0x3ffb94b0 0x400ea4f1:0x3ffb94d0 0x400ea4f1:0x3ffb94f0 0x400ea4f1:0x3ffb9510 0x400ea4f1:0x3ffb9530 0x400ea4f1:0x3ffb9550 0x400ea4f1:0x3ffb9570 0x400ea4f1:0x3ffb9590 0x400ea4f1:0x3ffb95b0 0x400d1063:0x3ffb95d0 0x400ea4dd:0x3ffb9620 0x400e6e65:0x3ffb9640 0x400e5f30:0x3ffb9670 0x40087c9d:0x3ffb9690

Rebooting...
ets Jun  8 2016 00:22:57

```

atanisoft, 2019-07-30:
You are triggering the WDT since your std::thread impl doesn't delay to allow timeslicing. We have covered that in your issue about std::thread previously.

dsyleixa, 2019-07-30:
I have to disagrre:
leaving main loop by default 1,
and setting 
vTaskPrioritySet(NULL, 0);
in either std::thread, then it works like a charm

but being always  limited to thread prios 0, nothing different. (prio 1 would brak it, too).

atanisoft, 2019-07-30:
> E (12195) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
> E (12195) task_wdt:  - IDLE0 (CPU 0)
> E (12195) task_wdt: Tasks currently running:
> E (12195) [...]

You can disagree all you want but the above is what is causing your failure. You *MUST* have a delay in your tasks otherwise there will not be any task switching and the IDLEX task will be starved from execution and trigger the watchdog timer.

You are not limited to priority zero, it works at that level as it is at the same priority as the IDLEX tasks. Nothing more, nothing less. If you add a small delay, like vTaskDelay(1), you can use almost any priority and it will work as you expect.

dsyleixa, 2019-07-30:
no, you're wrong: I do not have to have a delay.
As stated, all runs fine with this program, no delays at all:
```

// std::thread for ESP32, Arduino IDE

// ver 0.0.6 fibonacci

#include <Arduino.h>
#include <thread>
#include <freertos/task.h>

#ifndef LED_BUILTIN
#define LED_BUILTIN 13
#endif

const auto one_sec = std::chrono::seconds
{
    1
};

int fibbonacci(int n) {
   if(n == 0){
      return 0;
   } else if(n == 1) {
      return 1;
   } else {
      return (fibbonacci(n-1) + fibbonacci(n-2));
   }
}

void blinker_loop() {
    thread_local uint32_t counter = 0;
    Serial.println((String)"blinker_loop Current priority :" + uxTaskPriorityGet(NULL));
    vTaskPrioritySet(NULL,0);//set Priority
    Serial.println((String)"blinker_loop Current priority :" + uxTaskPriorityGet(NULL));
    while(true) {
        digitalWrite(LED_BUILTIN, HIGH);
        Serial.println((String)"blinker_loop (HIGH) counter: "+ counter);
        std::this_thread::sleep_for(one_sec);
        
        digitalWrite(LED_BUILTIN, LOW);
        Serial.println((String)"blinker_loop (LOW) counter: "+ counter);
        std::this_thread::sleep_for(one_sec);
        
        counter++;
    }
}

void fibonacci_loop() {
    thread_local uint32_t counter = 0, i=0;
    Serial.println((String)"fibonacci_loop Current priority :" + uxTaskPriorityGet(NULL));
    vTaskPrioritySet(NULL,0);//set Priority
    Serial.println((String)"fibonacci_loop Current priority :" + uxTaskPriorityGet(NULL));
    while(true) {
        for(i=30; i<41; i++) {    // limits: test, debug
          Serial.println( (String)"Fibbonacci of "+i+"="+fibbonacci(i));            
        }        
        Serial.println((String)"fibonacci_loop counter: "+counter);  
        counter++;      
    }
}

std::thread *thread_1;
std::thread *thread_2;

void setup() {
  Serial.begin(115200);
  delay(1000);
  thread_1 = new std::thread(blinker_loop);
[... 153 lines cut ...]
blinker_loop (HIGH) counter: 49
blinker_loop (LOW) counter: 49
main loop: 18
Fibbonacci of 34=5702887
blinker_loop (HIGH) counter: 50
blinker_loop (LOW) counter: 50
blinker_loop (HIGH) counter: 51
Fibbonacci of 35=9227465
blinker_loop (LOW) counter: 51
blinker_loop (HIGH) counter: 52
main loop: 19
blinker_loop (LOW) counter: 52
blinker_loop (HIGH) counter: 53
blinker_loop (LOW) counter: 53
blinker_loop (HIGH) counter: 54
Fibbonacci of 36=14930352
blinker_loop (LOW) counter: 54
blinker_loop (HIGH) counter: 55
main loop: 20
blinker_loop (LOW) counter: 55

```
[/CODE]

ntl nur der jetzige Stand, möglicherweise erscheinen doch wieder irgendwann neue Erkenntnisse... ;) 8)

dsyleixa, 2019-07-30:
as you can see, it runs fine with thread prios=0, but just 0,
because main loop has prio=1.

My point in this topic is about raising the main loop prio to e.g. =3,
so then threads may have prio=0, 1,or 2 w/o blocking and watchdog issues.

atanisoft, 2019-07-30:
> as you can see, it runs fine with thread prios=0, but just 0,
> because main loop has prio=1, whilst the watchdog is feeded then by the main loop.

Sorry but your loop() task does *NOT* feed the watchdog. In fact none of your code does! That is why you are only able to use priority 0 for your tasks, they share the same priority as the IDLEX task.

The way the ESP32 FreeRTOS task scheduler works is based on task priority, it will give cycles to the highest priority task(s) until they yield (delay) to the scheduler for other tasks at lower priorities to run. There is no guarantee that lower priority tasks will be given any cycles without the higher priority tasks yielding.

Increasing the loop task priority will not solve the watchdog errors you are creating in your tasks, it will simply shift where they occur.

dsyleixa, 2019-07-30:
ok, then shift it, to enable more different thread prios!

atanisoft, 2019-07-30:
> (after all your hint
> 
> > You can call vTaskPrioritySet(NULL, 3); from setup and it will reset the current task's priority.
> 
> also didn't solve problem!)

Of course not, you didn't solve the problem in *YOUR* code which triggered the watchdog. This is a problem in *YOUR* code and not in arduino-esp32 or ESP-IDF.

> raise even the IDLEX prio additionally

This is not feasible or supported by FreeRTOS itself.

dsyleixa, 2019-07-30:
my code is fine, it **finally has to work** by **preemptive** MT scheduling.
A preemptive scheduler has to switch the timeslices no matter of delays or not (given the thread prios are equal or less).
Only _cooperative_ MT requires delays in between.
Also with pthread on Linux (preemptive, too) my code works fine without requiring any delays.
Also note that a stalling may happen anywhere at any time randomly in a thread e.g. because of unforeseen hardware or driver problems in a thread, so either delays wouldn't help anyway.

As I think that you are no RTOS developer on ESP32 please let a developer answer and please don't capture my request any more.

me-no-dev, 2019-07-30:
how about you use the FreeRTOS API for threads? have you tried that? Do you kniow on which core your threads are allocated or you do not care for affinity? what I see is that Core0 is calling WDT and Arduino runs on Core1

me21, 2019-07-30:
@dsyleixa please be more patient, no one owes you anything here. You seem to be unfamiliar with how the preemptive scheduler in FreeRTOS works. While higher-priority tasks are active, no lower-priority tasks will get any CPU time. This is different from desktop computers and even Raspberry Pi, where lower-priority tasks still get a chance to run.

But all the tasks with the highest priority will get CPU time, that's why the scheduler is still preemptive. It switches between runnable tasks with the highest priority found in the system. As you can see, it uses scheduling algorithm different from bigger machines.

If your tasks can stall, the primary cause for that is a bug somewhere in your task. On bigger machines you can kill the task and restart it, but it merely hides the bug, not fixes it. Microcontrollers should never ever stall.

@me-no-dev if Core0 calls WDT and Arduino runs on Core1, then `loop()` can hang up and it won't reboot?

me21, 2019-07-30:
@dsyleixa please be more patient, no one owes you anything here. You seem to be unfamiliar with how the preemptive scheduler in FreeRTOS works. While higher-priority tasks are active, no lower-priority tasks will get any CPU time. This is different from desktop computers and even Raspberry Pi, where lower-priority tasks still get a chance to run.

But all the tasks with the highest priority will get CPU time, that's why the scheduler is still preemptive. It switches between runnable tasks with the highest priority found in the system. As you can see, it uses scheduling algorithm different from bigger machines.

Here's a link to FreeRTOS explanation page: https://www.freertos.org/implementation/a00007.html

If your tasks can stall, the primary cause for that is a bug somewhere in your task. On bigger machines you can kill the task and restart it, but it merely hides the bug, not fixes it. Microcontrollers should never ever stall.

Raising IDLEX priority is useless, because it should be the task with the lowest priority, so that WDT will be reset if some other task hangs up and not gives CPU back. If it had a higher priority, then the task with lower priority could hang up undetectably. Think about it: how would you detect if your code stalls?

@me-no-dev if Core0 calls WDT and Arduino runs on Core1, then `loop()` can hang up and it won't reboot?

dsyleixa, 2019-07-30:
While higher-priority tasks are active, no lower-priority tasks will get any CPU time. 
That is true also for pthread on Linux.
As main() is running at prio 1 and IDLEX is running probably at prio 0 , all std::threads must run also at prio 0 not to ever block main() and the entire program.

The purpose is for bigger mobile robots to safely handle unforeseen (but possible) emergency events, not to damage the environment or even hurt or injure humans.

My request is about raising the main (and IDLEX ?) prios to have a wider range for arbitrary threads, not blocking the entire program.

me21, 2019-07-30:
But your thread which executes `fib` function takes 100% time and does not yield for a long time. If it has priority higher than 0, then Idle task doesn't get any CPU time and we have WDT reset.

From FreeRTOS point of view, your task looks like stalled - it executes and executes and never yields.

As Idle task should run at priority 0 to catch all other stalled tasks, it's not feasible to raise its priority.

Simply put, you should not execute long functions because they cannot be differentiated from stalled ones.

Your robots should avoid long functions and handle stalled ones gracefully.

And try PlatformIO instead of Arduino IDE, it's way better ;-)

dsyleixa, 2019-07-30:
my robots don't avoid long functions and also can't avoid using libs (e.g. for MPU6050) which hang up if  i2c fails.
So the tasks produce global timestamps (sort of heartbeat) which can be checked by main() if ok or stalling, and in that case (at least) the robot can quickly and safely be stopped.

That is the plan for ESP32 and that works already on my Pi.

The question is:
Am I strictly limited to prio=0 for unstable threads or is there a way to enable larger ranges?

me21, 2019-07-30:
You are strictly limited. But idle task already does what you want, it IS watcher thread for your stalled tasks. Watchdog resetting is like a heartbeat.

I don't understand what the fuss is about.

Read about watchdog in Espressif docs (I gave the link), I think it's possible to achieve safe robot operation and handling stalled tasks.

dsyleixa, 2019-07-30:
I don't understand RTOS, I only have knwledge aobout Linux pthread and a little about std::thread.
RTOS design and functionality  is far beyond my skills.
But ok, if I am stritly limited to prio=0 , then there's no choice and I have to resign about that.
Perhaps I'll try it with the limited ESP32 std::thread features, and if I fail I'll stay with my Pi.

me21, 2019-07-30:
That's ok, of course you should stick to what's familiar with big projects.
It takes sample small programs, trial and error, to learn something new, be it RTOS or web programming, for example.

Alan5142, 2019-07-30:
std::thread is implementation defined, there's no place in the standard that defines how it must be implemented, also pthread and std::thread in esp32 are just wrappers over a FreeRTOS task, you must yield tasks in order to allow other tasks to run

me21, 2019-07-31:
As a side note, I thought I should let you know that (as far as I understand RTOS concepts) it is guaranteed that your firmware running under RTOS will react to coming events not later than some well defined interval.
Desktop OS isn't RTOS (without special kernel tuning) and therefore cannot provide such guarantee. It will react satisfactorily most of the time, especially if the machine has some extra CPU power, but it can't be proven mathematically.
So if your robot runs on Linux, there's always a possibility that OS might decide to run its internal tasks (swapping, logging, disk checking, for examples) and miss the critical event it should have reacted to.

dsyleixa, 2019-08-01:
that is not true:
- preemptive multitasking does not require yields for equal thread prios
- on Linux the thread prios can be defined higher than standard  kernel threads 
- and Linux programs can be compiled to use cpu cores exclusively without enabling  kernel access

Alan5142, 2019-08-01:
> 
> 
> that is not true:
> 
>     * preemptive multitasking does not require yields for equal thread prios
> 
>     * on Linux the thread prios can be set higher than standard  kernel thread prios (e.g., SCHED_RR, prio=80)
> 
>     * and if required: additionally Linux programs can be compiled to use cpu cores exclusively without enabling  kernel access (no need to use that so far for my personal purposes yet though)
> 
>     * but as stated, for ESP32 the program above is running fine  by thread prios limited to =0

I'm not talking about Linux when i said that you must yield tasks in order to allow other tasks to run (indeed CPU starvation is bad even in a GPOS), I'm talking about ESP32, indeed ESP32 have a preemptive scheduler, but running all tasks in the same priority as the idle task is a bad design. Please, read [Watchdog](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/wdts.html) and this "fundamental" section [FreeRTOS Task](https://www.freertos.org/implementation/a00004.html)

Even if you use SCHED_RR in pthread, a GPOS does not provide a warranty that your thread should be executed in a exact time slice because there are other tasks with higher priorities (kernel tasks), so, Linux is not suitable for real time applications, so, if you need **real time processing** Linux may be a bad option even with SCHED_RR because it's not truly real-time.

If you want a full featured preemptive scheduler, why not use Linux? 🤷‍♂

Axel1400, 2019-08-01:
Hey dude, if you don't know anything about embedded system, why you want to implement one and don't even care about recommendations?, also you want a microcontroller to work as a PC and it won't work in that way, people told you to read documentation about ESP32 which is in IDF, not in life is people give you a working example, sometimes you need to modified something that exists, so if you want your project to work without learning a RTOS, then keep it up with Raspberry, or then work and attend recommendation to feed the watchdog in tasks.

dsyleixa, 2019-08-01:
as already stated, I am already using pthread on Linux (RPi) successfully but I tried to port programs to a ESP32 for a Arduino-based community project 
This due to convenience reasons; for most Arduino uses C++ on Pi is far too cumbersome, but that is also true for RTOS; Aduino is designed for non-professional programmers, e.g. hobbyists, artists,  and pupils. OTOH, also for Linux I do not have to know about how a Linux OS works in order to work with pthread, so I actually also do not want to care about RTOS works, I just use the high level functions (C++14 std::thread, C99 pthread).

me21, 2019-08-01:
Well , it won't work that way. You'll have to learn how RTOS works if you want to run real-time applications like controlling robots.

> I actually also do not want to care about how RTOS works, I just use the high level functions (C++14 std::thread, C99 pthread).

You'll have to care about how RTOS works. You'll have to call `yield` from time to time to let idle task run. Your threads working on Linux without having to call `yield` is pure coincidence, that behaviour is not required by any standard.

> that is not true - instead, true is:
> 
>     * preemptive multithreading does not require yields for equal thread prios compellingly
> 

I never said the opposite. ESP32 RTOS works that way.

>     * on Linux the thread prios can be set higher than standard  kernel thread prios (e.g., SCHED_RR, prio=80)
> 
>     * std::thread prios can be set by pthread thread ID prios (via native_handle() )
> 

You can't set std::thread priority in C++ standard portable way, am I missing something?

>     * and if required: additionally Linux programs can be compiled to use cpu cores exclusively without enabling  kernel access (no need to use that so far for my personal purposes yet though)
> 
> 
> ...but as stated, for ESP32 the program above is running fine by thread prios limited to =0

So, you need to know something about how Linux OS works after all? SCHED_RR, various schedulers, exclusive CPU core use, etc...

But in the end you're mistaken, it can't be guaranteed that your thread will have latency low enough to perform robot controlling tasks, even with exclusive CPU core use. Because to control robot, you'll need to perform I/O, and that means your thread will halt and give control to kernel (peripheral drivers). And it may be busy with its internal tasks, keeping your task frozen.

Think about it: if general purpose Linux would be suitable for real-time applications, why would people invent real-time Linux kernel patches?

dsyleixa, 2019-08-01:
no, I don't have to learn about RTOS, and I won't ever.
I am just using just C++ std::thread API functions, and perhaps C99 pthread.
what pthread provides is accessable via cplusplus.com and man pthread - that is enough to know.

me21, 2019-08-01:
It's up to you.
I think this issue now may safely be closed as non-issue.

dsyleixa, 2019-08-01:
no, you're wrong, because watchdog then makes the program break and reboot (see above).

me21, 2019-08-01:
Since you don't want to learn and read Espressif docs on ESP32 watchdog, you're right and I'm wrong 😀

dsyleixa, 2019-08-01:
yes, thank you, I already know that I'm right . 8-)

Alan5142, 2019-08-01:
Well, it's up to you if you want to use a higher level functionality, but you need to understand how it works and why the task is restarting, the right place to learn it is the esp32 idf 🤷‍♂️, we can't do nothing more for you

Axel1400, 2019-08-01:
If you don't want to use RTOS, then use the Raspberry and this Issue is over.
You are completely wrong and you don't even care, you don't want to learn something that isn't hard at all and also you expect things to work in the same way in two systems. I suggest you to left all about microcontroller, don't try to do your project in a microcontroller because you don't want to learn anything about embedded systems, so this issue is over.

closed 2019-08-02 by everslick

dsyleixa, 2022-01-01:
so why isn't it possible to raise the main loop prio to >2 by default and by design (instead of 1)?

me-no-dev, 2022-01-19:
why should it be? need a task with higher prio? just start one...

dsyleixa, 2022-01-19:
no, it is about lower threads of different prio (e.g., 0,1,2) which cannot block main prio (e.g., 3), lime I have already sstated before.
JTM, in Linux (Raspi) by pthrea it's similar:
main() has an intermediate prio (IIRC, 50 or 60?), and one can choose different pthread prios either  higher or lower than that.
So one is not limited to just 1 prio lower than main().

dsyleixa, 2022-01-19:
no, it is about lower threads of different prio (e.g., 0,1,2) which cannot block main prio (e.g., 3), like I have already stated before.
JTM, in Linux (Raspi) by pthread it's similar:
main() has an intermediate prio (IIRC, 50 or 60?), and one can choose lots of different pthread prios either  higher or lower than that (highest pio -20, IIRC).
So one is not limited to just 1 prio lower than main().

dsyleixa, 2022-01-19:
no, it is about lower threads of different prio (e.g., 0,1,2) which cannot block main prio (e.g., 3), like I have already stated before.
JTM, in Linux (Raspi) by pthread it's similar:
main() has an intermediate prio (IIRC, 50 or 60?), and one can choose lots of different pthread prios either higher or lower than that.
So one is not limited to just 1 prio lower than main().

Alan5142, 2022-01-19:
It is possible to raise the priority of the loop task, to accomplish that, you can use `vTaskPrioritySet` from FreeRTOS API, although I don't understand why you need to do that, tasks should not block for a long time

dsyleixa, 2022-01-19:
as far I have obeserved: when a task prio > main prio ans acquires calculation time >10sec then the watchdog timer reboots.
long calculating threads (e.g., multiple recursive calculations like the fibinacci loop in the TOP code) only work in the scheduler by prios < main prio ( =1), which means: just by 0, because then the scheduler is switching the time slices anyway.
I just would like to have more hierarchical prios below the main prio, which means: the mein prio must be erased to sth higher.

But I would like to have more options below

igrr, 2022-01-19:
As mentioned by others in the thread (https://github.com/espressif/arduino-esp32/issues/3045#issuecomment-516468368), it's not the main task priority that matters in this case. As soon as you have a long running task with priority higher than the idle task, you need to consider how that task will interact with the watchdog. So raising the priority of the main task will not give you more possible values of task priorities — any value higher than 0 will prevent the idle task from running (unless your task yields) and eventually cause a watchdog timeout.

dsyleixa, 2022-01-19:
then the watchdog will have to run by a prio higher than 1.
The basis of pre-emptive scheduling is that you do NOT have to make any cooperative arrangements, this is only necessary in cooperative MT.
With pre-eptive MT, the time slice scheduler takes over this task, provided that the thread to be interrupted has a lower priority than other threads. 
This is why perio 0 works, but 1 does not yet.
Increasinf the watchdog prio to 2 or 3 will then also work for 0,1, and perhaps 2.

dsyleixa, 2022-01-19:
then the watchdog (idle task?) will have to run by a prio higher than 1.
The basis of pre-emptive scheduling is that you do NOT have to make any cooperative arrangements, this is only necessary in cooperative MT.
With pre-emptive MT, the time slice scheduler takes over this task, provided that the thread to be interrupted has a lower priority than other threads. 
This is why prio 0 works currently, but 1 or more does not yet.
https://github.com/espressif/arduino-esp32/issues/3045#issuecomment-516401345
Increasing the watchdog (idle task?) prio to 2 or 3 will then also work for 0,1, and perhaps 2.

me21, 2022-01-20:
It's not about cooperative or preemptive, it's about hung task detection mechanism. If a task doesn't release the CPU, the idle task will not get a chance to run, the watchdog will time out and reset the system, as simple as that.

dsyleixa, 2022-01-20:
You're wrong, i'ts exactly about cooperative vs. preemptive,  and that's why the idle task has to become a higher prio.
That's also exactly what one does in LINUX when using POSIX thread polycicy SCHED__RR + .sched_priority and what happens with competing busy parrallel  threads  by equal, lower, or higher prios.
The Round-Robin always switches the time slices of equal-prio-threads without consideration if they are busy or not: if the timeslice is over, it's over, and it's the next one's turn, whilst always either higher-prio thread will keep it's control though.
OTOH it's unreasonable to have just 1 prio lower than the watchdog thread and 4 prio higher than that which zhen may lead to watchdog issues. It would be much more reasonable to have an intermediate watchdog prio and a couple of thread prio to run benath or beyond it.

dsyleixa, 2022-01-20:
You're wrong, i'ts exactly about cooperative vs. preemptive,  and that's why the idle task has to become a higher prio.
That's also exactly what one does in LINUX when using POSIX thread policy SCHED__RR + .sched_priority and what happens with competing busy parrallel  threads  by equal, lower, or higher prios.
The Round-Robin always switches the time slices of equal-prio-threads without consideration if they are busy or not: if the timeslice is over, it's over, and it's the next one's turn, whilst always either higher-prio thread will keep it's control though.
OTOH it's unreasonable to have just 1 prio lower than the watchdog idle thread and 4 prios higher than that which then may cause  watchdog issues. It would be much more reasonable to have an intermediate watchdog idle thread prio and a couple of thread prios to run beneath or beyond it.

me21, 2022-01-20:
No, you're wrong, it's not about cooperative or preemptive, it's about hung task detection.
The scheduler here is preemptive, it will not give idle task any timeslice if a higher priority task is always ready to run (for example, when it's hung, stuck in endless loop). Hence the watchdog will time out and reset the system.
Linux does not have this feature: if the task is hung there in endless loop, it may consume all the CPU time it can get.

dsyleixa, 2022-01-20:
yes, the hung detection has to run by a higher prio.
But when running the hung-detection thread at a higher prio then it can be scheduled pre-emptivelyto all other lower prio threads  whilst leaving it at prio 1 then equal or higher prio threads must have prio=0 or must handle it cooperatively.
That is exactly whyt preemtive MT is about.
Nonethekess, finally NOTHING  speaks against a higher hung-detection-idle-watchdog-thread-thing prio (e.g. 3), but everything speaks for it .

dsyleixa, 2022-01-20:
yes, the hung detection has to run by a higher prio.
But when running the hung-detection thread at a higher prio then it can be scheduled pre-emptively to all other lower prio threads  whilst leaving it at prio 1 then  threads must have prio=0 or (if at equal or higher prio) must handle it cooperatively.
That is exactly whyt preemtive MT is about.
Nonethekess, finally NOTHING  speaks against a higher hung-detection-idle-watchdog-thread-thing prio (e.g. 3), but everything speaks for it .

JTM, also in Linux it may happen that for high-prio SCHED_RR ("realtime") threads may block important system threads from running and make one's system unresponsive.

me21, 2022-01-20:
I can't see how would hung detection (idle) task work at higher priorities instead of the lowest one. We need to trigger the watchdog if any of the tasks hang, right? This means the idle task should have the lowest priority - when any task hangs (runs endlessly), it will block the idle task and the watchdog will time out.
If it had some higher priority, the lower priority task could hang and I don't see any way of knowing that then.

Cooperatively yielding the task from time to time to feed the watchdog does not make the scheduler cooperative, it is still preemptive, because scheduler type is related only to the way it switches tasks.

> JTM, also in Linux it may happen that high-prio SCHED_RR ("realtime") threads may block important system threads from running and make one's system unresponsive.

Yes, and ESP will reboot in this case to resolve the situation.

dsyleixa, 2022-01-20:
why is that so hard to understand for you?
as to Linux: exactly this can be adjusted by arbitrary program design,
` 0 <= threadlowest < commonthread < **main** < higherthread <= maxthreadprio.`

but always to reboot if a thread is consuming long cpu time is not a feature, that is a bug.
Let the programmer decide, how to handle time-consuming threads, hierarchically. Not always time-consuming threads need to reboot the entire system, and currently that's true for prio=0 threads (fibonacci, see above).
There might exist additionl threads with higher prio than fibonacci, e.g. blink(), but still no thread to system safety , nonetheless able to take over the fibonacci 0 prio.
And then there is the watchdog thread thing which can be more important  
So give us 2 or 3  prios below the watchdogthreadthing and 2 or 3 above it:
` 0 <= commonthread < **watchdogthing** < higherthread <= maxthreadprio.`

dsyleixa, 2022-01-20:
why is that so hard to understand for you?
as to Linux: exactly this can be adjusted by arbitrary program design,
` 0 <= threadlowest < commonthread < **main** < higherthread <= maxthreadprio.`

but always to reboot if a thread is consuming long cpu time is not a feature, that is a bug, given the timeslice  can be switched by the Round-Robin.
Let the programmer decide, how to handle time-consuming threads, hierarchically. Not always time-consuming threads need to reboot the entire system, and currently that's true for prio=0 threads (fibonacci, see above).
There might exist additionl threads with higher prio than fibonacci, e.g. blink(), but still no threat to system safety , nonetheless able to take over the fibonacci 0 prio.
And then there is the watchdog thread thing which can be more important  
So give us 2 or 3  prios below the watchdogthreadthing and 2 or 3 above it:
` 0 <= threadlowest < commonthread < **watchdogthing** < higherthread <= maxthreadprio.`

dsyleixa, 2022-01-20:
why is that so hard to understand for you?
as to Linux: exactly this can be adjusted by arbitrary program design,
0 <= threadlowest < commonthread < **main** < higherthread <= maxthreadprio.

but always to reboot if a thread is consuming long cpu time is not a feature, that is a bug, given the timeslice can be switched by the Round-Robin.
Let the programmer decide, how to handle time-consuming threads, hierarchically. Not always time-consuming threads need to reboot the entire system, and currently that's true for prio=0 threads (fibonacci, see above).
There might exist additional threads with higher prio than fibonacci(), e.g. primesieve(), able to take over the fibonacci 0 prio, but still no threat to system safety (because of preemptive timeslice scheduling).
And then there is the watchdog thread thing which can be more important.

So give us 2 or 3 prios below the watchdogthreadthing and 2 or 3 above it:
0 <= threadlowest < commonthread < **watchdogthing** < higherthread <= maxthreadprio.

me21, 2022-01-20:
I don't know why, but I just can't see how your proposed design is going to work.

Rebooting if the thread is taking long time is a feature, not a bug. It's a design decision to count all long-executing threads as hung. Adds robustness to the system. I agree they may be not necessarily hung, just executing a long operation, but the OS cannot tell the difference.

Why is it so hard for you to let the idle task run from time to time?

Time-consuming threads can pose a threat to the system safety if they prevent some other lower priority thread from running, just as you've shown on Linux with realtime priority. Here, this situation is resolved by reboot instead of making the system unresponsive.

In addition, if the thread is really hung, a threat to the system safety arises because some function is not properly executed.

So the system is rebooted to stay on the safe side.

If the idle task priority is risen, it won't be possible to detect hung tasks with priority less or equal to idle task priority (like now with priority == 0). So I think the devs aren't going to make this change.

dsyleixa, 2022-01-20:
I don't need to detect hung tasks if I put them arbitrarily to a prio lower than the watchdog thread.
IIcan do it now juat by prio =0, but I want to have more choices.
If I want to reboot the system when a thread might hang and I want that to be detected, then  I put them to a prio higher than the watchdog task.
I understand that you have conceptual problems in understanding the scope of the problem, and therefore it does not make much sense for me at the moment to repeat everything that has been said for the umpteenth time.
I just hope the devs have greater conceptual skills to understand, appreciate and improve on this. ;-)

Axel1400, 2022-01-20:
You are still wrong after three years, you are not understanding the purpose of Watchdog (it's a feature).
There is no need to increase the prio of idle task as you will loose hung task detections if they are low priority than idle task.
For your problem, just use Linux, you are thinking a uC need to be the same as PC.
If you still don't want to use Linux, disable Watchdog.

atanisoft, 2022-01-20:
> If the idle task priority is risen, it won't be possible to detect hung tasks with priority less or equal to idle task priority (like now with priority == 0). So I think the devs aren't going to make this change.

The idle task is intentionally created at priority zero and is documented [here](https://www.freertos.org/RTOS-idle-task.html).

> The basis of pre-emptive scheduling is that you do NOT have to make any cooperative arrangements, this is only necessary in cooperative MT.

I agree with your logic *BUT* that is not how FreeRTOS is configured or running. Refer to [this](https://www.freertos.org/single-core-amp-smp-rtos-scheduling.html) page which describes how the scheduling works, for reference it is configured with [prioritized preemptive scheduling](https://github.com/espressif/esp-idf/blob/release/v4.4/components/freertos/include/esp_additions/freertos/FreeRTOSConfig.h#L137). It is running as SMP on ESP32 devices which support it (ESP32, ESP32-S3) with a single scheduler but otherwise uses the preemptive model.

dsyleixa, 2022-01-20:
@Axel1400 :
never heard of subsumption architecture, did you?

[56 more replies]