espressif/arduino-esp32 #3046: WebServer->handleClient on CPU0 (xTaskCreatePinnedToCore) crashes with invoke_abort!

alorbach, 2019-07-30:
So I am trying to use CPU0 to handle WebServer->handleClient within a Task Loop but it crashes with this stack trace: 

```
Decoding stack results
0x4008f10c: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 155
0x4008f33d: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 170
0x400f0a9b: task_wdt_isr at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/task_wdt.c line 174
0x4008c931: vTaskExitCritical at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/tasks.c line 4304
0x4008b9f9: xQueueGenericSend at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/queue.c line 828
0x40118a36: sys_mutex_unlock at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/port/esp32/freertos/sys_arch.c line 93
0x40106300: lwip_accept_r at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/api/sockets.c line 3481
0x400d9dc7: WiFiServer::available() at C:\Users\AL.ADISCON\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2\libraries\WiFi\src\WiFiServer.cpp line 52
0x400de8b1: WebServer::handleClient() at C:\Users\AL.ADISCON\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2\libraries\WebServer\src\WebServer.cpp line 279
0x400d2fe8: TaskWebserverCode(void*) at C:\Users\AL8775~1.ADI\AppData\Local\Temp\arduino_build_205336\sketch\ambiwebserver_esp32.cpp line 59
0x4008b0a9: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143
```

It works if I have the Task created with xTaskCreate.

lbernstone, 2019-08-01:
You have an interrupt that is taking to long and tripping the watchdog timer.  Do not try to interact with hardware in interrupts.  Set a flag/semaphore/mutex to another process that can execute it.

alorbach, 2019-08-01:
Sorry to ask but I am very new to ESP32 dev, this is my Task Function: 

```
void TaskWebserverCode( void * pvParameters ){
  // Continue forever
  while(true) {
    pStaticWebServer->handleClient(); 
    yield(); 
  } 
}
```

I want to PIN this to core 0 to have the webserver more responsive:

```
    xTaskCreatePinnedToCore(
                      TaskWebserverCode,  /* Task function. */
                      "TaskWebserver",    /* name of task. */
                      10240,              /* Stack size of task */
                      this,               /* parameter of the task */
                      5,                  /* priority of the task */
                      &taskWebserver,     /* Task handle to keep track of created task */
                      0);                 /* pin task to core 1 */
```

How would I add flag/semaphore/mutex  handling here?

lbernstone, 2019-08-01:
First, take a look at https://github.com/me-no-dev/ESPAsyncWebServer.  No need to reinvent the wheel.
It is not this task that is causing the problem.  It is an interrupt handler that is being activated somewhere in the WebServer processing that is blocking.

alorbach, 2019-08-20:
I am actually using "WebServer.h" and not "ESPAsyncWebServer", I thought "ESPAsyncWebServer" is not compatible with ESP32. With ESPAsyncWebServer I know that there is no need to call "handleClient" in a loop at all.

atanisoft, 2019-08-20:
@alorbach It is 100% compatible with ESP32 and ESP8266.

alorbach, 2019-08-21:
Oh ok... I thought WebServer was some kind of ESPAsyncWebServer made for esp32 as coding convention is very simular. 

In this case I need to try ESPAsyncWebServer working in my code for esp32 again.

alorbach, 2019-10-10:
I have switched my project to ESPAsyncWebServer for esp32 successfully. Now I am running into the interrupt problem again when I try to start mjpeg streaming which is of course blocking again.

alorbach, 2019-10-14:
Fixed the problem by running the streaming code within its own task detached from the AsyncWebServerRequest Handler.

closed 2019-10-14 by alorbach

alorbach, 2020-01-08:
UPDATE, the actual problem here was that the Watchdog because I was using Core0 for the Task! 
You can use disableCore0WDT() and enableCore0WDT() to disable the watchdog. 

However using ESPAsyncWebServer was a way better method anyway.