espressif/arduino-esp32 #3047: Proxy setting for the wifi

thedim-witted, 2019-07-31:
Hi,

I am working with ESP32 and I want to connect to a wifi with proxy on it. Is there a way to enter the proxy server IP and port number for the same and connect to the wifi?

atanisoft, 2019-07-31:
A proxy is primarily used for http connections, the HTTPClient library does not currently expose support for using a proxy on the outbound connection itself, but you can do this with code similar to the below (this is 100% untested):
```
WiFiClient client;
if(client.connect(proxy_host, proxy_port)) {
  HTTPClient http;
  if(http.begin(client, url)) {
    // do something with the http client...
  }
}
```
If you are needing to create a tunnel to a remote host through the proxy it is doable, instead of using HTTPClient inside the connect condition check you could follow the process listed [here](https://en.wikipedia.org/wiki/HTTP_tunnel).

thedim-witted, 2019-07-31:
I don't want to do a tunneling. The place where I have to set up my device has a proxy server set-up over it's entire internet connection. I need to access the internet to send the data over.

a-c-sreedhar-reddy, 2019-07-31:
https://gitter.im/espressif/arduino-esp32?utm_source=share-link&utm_medium=link&utm_campaign=share-link  you can ask these questions here instead of issues

closed 2019-07-31 by thedim-witted

thedim-witted, 2019-07-31:
I have one doubt though. With this implementation we will not be able to use the functions of the WiFiClient library as it is connecting through the HTTPClient, right? So, is there a way to tunnel it back to WiFiClient and take the advantage of the library?

atanisoft, 2019-07-31:
HTTPClient uses a WiFiClient instance internally as the transport layer. Are you trying to make an HTTP request via HTTP proxy or are you trying to tunnel non-HTTP traffic through an HTTP proxy?

For the tunneling case, instead of using the HTTPClient you can directly write "CONNECT host:port HTTP/1.0\r\n" and wait for the 200 OK to come back (see the wikipedia link for specifics) and then you can use the client as-is since it will a proxied connection to the remote host/port. Note though that the proxy connect may fail!