espressif/arduino-esp32 #3049: ESP32 freezes when it restarts during WiFi.begin or WiFi.mode

thehellmaker, 2019-07-31:
Show the shortest possible code that will duplicate the error

Show the EXACT error message(it doesn't work is not enough)

### Hardware:
Board:							?ESP32 Dev Module?
Core Installation/update date:		?11/jul/2017?
IDE name:						?Arduino IDE?
Flash Frequency:					?40Mhz?
PSRAM enabled:                                      ?no?
Upload Speed:					?115200?
Computer OS:                                           ?Mac OSX?

### Description:
I am using an ESP32 NodeMCU development board and using esp32-arduino using platformio. ESP32 hangs/freezes and watchdog doesn't restart the system in a peculiar case. Most of the times it restarts but in some peculiar case it freezes.

### StackTrace 

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>
#include "WiFi.h"
#include "esp_task_wdt.h"
#include "esp_int_wdt.h"
#include "esp32-hal.h"

static volatile long lastACMainsTimestamp = 0;
static volatile int current_count_ac[] = {0,0,0,0};

void IRAM_ATTR recordACInterrupt0() {
	ESP.restart();
	current_count_ac[0]++;
}

void IRAM_ATTR acMainsInterrupt() {
 	lastACMainsTimestamp = micros();
}

void setup() {
  enableLoopWDT();
  enableCore0WDT();
  enableCore1WDT();
  Serial.begin(115200);
  Serial.setDebugOutput(true);
  pinMode(33, INPUT_PULLUP);
	attachInterrupt(digitalPinToInterrupt(33), acMainsInterrupt, CHANGE);
  pinMode(23, INPUT_PULLUP);
	attachInterrupt(digitalPinToInterrupt(23), recordACInterrupt0, CHANGE);
  WiFi.mode(WIFI_STA);
  WiFi.disconnect(true);
  WiFi.begin("ssid-test", "password-test");
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

thehellmaker, 2019-08-01:
Related Issue: https://github.com/espressif/arduino-esp32/issues/2686 

I am testing with disabling interrupt during WiFi connect as suggested by @atanisoft

closed 2019-10-14 by stale[bot]

atanisoft, 2022-10-07:
@roysG If you have a reproducible test case for this please open a new issue and it can be reviewed. If there is a bug in the arduino-esp32 code it can be fixed, if the bug is in your code someone will likely be able to provide advise on how to fix.

VojtechBartoska, 2022-10-07:
Hello @roysG, yea, please raise new issue if needed.

Thanks for nice answer @atanisoft ;)