espressif/arduino-esp32 #3050: How to set esp-idf commit version properly

Nathan-ma, 2019-08-01:
Guys, its a super simple and really quick question.

I wanna use the new bt.c commit to fix a bug on my a2dp_source code, I'm currently using arduino-esp32 1.0.2, this new commit is on esp-idf it self and it was made 28 days ago, my idea is to use arduino as a component with the last usable esp-idf commit for the arduino-esp32 in order to generate the lib32.a file and compile it all using arduino ide.

the thing is, I'm not been able to get the right commit from the esp-idf github repository in order to run 1.0.3 arduino-esp32.

Can you guys help me with the command? on the 1.0.3 says I need to set esp-idf to v3.2 977854975

I tried git checkout v3.2 977854975 with no succes, tried to clone the repository 977854975 no success either.

Any good soul can please give me a simple git command line just to set the commit to this v3.2 977854975? thank you

lbernstone, 2019-08-01:
https://github.com/espressif/arduino-esp32/issues/1142#issuecomment-388913018
as of today:
   Update IDF to a8916daeb (#2992)

Nathan-ma, 2019-08-01:
Thank you @lbernstone , gonna close this topic

closed 2019-08-01 by Nathan-ma