espressif/arduino-esp32 #3052: Substantial differences from normal Arduino in SD library

embedded-creations, 2019-08-01:
There seem to be quite a few differences in the library behavior that are making it very difficult to write an Arduino sketch that is compatible with both libraries.  Compare the output from [this example sketch](https://www.arduino.cc/en/Reference/FileIsDirectory) on both ESP32 and Teensy:

ESP32

```
/gifs/
    /gifs/bigbuck1-64x32.gif        148635
    /gifs/._bigbuck1-64x32.gif      4096
    /gifs/ezgif-259182266.gif       199781
    /gifs/._ezgif-259182266.gif     4096
```

Teensy
```
GIFS/
    BIGBUC~1.GIF        148635
    _BIGBU~1.GIF        4096
    EZGIF-~1.GIF        199781
    _EZGIF~1.GIF        4096
```

More details here: https://github.com/espressif/arduino-esp32/issues/1036#event-2526032254

atanisoft, 2019-08-01:
this is not limited to just SD, the underlying VFS layer uses the fully qualified path as the value for name. It looks like it might be possible to return the last segment of the name (everything after the last /) though. @me-no-dev thoughts?

everslick, 2019-08-03:
the reason being, I guess, to maintain SPIFFS compatibility, which does not have directories. also it looks like as Teensy lacks support for VFAT names.

I'd suggest setting your filessystem structure to the lowest common denominator, if you need to support both with the same code.

embedded-creations, 2019-10-11:
It's unfortunate that the stale bot seems to be doing more commenting on issues than espressif.

atanisoft, 2019-10-11:
@embedded-creations Feel free to submit a PR that helps to bring the two libraries more into alignment.

closed 2019-12-24 by stale[bot]